areaDetector/ADCore #280: NDPluginDriver effectively limits the maximum size of arrayPort to 20 chars

lerwys, 2017-06-22:
Hello everyone,

Although the constructor can correctly set the parameter NDPluginDriverArrayPort to any string length, the function connectToArrayPort () limits the maximum number of chars to 20, as seen here in the current master:
https://github.com/areaDetector/ADCore/blob/515a7d80ba8b0ee0301b67e3ccaf31b478316ed0/ADApp/pluginSrc/NDPluginDriver.cpp#L518, breaking plugins using more than 20 chars for the arrayPort.

How do we proceed from here?

A couple of options from the top of my head:

1) We could create a new function in the paramVal class like `char*` `paramVal::getStringLength()` and then the associated asynPortDriver `asynPortDriver::getStringParamLength ()` methods. With this we can allocate the string dynamically.

2) Maybe we could have the paramVal class allocate this string automatically for us and return it like `char* paramVal::getAndAllocString()` and the associated asynPortDriver `asynPortDriver::getStringParam(int index, char **value)`

Or maybe there is a more elegant solution?

Thanks,

Lucas Russo

MarkRivers, 2017-06-22:
There is already a variant of asynPortDriver::getStringParam that takes an std::string argument rather than char * argument.  It does not take a maximum length argument.

So this patch should fix the problem:
```
corvette:ADCore/ADApp/pluginSrc>git diff NDPluginDriver.cpp
diff --git a/ADApp/pluginSrc/NDPluginDriver.cpp b/ADApp/pluginSrc/NDPluginDriver.cpp
index 4d6cb59..14d6261 100644
--- a/ADApp/pluginSrc/NDPluginDriver.cpp
+++ b/ADApp/pluginSrc/NDPluginDriver.cpp
@@ -515,11 +515,11 @@ asynStatus NDPluginDriver::connectToArrayPort(void)
     asynStatus status;
     asynInterface *pasynInterface;
     int enableCallbacks;
-    char arrayPort[20];
+    std::string arrayPort;
     int arrayAddr;
     static const char *functionName = "connectToArrayPort";

-    getStringParam(NDPluginDriverArrayPort, sizeof(arrayPort), arrayPort);
+    getStringParam(NDPluginDriverArrayPort, arrayPort);
     getIntegerParam(NDPluginDriverArrayAddr, &arrayAddr);
     getIntegerParam(NDPluginDriverEnableCallbacks, &enableCallbacks);

@@ -534,11 +534,11 @@ asynStatus NDPluginDriver::connectToArrayPort(void)
     this->connectedToArrayPort_ = false;

     /* Connect to the array port driver */
-    status = pasynManager->connectDevice(this->pasynUserGenericPointer_, arrayPort, arrayAddr);
+    status = pasynManager->connectDevice(this->pasynUserGenericPointer_, arrayPort.c_str(), arrayAddr);
     if (status != asynSuccess) {
         asynPrint(this->pasynUserSelf, ASYN_TRACE_ERROR,
                   "%s::%s Error calling pasynManager->connectDevice to array port %s address %d, status=%d, error=%s\n",
-                  driverName, functionName, arrayPort, arrayAddr, status, this->pasynUserGenericPointer_->errorMessage);
+                  driverName, functionName, arrayPort.c_str(), arrayAddr, status, this->pasynUserGenericPointer_->errorMessage);
         return (status);
     }

@@ -547,7 +547,7 @@ asynStatus NDPluginDriver::connectToArrayPort(void)
     if (!pasynInterface) {
         asynPrint(this->pasynUserSelf, ASYN_TRACE_ERROR,
                   "%s::connectToPort ERROR: Can't find asynGenericPointer interface on array port %s address %d\n",
-                  driverName, arrayPort, arrayAddr);
+                  driverName, arrayPort.c_str(), arrayAddr);
         return(asynError);
     }
     pasynGenericPointer_ = (asynGenericPointer *)pasynInterface->pinterface;
```
This compiles OK.  Can you test it and if it is OK I will push the change.

lerwys, 2017-06-22:
Hi Mark,

Nice! Thanks for the fast feedback.
I will compile master ADCore and asyn and test them together your patch.

Thanks!

Lucas

lerwys, 2017-06-23:
Hi again,

I've tested my IOC with the master asyn (15f744b) + ADCore-R2-2 (would that be a problem?), without the fix to see if I could reproduce the error first, but I've got an exception from asynPortDriver:

```
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_S_construct null not valid
```

Stack trace:

```
Program received signal SIGABRT, Aborted.
0x00007ffff15301d7 in raise () from /usr/lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install glibc-2.17-157.el7_3.2.x86_64 hdf5-1.8.12-8.el7.x86_64 jbigkit-libs-2.0-11.el7.x86_64 libgcc-4.8.5-11.el7.x86_64 libjpeg-turbo-1.2.90-5.el7.x86_64 libstdc++-4.8.5-11.el7.x86_64 libtiff-4.0.3-27.el7_3.x86_64 libxml2-2.9.1-6.el7_2.3.x86_64 ncurses-libs-5.9-13.20130511.el7.x86_64 readline-6.2-9.el7.x86_64 szip-2.1-1.sdl7.x86_64 xz-libs-5.2.2-1.el7.x86_64 zlib-1.2.7-17.el7.x86_64
(gdb) bt
#0  0x00007ffff15301d7 in raise () from /usr/lib64/libc.so.6
#1  0x00007ffff15318c8 in abort () from /usr/lib64/libc.so.6
#2  0x00007ffff1b329d5 in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib64/libstdc++.so.6
#3  0x00007ffff1b30946 in ?? () from /usr/lib64/libstdc++.so.6
#4  0x00007ffff1b30973 in std::terminate() () from /usr/lib64/libstdc++.so.6
#5  0x00007ffff1b30b93 in __cxa_throw () from /usr/lib64/libstdc++.so.6
#6  0x00007ffff1b85757 in std::__throw_logic_error(char const*) () from /usr/lib64/libstdc++.so.6
#7  0x00007ffff1b915b1 in char* std::string::_S_construct<char const*>(char const*, char const*, std::allocator<char> const&, std::forward_iterator_tag) () from /usr/lib64/libstdc++.so.6
#8  0x00007ffff1b91968 in std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&) () from /usr/lib64/libstdc++.so.6
#9  0x00007ffff796eaee in paramList::setString (this=0x712ba0, index=index@entry=8, value=<optimized out>)
    at ../../asyn/asynPortDriver/asynPortDriver.cpp:241
#10 0x00007ffff796ec08 in asynPortDriver::setStringParam (this=0x712fd0, list=0, index=8, value=<optimized out>)
    at ../../asyn/asynPortDriver/asynPortDriver.cpp:1298
#11 0x00007ffff796d05d in asynPortDriver::setUInt32DigitalInterrupt (this=0x712fd0, list=10, index=0, mask=<optimized out>, 
    reason=<optimized out>) at ../../asyn/asynPortDriver/asynPortDriver.cpp:1201
#12 0x00007ffff796db9d in asynPortDriver::getStringParam (this=0x712fd0, list=0, index=0, maxChars=<optimized out>, value=<optimized out>)
    at ../../asyn/asynPortDriver/asynPortDriver.cpp:1454
#13 0x00007ffff58cbdc9 in asynNDArrayDriver::asynNDArrayDriver (this=0x712fd0, portName=0x7129a0 "SI-04C1:DI-BPM-1:", maxAddr=48, 
    numParams=<optimized out>, maxBuffers=0, maxMemory=25000000, interfaceMask=4410, interruptMask=4408, asynFlags=3, autoConnect=1, priority=0, 
    stackSize=0) at ../asynNDArrayDriver.cpp:662
#14 0x00007ffff7bcde5f in drvBPM::drvBPM (this=0x712fd0, portName=0x7129a0 "SI-04C1:DI-BPM-1:", endpoint=0x7129b2 "ipc:///tmp/malamute", 
    bpmNumber=15, verbose=0, timeout=2000, maxPoints=100000, maxBuffers=0, maxMemory=25000000) at ../drvBPM.cpp:565
#15 0x00007ffff7bd0ab1 in drvBPMConfigure (portName=0x7129a0 "SI-04C1:DI-BPM-1:", endpoint=0x7129b2 "ipc:///tmp/malamute", bpmNumber=15, 
    verbose=0, timeout=2000, maxPoints=100000, maxBuffers=0, maxMemory=25000000) at ../drvBPM.cpp:4301
#16 0x00007ffff7bd0b0d in initCallFunc (args=<optimized out>) at ../drvBPM.cpp:4327
#17 0x00007ffff5da1cc1 in iocshBody (pathname=<optimized out>, commandLine=0x0) at ../../../src/libCom/iocsh/iocsh.cpp:771
#18 0x0000000000406b32 in main (argc=<optimized out>, argv=<optimized out>) at ../BPMMain.cpp:17
```

Maybe this doesn't happen with the current ADCore?

MarkRivers, 2017-06-23:
Are you sure that you rebuilt areaDetector completely after updating asyn?

At the top-level of ADCore
```
make clean uninstall
make
```
I think ADCore R2-2 should build OK with the latest asyn, it should be backwards compatible.

lerwys, 2017-06-23:
Looking at asynNDArrayDriver.cpp from both ADCore-R2-2 and ADCore master (515a7d8), the offending function call `setStringParam (NDPortNameSelf, portName);`, available at line 662 in ADCore-R2-2 and at line 758 in ADCore master (515a7d8) looks the same. So, I'm not sure if ADCore version would change anything.

lerwys, 2017-06-23:
Yes Mark. You're right.

Now, I recompiled ADCore and applied the patch. After that the problem was fixed.

Thanks for all the help!

MarkRivers, 2017-06-23:
OK, good news.  I have pushed that change to the master branch of ADCore on Github.

closed 2017-06-23 by MarkRivers