areaDetector/ADCore #281: Add Direct Chunk Write feature to HDF5 file writer plugin

ulrikpedersen, 2017-06-28:
The performance of the internal HDF5 library pipeline when doing basic I/O operations can be a bottleneck for high datarates (somewhere above 1GB/s). 

The HDF Group have implemented a way to circumvent this internal pipeline (and chunk cache) with the [Direct Chunk Write](https://support.hdfgroup.org/HDF5/doc/Advanced/DirectChunkWrite/) feature.

See also tech-talk discussion: http://www.aps.anl.gov/epics/tech-talk/2017/msg01034.php

We should add the Direct Chunk Write feature to the HDF5 file writer.

ulrikpedersen, 2017-06-28:
There are some subtleties to using Direct Chunk Write. It cannot use the internal HDF5 chunk cache so our chunk configuration would have to be restricted to fit within an individual frame. Otherwise we would have to implement our own pipeline with copies and chunk cache...

xiaoqiangwang, 2017-06-29:
Since that requires an pre-compressed NDArray, would it be a new NDPluginCompressor? That would bring it in line with NTNDArray specification.

ulrikpedersen, 2017-06-29:
> Since that requires a pre-compressed NDArray

No, it does not __require__ a pre-compressed NDArray. Direct Chunk Write works just as well (and with a performance increase) with uncompressed data.

> would it be a new NDPluginCompressor?

It doesn't have to be (see my comment just above) - but it is an interesting suggestion and I would like to see something like that.

> That would bring it in line with NTNDArray specification.

Again, that would be a good idea to have but there is no need to tightly couple the work of adding Direct Chunk Write functionality to changing the NDArray definition.

It is possible to "simply" add Direct Chunk Write functionality without coupling it with major changes to ADCore and NDArrays. I like the idea of small incremental changes to continuously improve performance/usability.

xiaoqiangwang, 2017-06-29:
Ok. I was assuming the file writer will be similar to the Dectris Eiger detector.

The compressor plugin would equally reduce the traffic when transferred though NDStdArray plugin.

ulrikpedersen, 2018-12-07:
That's effectively #349 

I agree this has sat around for long enough now without any real effort. I'll discuss priorities with @ajgdls, @GDYendell and @skyfrench.

GDYendell, 2018-12-13:
I am starting work on this now.

closed 2019-05-22 by ulrikpedersen