areaDetector/ADCore #291: Problem with NDPluginFile when AutoSave=1

MarkRivers, 2017-09-28:
When using the NDFileTIFF plugin, AutoSave=1 and high frame rate in ADSimDetector the IOC crashes.

The problem has been traced to 2 different threads writing files, the NDFileTIFF_Plugin1 thread (which is correct) and the main NDFileTiff asyn port thread (which is incorrect).  The problem has almost certainly been introduced by the changed behavior of the busy record asyn device support in the current master branch of epics-modules/busy.  The record now processes whenever it gets a callback.  If it is processing because of a callback it should not call the driver. 

NDPluginFile sets NDWriteFile to 1 and does callbacks each time it writes a file in autosave mode.  It then sets NDWriteFile to 0 when the write is complete.  This is done in order to give feedback in the OPI that a write operation is happening.  The problem is that if the callbacks come too quickly the busy record actually **does** call the driver.  This means that a second thread is calling the TIFF library at the same time as the plugin thread, which leads to corruption and crashes. 

I have temporarily fixed the problem with 9c14ce950281f358eaaa4a2459efaecefd03c0e2. This removes the calls to set NDWriteFile to 1, so the busy record won't get these callbacks.

The correct solution is to fix the asyn device support for the busy record so that it does not call the driver due to callbacks, even if they come at a very fast rate.  It currently detects if multiple callbacks have occurred since it last processed, but it does not handle it correctly.

ulrikpedersen, 2017-10-10:
is this related to #287 ?

MarkRivers, 2017-10-10:
>is this related to #287 ?

It might be the same problem. The problem I am seeing is that when the busy record gets multiple callbacks quickly it can mistakenly send the readback value to the driver during record processing, rather than just processing the record with the new value and not sending it to the driver.  When it sends 1 to the driver this can cause 2 threads to be talking to the file library (TIFF, HDF5, etc.) at the same time, which causes crashes.  

#287 is a problem at iocInit.  I am not sure that there are multiple callbacks there, but possibly.  It will be interesting to see if #287 goes away once the busy record is fixed.

The correct fix is in asyn device support.

MarkRivers, 2017-10-24:
This is a more detailed description of the issue.  The problem is that devBusyAsyn.c does  behave properly when there are rapid callbacks from the driver.  When there are callbacks from the driver this is what should happen:
- The record should process once for each callback value
- The device support should detect that it is being called due to a driver callback, and should not call the driver with this value from the callback.

The version of devBusyAsyn.c starting with 2c1cfca916767651b92d90bc8ef81f07b1212b79 fails to do both of these things.
- It only had a single variable to store the callback values from the driver, not a ring buffer.  So it even when it processed multiple times it was always using the last callback value, not all callback values.
- It did have a counter to keep track of how many callbacks had occurred since the record last processed.  However, it did not decrement this counter when it processed, it set it to zero and printed a warning message.
- Because each driver callback did result in a call to scanOnce(), the record would process N times if there were N callbacks in a row before it processed.  But only the first one was properly handled and did not result in a call to the driver.  N-1 times it would call the driver.

There was an additional problem with the logic.  It did not correctly distinguish between processing due to driver callbacks, and processing due to a real write to the record.  If the record is actually written to this value must be sent to the driver, but this was not guaranteed if the driver was doing rapid callbacks.  

This is a problem that is shared with the standard asyn device support for output records that have the info tag asyn:READBACK.  They also do not have a way to distinguish between processing due to a record callback and that due to a write to the record.

MarkRivers, 2017-11-05:
The busy record asyn device support has been fixed.  The problem no longer occurs.  I tested with the TIFF file plugin with AutoSave=1 and the simDetector running at ~500 frames/s.

closed 2017-11-05 by MarkRivers