areaDetector/ADCore #292: Overlay Plugin Behaviour Changes

MJGaughran, 2017-10-19:
The updated Overlay plugin (as of cf92fbf4974fbccfdd6ec757dc45685017513a65) appears to have a number of inconsistent / incorrect behaviours. The important issues are (with reference to the cross overlay in particular):

* The centre of the cross moves when changing the size.
* The top-left corner of the cross is now bounds-checked, as opposed to the centre, preventing it from being used near the edge of the image.
* Setting both overlay position values outside of the image area causes the IOC to crash.

I am using the version as of ADCore 2-6. Looking through later commits I did not notice any changes for the above behaviour, except maybe the last point.

According to the commit message, the change to using the corner as the position parameter was done for consistency. For our users, however, the centre is a much more useful parameter. It also removes rounding issues with cross placement.

Is it possible to revert back to using the centre as the position parameter?

It would be useful if we can work towards a common solution to the above. I can provide a pull request once we have agreed on this.

MarkRivers, 2017-10-19:
Thanks for finding these problems.

> According to the commit message, the change to using the corner as the position parameter was done for consistency. For our users, however, the centre is a much more useful parameter. It also removes rounding issues with cross placement.

The position parameters do indeed now control the upper left position of the cross.  However, all shapes now also have CenterX and CenterY records, so you can control the center.  Are you using these?

> The centre of the cross moves when changing the size.

Yes that is now true.  This was always true for the other shapes, so the behavior is now consistent.  When changing the size a decision needs to be made as to whether PositionX/Y or CenterX/Y is preserved.  Currently Position is preserved, but we could change this to Center.  That is probably what would be desired in most cases.

> The top-left corner of the cross is now bounds-checked, as opposed to the centre, preventing it from being used near the edge of the image.

I can reproduce this problem for the X=0 and Y=0 edges. It does not bounds check at X=XMax and Y=YMax edges of the image.  However, when PositionX > XMax then the vertical bar of the cross wraps around to the left edge of the image. Is this what you see?  There are definitely bugs here that need to be fixed.  The bounds checking should be done on the final drawn pixel locations, not on the center or position.

> Setting both overlay position values outside of the image area causes the IOC to crash.

I cannot reproduce this.  Can you post settings you used to cause the crash?  I have tested with the simDetector with 1024x1024 images, Cross shape, overlay SizeX=200, SizeY=100, PostionX=2000, PositionY=2000 and it does not crash.  I also tested -500,-500 and that does not crash either.

MJGaughran, 2017-10-19:
> Thanks for finding these problems.

> > According to the commit message, the change to using the corner as the position parameter was done for consistency. For our users, however, the centre is a much more useful parameter. It also removes rounding issues with cross placement.

> The position parameters do indeed now control the upper left position of the cross. However, all shapes now also have CenterX and CenterY records, so you can control the center. Are you using these?

I had already modified our screens for this.

> > The centre of the cross moves when changing the size.

> Yes that is now true. This was always true for the other shapes, so the behavior is now consistent. When changing the size a decision needs to be made as to whether PositionX/Y or CenterX/Y is [...]

I had already made that change; it was how I first noticed the rounding errors.

> > The top-left corner of the cross is now bounds-checked, as opposed to the centre, preventing it from being used near the edge of the image.

> I can reproduce this problem for the X=0 and Y=0 edges. It does not bounds check at X=XMax and Y=YMax edges of the image. However, when PositionX > XMax then the vertical bar of the cross wraps [...]

Yes, that is correct.

As with the above point, it is possible to get around these issues, but making it behave as before is simpler if we keep the centre as the position parameter.

We don't have to care about the rounding errors; I'm just not convinced its worth introducing them for no benefit.

> > Setting both overlay position values outside of the image area causes the IOC to crash.

> I cannot reproduce this. Can you post settings you used to cause the crash? I have tested with the simDetector with 1024x1024 images, Cross shape, overlay SizeX=200, SizeY=100, PostionX=2000, PositionY=2000 and it does not crash. I also tested -500,-500 and that does not crash either.

It crashes when it processes an overlay (at 10 frames / s, in my case), not when you set the position value. I will test this out on the sim detector, and give you the settings if it still occurs.

MarkRivers, 2017-10-19:
For the crash issue I was testing with simDetector at 20 frames/s and it did not crash when processing the overlays.

MJGaughran, 2017-10-19:
I cannot reproduce the issue using simDetector.

MarkRivers, 2017-10-19:
> As with the above point, it is possible to get around these issues, but making it behave as before is simpler if we keep the centre as the position parameter.

I don't want the PositionX/Y to be the center for the cross because that is inconsistent with the other shapes.  With the current version if one has both a cross and a rectangle with the same Size and Position or Center then the result is a cross circumscribed by the rectangle.  I believe that is the logical behavior, but it was not the old behavior.  The old behavior was that the cross would be centered on the upper left corner of the rectangle.

I propose to do the following:
- Fix the clipping problems.  This would be done by allowing PositionX/Y and CenterX/Y to have any values, i.e. inside or outside of the image.  Each pixel in the object will be tested to see if it is inside the image.  If it is inside then it will be added to the vector of active pixels for that overlay, otherwise it is ignored.
- For both X and Y remember wheher Position or Center were last set.  If Position was last set then preserve Position when Size is changed.  If Center was last set then preserve Center when Size is changed.

Can you explain the rounding problem you have observed?

MJGaughran, 2017-10-20:
If I make the (naive) change to keep the centre constant with size then it will jump across by one pixel depending on whether the size is odd or even, for example.

It has similar behaviour at the moment when switching between setting centre and position (put a value in the position box, then press enter on the new centre. Then press enter on the new position). This is because all values ending in .5 are rounded up, so there is an asymmetry.

I think I have a solution that suits both our needs:

* Fix the clipping problems, as in your comment.
* Add a new parameter for the resizing mode.
* When setting the size (with centre to be fixed), or the centre itself, increment the size value if it is an odd number. Then both position and centre are integers.
* Autosave the position, size and resizing mode (but not the centre).

This also gets rid of the duplicated autosave entries. Do you have any issues with the above?

MJGaughran, 2017-10-20:
I still have the segmentation fault when using simDetector, but it is less consistently failing (I used the wrong version before).

I think the problem is caused by no CLIPX when drawing the horizontal line. pValue is then pointing to an arbitrary memory location in setPixel.

This section needs to be modified anyway, as you mentioned above (the vertical line still displays when you set X too high).

MarkRivers, 2017-10-23:
I have fixed the problems.  I did it slightly differently than @MJGaughran proposed.  The new version is in the overlay_fix branch.  This is the commit message:
```
    Fixed problems:
    - Overlays were previously constrained to fit in image on X=0 and Y=0 edges.
      Users may want part of the overlay outside the image area.
      The location of the overlay can now be set anywhere, including negative positions.
      Each pixel in the overlay only added if it is within the image area.
    - Fixed problems with incorrect drawing and crashing if part of overlay was outside image area.
    - Removed rounding when setting center from position or position from center.
      This was causing the location to change when setting the same center or position.
    - Changed cross so that it is drawn symmetrically, same number of pixels on each side of center.
      This means the actual size is 2*Size/2 + 1, which will be Size+1 if Size is even.
    - Previously the Position was always preserved when Size was changed.
      This was not the desired behavior when the user had set the Center rather than Position.
      Now the code remembers whether Position or Center was last set, and preserves the
      appropriate value when the Size is changed.
```
I did not add a new parameter for the resizing mode.  I just remember whether Position or Center was last set, and preserve that value when Size is changed.  This simplifies the user interface and means we don't need new OPI screens.

By removing all rounding I believe I have fixed the problem with overlays moving when pressing Enter in Position or Center.

> Autosave the position, size and resizing mode (but not the centre).

I think you may have been seeing a problem because you were using R2-6 which was restoring both Position and Center.  This was fixed in cd298202e4bbae3b35c29df39f59d3e5350bb0cb which removed PINI=YES from CenterX/Y.  They are still in autosave but the records are not processed at iocInit.

Please test the new branch and see if it fixes your problems.

MJGaughran, 2017-10-31:
I got it working in our R2-6 ADCore. I haven't been able to find any bugs in it, and it solves my problem.

Because the resize mode isn't autosaved, it changes whenever the IOC is restarted. I didn't find a way to correct that, but I'm not too worried since the scientists usually resize alongside position changes.

Thank you very much, Mark.

MarkRivers, 2017-10-31:
OK, good news I will merge it.

MarkRivers, 2017-10-31:
Merged.

closed 2017-10-31 by MarkRivers