areaDetector/ADCore #294: support caQtDM caCamera widget with a .ui screen

prjemian, 2017-11-09:
caQtDM has a widget that will visualize the image from AreaDetector.  It is useful to provide a .ui screen for caQtDM that has good default values specified for the necessary PVs.

prjemian, 2017-11-09:
Here is an example of such a screen using the sim detector from AD 2.5 and caQtDM 4.1.6 (from the master branch):

[image]

monochromatic, 8 bits/pixel, 2000x1500 image binned by 50 in both x&y (which is why the image looks chunky)

The pop-up shows the connected PVs.
ad_cam_image.ui, macro: P=13SIM1:,R=cam1:

Important configuration parameters are shown here:

[image]

prjemian, 2017-11-09:
This is helpful since it relieves the need to install a separate image visualization tool such as ImageJ if one just wants to watch the images.

MarkRivers, 2017-11-09:
That's nice.  Does it support color?

prjemian, 2017-11-09:
see: https://github.com/caqtdm/caqtdm/blob/Development/caQtDM_QtControls/src/cacamera.cpp#L1179

I believe that is what the channelCode is used for but @timmmooney tells me that this must be a PV that provides the value of "1".  If we use the `$(P)$(R)ColorMode (mbbo)` for channelCode, "1" means a "RGB1" image.

For now, it works with `DataType = Uint8` and `ColorMode = Mono`

Selecting `DataType = Int8` and `ColorMode = Mono` replaces the image with text that says:

```
specified camera conversion not supported
code=1 bpp=1: monocrome 1bpp
code=1 bpp=2: monocrome 2bpp
code=1 bpp=3: monocrome 2bpp (used only 12bits)
code=3 bpp=3: color rgb 3bpp (r,g,b)
```

It may be possible update this with the caCalc widget making the conversion from what the ColorMode  PV provides and what the channelCode property expects.  That would enable RGB support.

prjemian, 2017-11-09:
Here is where the *channelCode* property (`m_code`) is used in the caQtDM source:
https://github.com/caqtdm/caqtdm/blob/Development/caQtDM_QtControls/src/cacamera.cpp#L295

Is there a different areaDetector PV that can be used to show color?

MarkRivers, 2017-11-10:
The code assumes that m_code is 1 for mono, and 3 for rgb.  In mono mode m_bpp (bytes per pixel) is 1 for 8bit, 2 for 16-bit, 3 for 12-bit in 16-bit data.  If m_code is 3 then it only supports m_bpp=3, i.e. 8-bit data.  Your screen maps both m_code and m_bpp to DataType_RBV.  This does not seem correct.  The DataType_RBV comes from this enumeration:
```
/** Enumeration of NDArray data types */
typedef enum
{
    NDInt8,     /**< Signed 8-bit integer */
    NDUInt8,    /**< Unsigned 8-bit integer */
    NDInt16,    /**< Signed 16-bit integer */
    NDUInt16,   /**< Unsigned 16-bit integer */
    NDInt32,    /**< Signed 32-bit integer */
    NDUInt32,   /**< Unsigned 32-bit integer */
    NDFloat32,  /**< 32-bit float */
    NDFloat64   /**< 64-bit float */
} NDDataType_t;
```
The ColorMode_RBV PV uses this enumeration:
```typedef enum
{
    NDColorModeMono,    /**< Monochromatic image */
    NDColorModeBayer,   /**< Bayer pattern image, 1 value per pixel but with color filter on detector */
    NDColorModeRGB1,    /**< RGB image with pixel color interleave, data array is [3, NX, NY] */
    NDColorModeRGB2,    /**< RGB image with row color interleave, data array is [NX, 3, NY]  */
    NDColorModeRGB3,    /**< RGB image with plane color interleave, data array is [NX, NY, 3]  */
    NDColorModeYUV444,  /**< YUV image, 3 bytes encodes 1 RGB pixel */
    NDColorModeYUV422,  /**< YUV image, 4 bytes encodes 2 RGB pixel */
    NDColorModeYUV411   /**< YUV image, 6 bytes encodes 4 RGB pixels */
} NDColorMode_t;
```
I don't see how this is working at all assuming m_code comes from channelCode and m_bpp comes from channelBPP in your camera property screen.  This is because 8-bit data would have m_code=m_bpp=DataType_RBV=0 which is not handled in either of the switch statements.  I'm confused.

MarkRivers, 2017-11-10:
Sorry, I missed your previous comment explaining that it worked with UInt8 and not Int8.  That makes sense.  It doesn't seem like this widget was really designed for areaDetector, but with some fiddling it could work.

prjemian, 2017-11-10:
I agree.  For now, it works with Mono, Uint8.  That covers a lot of use at our beam lines.

HelgeBrands, 2017-11-10:
The plan for the camera widget is to support some color modes, but yet only the RGB Mode ((m_code == 3) && (m_bpp == 3)) is implemented but not fully tested (too less color cams).

Best regards
                Helge

AntonMezger, 2017-12-20:
It should be straight forward to implement new routines into the cacamera widget to support the other cameras. We did not do this, while we do not have the hardware to test them. However feel free to submit us with these routines and we would implement them in new releases.

MarkRivers, 2017-12-20:
You can test all data types and color modes with the ADSimDetector, you don't need any hardware.

AntonMezger, 2017-12-22:
Dear Mark,

I have installed a softioc with the ADSim module in order to look into this beginning of next year.

Best regards, merry Christmas and a happy new year

Anton

MarkRivers, 2017-12-22:
Hi Anton,

Great, thanks very much.

Merry Christmas to you as well.

Mark

kmpeters, 2018-10-29:
@AntonMezger, has there been any progress on supporting more image modes in the caCamera widget?

AntonMezger, 2018-10-29:
Hi,

All modes of areadetector and even more are now implemented. Take a look tests.ui

Best regards

Anton

PR #295 (2017-12-19): the caQtDM camera screen as described
fixes #294