areaDetector/ADCore #297: apply adl2ui conversion for caQtDM

prjemian, 2017-11-17:
like https://github.com/areaDetector/ADSimDetector/issues/4

MarkRivers, 2017-12-19:
I have converted the ui screens using adl2ui from caQtDM 4.1.7.  These are committed to Github.

ADCore/ADApp/op/Makefile will now do the conversions from adl/ to ui/autoconvert, edl/autoconvert, and opi/autoconvert.

It relies on RULES_OPI which has been created in https://github.com/EPICS-synApps/configure/blob/master/RULES_OPI.  The definitions of the paths to the executables for adl2ui, adl2edl, and css are defined in https://github.com/EPICS-synApps/configure/blob/master/CONFIG_SITE.linux-x86_64.  

This arrangement allows these files to be accessed from any module in the synApps tree.  I will begin to implement it for all areaDetector drivers, asyn, quadEM, etc.

MarkRivers, 2017-12-19:
Closing this issue since it is now resolved.

closed 2017-12-19 by MarkRivers