areaDetector/ADCore #301: ntndArrayConverter.cpp does not compile with EPICS 7.0.1

MarkRivers, 2017-12-05:
ntndArrayConverter.cpp does not compile with EPICS 7.0.1

The error when building on Linux is:
```
make[1]: Entering directory `/home/epics/devel-base7/areaDetector-3-1/ADCore/ADApp/ntndArrayConverterSrc/O.linux-x86_64'
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE           -D_X86_64_  -DUNIX  -Dlinux     -O3 -g   -Wall      -mtune=generic     -m64 -fPIC 
-I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include       
-I/corvette/usr/local/epics-devel/base-7.0.1/include/compiler/gcc -I/corvette/usr/local/epics-devel/base-7.0.1/include/os/Linux -I/corvette/usr/local/epics-devel/base-7.0.1/include      
-I/corvette/home/epics/devel-base7/asyn-4-32/include  
-I/corvette/home/epics/devel-base7/areaDetector-3-1/ADSupport/include/os/Linux 
-I/corvette/home/epics/devel-base7/areaDetector-3-1/ADSupport/include   
-I/corvette/home/epics/devel-base7/areaDetector-3-1/ADCore/include                    -c ../ntndArrayConverter.cpp
../ntndArrayConverter.cpp: In member function 'void NTNDArrayConverter::fromUndefinedAttribute(epics::pvData::PVStructurePtr)':
../ntndArrayConverter.cpp:497:54: error: passing 'const PVFieldPtr {aka const std::tr1::shared_ptr<epics::pvData::PVField>}' 
as 'this' argument of 'void std::tr1::__shared_ptr<_Tp, _Lp>::reset() [with _Tp = epics::pvData::PVField; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2u]' discards qualifiers [-fpermissive]
     dest->getSubField<PVUnion>("value")->get().reset();
                                                      ^
make[1]: *** [ntndArrayConverter.o] Error 1
```

The error when building on Windows is:
```
ntndArrayConverter.cpp
H:/epics-devel/base-7.0.1/include\pv/pvData.h(534) : warning C4800: 'char' : forcing value to bool 'true' or 'false' (performance warning)
H:/epics-devel/base-7.0.1/include\pv/pvData.h(583) : warning C4250: 'epics::pvData::PVString' : inherits 'epics::pvData::PVScalarValue<T>::epics::pvData::PVScalarValue<T>::deserial
ize' via dominance
        with
        [
            T=std::string
        ]
        H:/epics-devel/base-7.0.1/include\pv/pvData.h(488) : see declaration of 'epics::pvData::PVScalarValue<T>::deserialize'
        with
        [
            T=std::string
        ]
../ntndArrayConverter.cpp(497) : error C2662: 'void std::tr1::shared_ptr<_Ty>::reset(void)' : cannot convert 'this' pointer from 'const epics::pvData::PVFieldPtr' to 'std::tr1::shared_ptr<_Ty> &'
        with
        [
            _Ty=epics::pvData::PVField
        ]
        Conversion loses qualifiers
H:/epics-devel/base-7.0.1/configure/RULES_BUILD:237: recipe for target 'ntndArrayConverter.obj' failed
make[5]: *** [ntndArrayConverter.obj] Error 2
H:/epics-devel/base-7.0.1/configure/RULES_ARCHS:58: recipe for target 'install.windows-x64-static' failed
make[4]: *** [install.windows-x64-static] Error 2
```
I suspect the problem is that a <static_cast> is needed, but I am not sure how to do it.

mdavidsaver, 2017-12-05:
I'm confused, is this with a branch?

https://github.com/areaDetector/ADCore/blob/b305f875c1efc7feb3e267ed9f07630e8272e636/ADApp/ntndArrayConverterSrc/ntndArrayConverter.cpp#L497

With "PVStructurePtr dest" should work.  However, the error looks like what I'd expect if the argument were "const PVStructurePtr dest".

MarkRivers, 2017-12-05:
The problem occurs on the master branch, with the code in your comment.  This built fine with EPICS-CPP-4.6.0.

brunoseivam, 2017-12-05:
[This](https://github.com/epics-base/pvDataCPP/commit/9b1e789e62b9b4b03d3f56143bf3656ecfe1a375) is the commit that has the likely culprit, `PVUnion::get()` in `pvData.h` was changed to return a `const` pointer.

MarkRivers, 2017-12-05:
How do we change ntndArrayConverter.cpp to accommodate this change to pvData.h?

brunoseivam, 2017-12-05:
The intent here was to have an union with no value selected. 

NDArray's NDAttribute has a `NDAttrUndefined` choice, but NTNDArray's NTNDArrayAttribute behaves more like an Option type (as far as I understand) that has either a PVField with a value or no PVField. 

Maybe something like this would work:

``` c++
    PVFieldPtr nullPtr;
    dest->getSubField<PVUnion>("value")->set(nullPtr);
```

Perhaps Michael has a better suggestion

MarkRivers, 2017-12-05:
That suggestion from @brunoseivam worked, and ADCore now compiles OK.

I now have a problem when linking the IOC applications:
```
make[1]: Entering directory `/home/epics/devel-base7/areaDetector-3-1/ADSimDetector/iocs/simDetectorIOC/simDetectorApp/src/O.linux-x86_64'
/usr/bin/g++ -o simDetectorApp -Wl,-Bstatic 
-L/home/epics/devel-base7/areaDetector-3-1/ADSimDetector/lib/linux-x86_64 
-L/home/epics/devel-base7/areaDetector-3-1/ADSimDetector/iocs/simDetectorIOC/lib/linux-x86_64 
-L/home/epics/devel-base7/areaDetector-3-1/ADCore/lib/linux-x86_64 
-L/home/epics/devel-base7/areaDetector-3-1/ADSupport/lib/linux-x86_64 
-L/home/epics/devel-base7/asyn-4-32/lib/linux-x86_64 
-L/home/epics/devel-base7/autosave-5-8/lib/linux-x86_64 
-L/home/epics/devel-base7/busy-1-7/lib/linux-x86_64 
-L/home/epics/devel-base7/calc-3-6-1/lib/linux-x86_64 
-L/home/epics/devel-base7/devIocStats-3-1-15/lib/linux-x86_64 
-L/home/epics/devel-base7/seq-2-2-4/lib/linux-x86_64 
-L/home/epics/devel-base7/sscan-2-10-2/lib/linux-x86_64 
-L/usr/local/epics-devel/base-7.0.1/lib/linux-x86_64            
-rdynamic -m64             simDetectorApp_registerRecordDeviceDriver.o simDetectorAppMain.o   
-lsimDetector -lNDPlugin -lADBase -lntndArrayConverter -lnt -lpvDatabase -lpvAccess -lpvData 
-lnetCDF -lMagick++ -lcoders -lMagick -ljbig -ljp2 -lbzlib -lpng -lwebp -llcms -lttf -lwmf 
-lfilters -ltiff -ljpeg -lxml2 -lNeXus -lhdf5 -lblosc -lszip -lzlib -lasyn -lautosave -lbusy -lcalc 
-ldevIocStats -lsscan -lseq -lpv -ldbRecStd -ldbCore -lca -lCom -Wl,-Bdynamic -lX11 -lXext -lpthread   
-lreadline -lm -lrt -ldl -lgcc
simDetectorApp_registerRecordDeviceDriver.o: In function `simDetectorApp_registerRecordDeviceDriver':
/home/epics/devel-base7/areaDetector-3-1/ADSimDetector/iocs/simDetectorIOC/simDetectorApp/src/O.linux-x86_64/simDetectorApp_registerRecordDeviceDriver.cpp:504: undefined reference to `pvar_func_refTrackRegistrar'
/home/epics/devel-base7/areaDetector-3-1/ADSimDetector/iocs/simDetectorIOC/simDetectorApp/src/O.linux-x86_64/simDetectorApp_registerRecordDeviceDriver.cpp:505: undefined reference to `pvar_func_registerStartPVAServer'
collect2: error: ld returned 1 exit status
make[1]: *** [simDetectorApp] Error 1
make[1]: Leaving directory `/home/epics/devel-base7/areaDetector-3-1/ADSimDetector/iocs/simDetectorIOC/simDetectorApp/src/O.linux-x86_64'
make: *** [install.linux-x86_64] Error 2
```
Is this a problem with the order of link libraries?  

When using EPICS 7 does one still need to link explicitly with pvAccess, pvDatabase, etc. or are those included in EPICS_BASE_IOC_LIBS or some other symbol?

brunoseivam, 2017-12-05:
You also have to link pvAccessIOC I believe

MarkRivers, 2017-12-05:
That fixed that problem.  I am now able to build all of areaDetector except pvaDriver.  I get this error:
```
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE           -D_X86_64_  -DUNIX  -Dlinux     -O3 -g   -Wall      -mtune=generic     -m64 -fPIC -I. -I../O.Common 
-I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include       
-I/corvette/usr/local/epics-devel/base-7.0.1/include/compiler/gcc -I/corvette/usr/local/epics-devel/base-7.0.1/include/os/Linux 
-I/corvette/usr/local/epics-devel/base-7.0.1/include      
-I/corvette/home/epics/devel-base7/asyn-4-32/include  
-I/corvette/home/epics/devel-base7/areaDetector-3-1/ADSupport/include/os/Linux 
-I/corvette/home/epics/devel-base7/areaDetector-3-1/ADSupport/include   
-I/corvette/home/epics/devel-base7/areaDetector-3-1/ADCore/include                    
-c ../pvaDriver.cpp
../pvaDriver.cpp: In constructor 'pvaDriver::pvaDriver(const char*, const char*, int, size_t, int, int)':
../pvaDriver.cpp:109:16: error: call to 'epics::pvAccess::getChannelProviderRegistry' declared with attribute error: ChannelProvider shared_ptr ownership rules have changed.
     m_provider = getChannelProviderRegistry()->getProvider("pva");
                ^
../pvaDriver.cpp: In constructor 'pvaDriver::pvaDriver(const char*, const char*, int, size_t, int, int)':
../pvaDriver.cpp:109:16: error: call to 'epics::pvAccess::getChannelProviderRegistry' declared with attribute error: ChannelProvider shared_ptr ownership rules have changed.
     m_provider = getChannelProviderRegistry()->getProvider("pva");
                ^
make[3]: *** [pvaDriver.o] Error 1
make[3]: Leaving directory `/home/epics/devel-base7/areaDetector-3-1/pvaDriver/pvaDriverApp/src/O.linux-x86_64'
make[2]: *** [install.linux-x86_64] Error 2
make[2]: Leaving directory `/home/epics/devel-base7/areaDetector-3-1/pvaDriver/pvaDriverApp/src'
make[1]: *** [src.install] Error 2
make[1]: Leaving directory `/home/epics/devel-base7/areaDetector-3-1/pvaDriver/pvaDriverApp'
make: *** [pvaDriverApp.install] Error 2
```
How does this code need to be changed to work with EPICS 7.x?

brunoseivam, 2017-12-05:
From the [pvAccess RELEASE notes](http://www.aps.anl.gov/epics/base/R7-0/1-docs/pvAccess_RELEASE.html):

> Removed access to singleton registry via getChannelProviderRegistry() and registerChannelProviderFactory() in favor of epics::pvAccess::ChannelProviderRegistry::clients() and epics::pvAccess::ChannelProviderRegistry::servers(). The "pva" and "ca" providers are registered with the clients() singleton. epics::pvAccess::ServerContext() looks up names with the servers() singleton.

So I believe the failing line should be replaced with:

``` c
m_provider = ChannelProviderRegistry::clients()->getProvider("pva");
```

brunoseivam, 2017-12-06:
I think it is worth noting that this is a breaking change, it will make pvaDriver rely on pvAccess >= 6.0

MarkRivers, 2017-12-06:
I have made this change be conditional on EPICS_PVA_MAJOR_VERSION >=6.  I have pushed this change to the master branch on Github.  I have also pushed the changed in ADCore ntndArrayConverter.cpp and a minor change to ADPointGrey. 

It all compiles both with base 7.0.1 and with base 3.15.5 with EPICS-CPP-4.6.0.

MarkRivers, 2017-12-07:
These issues are now fixed.  If @mdavidsaver can look at the change to ntnArrayConverter.cpp in 7dc961a60f07ea84b9534e085e1e727b356e0dce that would be good, but I am closing this issue for now.

closed 2017-12-07 by MarkRivers

mdavidsaver, 2017-12-07:
I'll take a look at this and comment when I have time.  This is a busy week for me.

zthurman, 2023-07-07:
```
ned_registerRecordDeviceDriver.cpp: undefined reference to `pvar_func_refTrackRegistrar'
/usr/bin/ld: ned_registerRecordDeviceDriver.cpp: undefined reference to `pvar_func_registerStartPVAClient'
/usr/bin/ld: ned_registerRecordDeviceDriver.cpp: undefined reference to `pvar_func_registerStartPVAServer'
```

When building nED with EPICS7 today. The internet brought me here, linking pvAccessIOC before qsrv got me past it. 

I know this isn't really the right channel but I just wanted to thank you guys, you're awesome.