areaDetector/ADCore #304: check use of the `when` attribute in NDAttributes.xsd

prjemian, 2017-12-19:
@MarkRivers points out that the `when` attribute is not *supported in the XML files used to define [NDAttributes](https://github.com/areaDetector/ADCore/blob/master/XML_schema/NDAttributes.xsd#L132).  That file is only parsed by `ADCore/ADApp/ADSrc/asynNDArrayDriver.cpp`, and it does not look for a "when" attribute.  There is no property of the NDAttribute class that corresponds to the "when" attribute.  That attribute does not really make sense in the context of an NDAttribute, because it is only related to file plugins.*

*Rather, the XML file in which the `when` attribute is supported is the [`hdf5_xml_layout_schema.xsd`](https://github.com/areaDetector/ADCore/blob/master/XML_schema/hdf5_xml_layout_schema.xsd#L30), i.e. the file which controls the layout of the XML file.*

prjemian, 2017-12-19:
Should check this with attributes files such as in https://github.com/areaDetector/ADSimDetector/tree/master/iocs/simDetectorIOC/iocBoot/iocSimDetector *before* declaring this issue resolved.

prjemian, 2017-12-19:
Ran this script with the newly revised Schema:

    for f in `locate areaDetectorGIT/ | grep -i attributes | grep xml`; do 
        xmllint --noout --schema NDAttributes.xsd $f
    done

and obtained this result:

```
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADExample/iocs/simDetectorIOC/iocBoot/iocHDF5Test/test1_attributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADExample/iocs/simDetectorIOC/iocBoot/iocHDF5Test/test2_attributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADExample/iocs/simDetectorIOC/iocBoot/iocHDF5Test/test3_attributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADExample/iocs/simDetectorIOC/iocBoot/iocHDF5Test/test4_attributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADExample/iocs/simDetectorIOC/iocBoot/iocHDF5Test/test5_attributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADExample/iocs/simDetectorIOC/iocBoot/iocSimDetector/ROIAttributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADExample/iocs/simDetectorIOC/iocBoot/iocSimDetector/ROIStatAttributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADExample/iocs/simDetectorIOC/iocBoot/iocSimDetector/StatsAttributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADExample/iocs/simDetectorIOC/iocBoot/iocSimDetector/netCDFAttributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADExample/iocs/simDetectorIOC/iocBoot/iocSimDetector/simDetectorAttributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADPICam/iocs/PICamIOC/iocBoot/iocPICam/PerkinElmerAttributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADPerkinElmer/iocs/perkinElmerIOC/iocBoot/iocPerkinElmer/PerkinElmerAttributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADPilatus/iocs/pilatusIOC/iocBoot/iocPilatus/pilatusAttributes.xml validates
/usr/local/epics/synApps_5_8/support/areaDetectorGIT/ADProsilica/iocs/prosilicaIOC/iocBoot/iocProsilica/prosilicaAttributes.xml validates
```

Using ADCore commit 54569acaf042ba33476eb6de64fe203d4bf4e0d6 and related modules from Nov 22.  Looks like AD 3.1.

MarkRivers, 2017-12-20:
Closed via #305 .

closed 2017-12-20 by MarkRivers