areaDetector/ADCore #307: PV to indicate when all arrays are done processing

tacaswell, 2018-01-12:
Discussed this with @MarkRivers at BNL today.

One problem with non-blocking callbacks is that when the acquire bit flips to low the arrays have not in general been processed by all of the plugins in the processing chain.  Knowing when the arrays have been fully-processed is important for higher-level sequencing.

It would be very useful to have a PV on the cam indicating when all of the arrays it has emitted into the processing pipeline have either been fully processed or dropped.

MarkRivers, 2018-01-12:
I've thought about this a bit.  It is not as easy as we first thought, where an additional counter would be added to the NDArray, indicating how many plugins were processing it, and when it got to 0 a PluginsDone record could process.  The problem is that plugins may not be processing the original array, but rather a copy of that array, which would have its own counter.

One solution I am thinking of is to combine this issue with the desire to move to a single NDArrayPool per driver, and plugins allocate from that same pool.  This solves 2 problems:
  - A single pool enables effective limits on memory usage.  Currently with multiple pools it is difficult to really limit the memory use because it is the total use that matters but there is no way to predict how much memory each pool will need.
- The single pool would be the object that could keep track of how many plugins are currently active for that driver, because it is the object they all have in common.  It is also associated with the driver, so the PluginsBusy PV will belong to that driver.  If there are multiple asynNDArray drivers in an IOC then each will have its own pool and its own PluginsBusy PV.

brunoseivam, 2018-01-12:
If I understand correctly, the problem is that in the non-blocking mode the _driver_ might be done producing arrays while the _plugin_ of interest is still processing the data. 

Wouldn't the following workaround work? At the beginning of an acquisition you can zero the plugin's `ArrayCounter` and `DroppedArrays` and then wait for `(driver's) Acquire == 0 && ArrayCounter_RBV + DroppedArrays_RBV == N`. This could even be done in a calc record.

mp49, 2018-01-12:
@brunoseivam That's how I've worked around this issue in the past. For example, if I have an ROI plugin that is read as soon as the acquire is completed by the camera driver it will almost always be an NDArray behind. But Marks proposal seems like a useful feature and a more generic solution.

tacaswell, 2018-01-12:
Watching the number of images on the plugins requires knowing which plugins to watch.  I would like to generically know when _all_ plugins are done without having to know anything about the topology of the processing chain at the next control layer up.

MarkRivers, 2018-05-20:
> One solution I am thinking of is to combine this issue with the desire to move to a single NDArrayPool per driver, and plugins allocate from that same pool. This solves 2 problems:
>
> A single pool [...]
The single pool would be the object that could keep track of how many plugins are currently active for that driver, because it is the object they all have in common. It is also associated with the driver, so the PluginsBusy PV will belong to that driver. If there are multiple asynNDArray drivers in an IOC then each will have its own pool and its own PluginsBusy PV.

I'm thinking about how to implement this.  The first thing would be the change the constructor for NDArrayPool from this:
```
NDArrayPool::NDArrayPool(int maxBuffers, size_t maxMemory)
  : maxBuffers_(maxBuffers), numBuffers_(0), maxMemory_(maxMemory), memorySize_(0), numFree_(0)
```

to this:

```
NDArrayPool::NDArrayPool(bool isPlugin, asynNDArrayDriver *pDriver, size_t maxMemory)
  : pDriver(0), maxMemory_(maxMemory), memorySize_(0), numFree_(0)

    if (!isPlugin) pDriver_ = pDriver;
```
Add 2 new methods to NDArrayPool:
```
   void incrementPluginCounter();
   decrementPluginCounter();
```

This does several things:
- Removes the maxBuffers limit in NDArrayPool.  This limited the total number of NDArrays.  I don't think this is useful, we just want to limit the total amount of memory.
- Plugins no longer have an NDArrayPool, only drivers do.  This  makes it much more useful because there will only be a single pool per driver/plugin chain, so effective memory allocation checking can actually be done.   Currently with multiple pools it is hard to enforce memory limits.
- We now have methods to count the number of active plugins associated with a single NDArrayPool, i.e. a single driver.  This will address the problem originally posed in this issue.

This will require minor changes to all drivers and plugins, so would require a major release (e.g. R4-0).

MarkRivers, 2018-05-21:
I now have a working implementation of this.  It is in the plugin_count branch of ADCore.  There are the major changes in this branch:

- Only drivers now have an NDArrayPool, plugins do not.  Plugins use the pointer passed to them in pArray->pNDArrayPool so that they allocate NDArrays from the driver's NDArrayPool, not from a private one.  This makes it possible to enforce memory limits for NDArrays (maxMemory in driver constructor).
- The maxBuffers argument to the driver is now ignored.  There is no limit on the number of NDArrays, only on the total amount of memory.
- The maxBuffers argument to ADDriver::ADDriver is still present so existing drivers will work with no changes.  It is ignored.  A second constructor will be added in the future and the old one will be deprecated.
- The maxBuffers and maxMemory arguments to NDPluginDriver::NDPluginDriver are now ignored.  They are still present so existing plugins will work with no changes.  A second constructor will be added in the future and the old one will be deprecated.
- The NDPluginDriver base class now increments a NumActivePlugins counter in the driver that owns each NDArray as it is queued, and decrements the counter after the processing is done.  There is thus a single NumActivePlugins PV for each driver that can be used to determine when plugins are done processing.
- The ADBuffers sub-screen now contains the NumActivePlugins counter PV.
- The ADBase screen now contains the ADBuffers screen, so it shows this PV.
[image]
- The simDetector screen now contains the ADBuffers screen, so it shows this PV.
[image]

This is the output of camonitor when the simDetector acquires 20 images.  It shows both the state of the Acquire PV and the NumActivePlugins PV:
```
corvette:simDetectorApp/op/adl>camonitor 13SIM1:cam1:Acquire 13SIM1:cam1:NumActivePlugins
13SIM1:cam1:Acquire            2018-05-21 17:17:14.348695 Done
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:14.405151 0
13SIM1:cam1:Acquire            2018-05-21 17:17:21.081349 Acquire
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.092176 1
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.092183 2
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.092186 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.092279 2
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.093834 1
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.102960 2
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.102966 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.102993 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.103049 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.104512 2
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.113527 1
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.113668 2
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.113675 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.113700 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.113757 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.115163 2
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.124374 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.124380 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.124389 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.124470 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.125758 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.133302 2
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.135146 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.135152 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.135165 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.135239 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.136536 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.145863 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.145867 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.145875 6
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.145991 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.147502 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.151028 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.156637 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.156643 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.156652 6
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.156780 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.158283 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.167105 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.167373 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.167380 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.167396 6
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.167468 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.168815 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.178073 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.178077 6
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.178083 7
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.178141 6
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.179487 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.182071 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.188768 5
[... 47 lines cut ...]
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.274721 7
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.274750 8
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.274794 7
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.275887 6
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.277872 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.285482 6
[... cut ...]
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.297467 6
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.300037 5
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.310661 4
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.321579 3
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.332656 2
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.343830 1
13SIM1:cam1:NumActivePlugins   2018-05-21 17:17:21.355255 0
```
Note that Acquire goes done about 60 ms before the plugins finish processing.

MarkRivers, 2018-05-21:
The unit tests in pluginTests still need some work.  They are failing because of the changes above.

MarkRivers, 2018-05-22:
Added a mutex to protect the NumActivePlugins parameter in asynNDArrayDriver, because it can be modified by multiple plugin threads.  It now appears to always go back to 0 when plugins finish, which was not previously the case if many plugins were processing quickly.

MarkRivers, 2018-05-22:
I have added some new records to ADBase.template.

- WaitForPlugins  This record determines whether AcquireBusy waits for NumActivePlugins to go to 0 before changing to 0 when acquisition completes.
- AcquireBusy  This is a busy record that is set to 1 when Acquire changes to 1.  It changes back to 0 when acquisition completes, i.e. when Acquire_RBV=0.  If WaitForPlugins is Yes then it also waits for NumActivePlugins to go to 0 before changing to 0.

I moved the NumActivePlugins from the ADBuffers sub-screen to the ADCollect sub-screen.  This is a screen shot of simDetector.adl when Acquire has gone back to 0 (Done) but AcquireBusy is still 1 (Acquiring) because the plugins have not finished processing (133 arrays still in queues).
[image]

These changes mean that ca_put_callback to the Acquire PV will now wait for the plugins to complete if WaitForPlugins is Yes.  This is very nice, it means a client does not need need to look at another PV to see if the plugins are done.

This is the output of time and caput command that waits for completion when WaitForPlugins=No. Note that the elapsed time is 1.04 seconds.
```
corvette:ADCore/ADApp/op>time caput -c -w 10 -l 13SIM1:cam1:Acquire 1
Old : 13SIM1:cam1:Acquire            2018-05-22 17:51:09.262987 Done
New : 13SIM1:cam1:Acquire            2018-05-22 17:52:44.490491 Done
0.004u 0.006s 0:01.04 0.0%      0+0k 0+0io 0pf+0w
```
This is the output of camonitor when that command was executing:
```
corvette:ADApp/op/adl>camonitor 13SIM1:cam1:Acquire 13SIM1:cam1:WaitForPlugins 
...
13SIM1:cam1:WaitForPlugins     2018-05-22 17:52:41.365223 No
13SIM1:cam1:Acquire            2018-05-22 17:52:43.480759 Acquire
13SIM1:cam1:AcquireBusy        2018-05-22 17:52:43.480774 Acquiring STATE MINOR
13SIM1:cam1:Acquire            2018-05-22 17:52:44.490491 Done
13SIM1:cam1:AcquireBusy        2018-05-22 17:52:44.490516 Done
```
This is the output of time and caput command that waits for completion when WaitForPlugins=Yes. Note that the elapsed time is 2.29 seconds.
```
corvette:ADCore/ADApp/op>time caput -c -w 10 -l 13SIM1:cam1:Acquire 1
Old : 13SIM1:cam1:Acquire            2018-05-22 17:52:44.490491 Done
New : 13SIM1:cam1:Acquire            2018-05-22 17:54:45.390593 Done
0.004u 0.008s 0:02.29 0.0%      0+0k 0+0io 0pf+0w
```
This is the output of camonitor when that command was executing:
```
corvette:ADApp/op/adl>camonitor 13SIM1:cam1:Acquire 13SIM1:cam1:WaitForPlugins 
...

13SIM1:cam1:WaitForPlugins     2018-05-22 17:54:42.448385 Yes
13SIM1:cam1:Acquire            2018-05-22 17:54:44.380873 Acquire
13SIM1:cam1:AcquireBusy        2018-05-22 17:54:44.380886 Acquiring STATE MINOR
13SIM1:cam1:Acquire            2018-05-22 17:54:45.390593 Done
13SIM1:cam1:AcquireBusy        2018-05-22 17:54:46.636855 Done
```
Note that Acquire went to Done about 1.2 second before AcquireBusy changed to Done, because AcquireBusy is waiting for the plugins to complete.

tacaswell, 2018-05-23:
Thanks @MarkRivers !  This looks like it will make our lives _much_ easier on the ophyd/bluesky side of things!

MarkRivers, 2018-06-25:
This issue is resolved with #328.

closed 2018-06-25 by MarkRivers