areaDetector/ADCore #309: consider using .gitattributes per repo

prjemian, 2018-01-16:
In your [review of #308](https://github.com/areaDetector/ADCore/pull/308#pullrequestreview-88848411) and in other pull requests, you make note of certain style guidelines such as indentation preference.  Is it possible for you to enforce some of these more easily using a [`.gitattributes` file](https://help.github.com/articles/dealing-with-line-endings/#per-repository-settings) (such as this example from [drupal](https://github.com/drupal/drupal/blob/8.5.x/.gitattributes))?

t-b, 2018-01-16:
@prjemian Enforcing them via `.gitattributes` only works in combination with having some CI service checking each PR. If you want to go into that direction I would instead add a `.clang-format` file, relayout all source code, and require that all PRs follow that.

MarkRivers, 2018-01-20:
I don't think this is worth the extra hassle.  The only real style aspects that I ask people to observe are no tabs in source files (.h, .cpp, .template, etc.) and to respect the existing indentation in the file (typically 2 or 4 spaces).

No tabs in C++ files is recommended by all style guides I have read because it makes the code legible in any editor.
https://google.github.io/styleguide/cppguide.html
http://archive.oreilly.com/oreillyschool/courses/cplusplus1/style.html
https://developer.mozilla.org/en-US/docs/Mozilla/Developer_guide/Coding_Style

closed 2018-01-20 by MarkRivers