areaDetector/ADCore #329: Revisiting name 'NumActivePlugins'

danielballan, 2018-06-19:
During a presentation at the recent EPICS meeting in Chicago, @MarkRivers covered the changes in #323 and mentioned that 'NumActivePlugins' counts the number of queued NDArrays, it is says in the issue title of #323, not the number of plugins that are active. Is it worth renaming this before R3-3 is tagged?

I'm happy to put up a PR making the change if people support it.

tacaswell, 2018-06-19:
I am :+1: on this.

MarkRivers, 2018-06-19:
I think this is worth doing.  I'll work on it soon.

MarkRivers, 2018-06-20:
I have renamed NumActivePlugins to NumQueuedArrays.  This has been done in asynNDArrayDriver.h and .cpp, ADBase.template, NDArrayBase.template, and ADCollect.adl.  

The increment/decrementPluginCount methods have been renamed to increment/decrementQueueArrayCount.  

ADSimDetector.adl and pvaDriver.adl have been changed, and all of the op/*/autoconvert files have been changed.

These changes have been merged into master and pushed to Github.

closed 2018-06-20 by MarkRivers

tacaswell, 2018-06-20:
Thanks @MarkRivers !