areaDetector/ADCore #337: "SDS" is an invalid NX_class

prjemian, 2018-07-24:
The default HDF5 XML layout specifies a NeXus structure that is correct except for the use of `SDS` as the value for an [instance of the `NX_class` attribute](https://github.com/areaDetector/ADCore/blob/master/ADApp/pluginSrc/NDFileHDF5LayoutXML.cpp#L58).

Valid NeXus [class names](http://download.nexusformat.org/doc/html/datarules.html?highlight=nx_class#id2) follow this [regexp](http://download.nexusformat.org/doc/html/datarules.html?highlight=nx_class#nexus-data-types): `^NX[A-Za-z0-9_]*$`
and take one of the names of the existing [NeXus base classes](http://download.nexusformat.org/doc/html/classes/base_classes/index.html#base-class-definitions).

The current use of `SDS` causes the HDF5 files produced by the default layout to fail [NeXus validation programs](http://download.nexusformat.org/doc/html/utilities.html#validation) such as [`punx`](http://punx.readthedocs.io/en/latest/).  This can be fixed with a `custom_layout.xml` file.  It would be simple to fix the default case provided here.

Also, this should be changed in the example HDF5 layout file in [ADSimDetector](https://github.com/areaDetector/ADSimDetector/blob/master/iocs/simDetectorIOC/iocBoot/iocSimDetector/hdf5_layout_nexus.xml#L13).

prjemian, 2018-07-24:
Removing the `NX_class` attribute from the image dataset should have zero adverse consequences.  No NeXus software uses or expects to find this attribute on any dataset.

MarkRivers, 2019-04-01:
@prjemian @ajgdls @ulrikpedersen should I make this change?  Any adverse consequences expected?

prjemian, 2019-04-01:
As [stated previously](https://github.com/areaDetector/ADCore/issues/337#issuecomment-407445694), zero adverse consequences for NeXus.  I am not aware of any software that would expect this invalid NeXus class name.  It should be corrected either by renaming the `NX_class` attribute or removing the attribute altogether.  For NeXus, a dataset does not need this attribute.

ulrikpedersen, 2019-04-02:
> The current use of SDS causes the HDF5 files produced by the default layout to fail NeXus validation programs such as punx. 

I'm surprised at that because my understaning of NeXus is that it specifically allows for non-NeXus groups, datasets and attributes to exist in the HDF5 file... (but my understanding of NeXus is definitely patchy!)

Our GDA application used to depend on SDS=1 and there are apparently still some references to it in the code-base but it's difficult to say if those particular codes are still used. 

As NeXus validation scripts **are** failing, I vote in favour of removing the SDS=1. :+1:  We will just have to untangle our own legacy codes here...

prjemian, 2019-04-02:
> the SDS=1. 👍 We will just have to untangle our own legacy codes here...

You have not read this issue correctly. The problem attribute is

    NX_class=SDS

NeXus has a strict list of allowed values for the NX_class attribute. The
value SDS is not in that list. It's easy to see this is an error since all
such values must begin with the letters NX shown in the regexp.

This rule has existed as long as I have been on the NIAC (2006).

BTW, NeXus does not define the attribute you mentoned

    SDS=1

which means it can appear anywhere in a NeXus data file.

Pete

ulrikpedersen, 2019-04-02:
> You have not read this issue correctly. The problem attribute is NX_class=SDS

Ah, I see. I thought I had the recollection that it was an HDF5 attribute SDS=1 and didn't think to check the code.

Let's get rid of it!

MarkRivers, 2019-04-02:
I have removed this attribute.

closed 2019-04-02 by MarkRivers