areaDetector/ADCore #338: Add FFTTimePerPoint as NDAttribute to NDPluginFFT

BenBradnick, 2018-07-27:
I think it would be useful to add the time per point parameter passed as an NDAttribute so that the frequency axis can be correctly scaled when the FFT NDArray is saved to disk using a file plugin.

MarkRivers, 2018-07-27:
I don't think I understand this issue.  It seems to me that you can do this today using an XML file to specify that the time-per-point parameter should be added to the NDArray as an NDAttribute and saved to disk with a file plugin.

I just tested this by adding this FFTAttributes.xml to the areaDetector/ADCSimDetector/iocs/ADCSimDetectorIOC/iocBoot/iocADCSimDetector directory.
```
corvette:ADCSimDetectorIOC/iocBoot/iocADCSimDetector>more FFTAttributes.xml
<?xml version="1.0" standalone="no" ?>
<!-- Attributes -->
<Attributes
    xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
    xsi:noNamespaceSchemaLocation="../../../../../ADCore/XML_schema/NDAttributes.xsd"
    >
    <Attribute name="TimePerPoint"        type="PARAM"    source="FFT_TIME_PER_POINT"        datatype="DOUBLE"        description="FFT time per point"/>
</Attributes>
```
I then read that XML attributes file into the FFT2 plugin using the NDPluginBaseFull screen:
[image]

Running the `asynReport 20 FFT2` command shows that the attribute list does now contain this new attribute with the correct value:

```
epics> asynReport 20 FFT2
...
FFT2: pAttributeList report

NDAttributeList: address=0x376cc30:
  number of attributes=1

NDAttribute, address=0x7f7e8c03a340:
  name=TimePerPoint
  description=FFT time per point
  source type=1
  source type string=NDAttrSourceParam
  source=FFT_TIME_PER_POINT
  dataType=NDAttrFloat64
  value=0.005000
  paramAttribute
    Addr=0
    Param type=1
    Param ID=71
```
I then saved the NDArrays from the FFT2 plugin into netCDF files with the NDFileNetCDF plugin.

Here is the output of ncdump on one of those files when the time per point was changed varied from .005 to .001.
```
corvette:~/scratch>ncdump -v Attr_TimePerPoint fft_test_006.nc
netcdf fft_test_006 {
dimensions:
        numArrays = UNLIMITED ; // (24 currently)
        dim0 = 1024 ;
        attrStringSize = 256 ;
variables:
        int uniqueId(numArrays) ;
        double timeStamp(numArrays) ;
        int epicsTSSec(numArrays) ;
        int epicsTSNsec(numArrays) ;
        double array_data(numArrays, dim0) ;
        double Attr_TimePerPoint(numArrays) ;

// global attributes:
                :dataType = 7 ;
                :NDNetCDFFileVersion = 3. ;
                :numArrayDims = 1 ;
                :dimSize = 1024 ;
                :dimOffset = 0 ;
                :dimBinning = 1 ;
                :dimReverse = 0 ;
                :Attr_TimePerPoint_DataType = "Float64" ;
                :Attr_TimePerPoint_Description = "FFT time per point" ;
                :Attr_TimePerPoint_Source = "FFT_TIME_PER_POINT" ;
                :Attr_TimePerPoint_SourceType = "NDAttrSourceParam" ;
data:

 Attr_TimePerPoint = 0.005, 0.005, 0.005, 0.005, 0.005, 0.004, 0.004, 0.004,
    0.004, 0.004, 0.004, 0.004, 0.003, 0.003, 0.003, 0.003, 0.003, 0.003,
    0.002, 0.002, 0.002, 0.001, 0.001, 0.001 ;
```
So it seems like you can do what you are asking today with no change to the FFT plugin.

BenBradnick, 2018-07-30:
Hi Mark,

Apologies, I will close this ticket - I was unaware of this feature as it is hidden on Diamond's set of plugin screens and is configured in our IOC build system instead.

Thank you for the prompt response, this is perfect.

closed 2018-07-30 by BenBradnick