areaDetector/ADCore #343: Hardcoded ":" in request file

hinxx, 2018-09-21:
The only place that hardcodes the ":" delimiter is NDStats_settings.req. Our site has special rules for ":" and therefore I need to change the stock ADCore file to overcome this (currently replacing with "-"):

https://github.com/areaDetector/ADCore/blob/5122f460705b57827e61da9136ba2867ab9673ad/ADApp/Db/NDStats_settings.req#L12

We could remove the offending line from the NDStats_settings.req; let use specify it in its auto_settings.req, or we could introduce a new macro for the "$(R)TS:" and let user set it in the caller.

MarkRivers, 2018-09-21:
These are the template files in ADCore that hardcode ":".  
```
corvette:ADCore/ADApp/Db>grep : *.template | grep -v #
NDAttrPlotData.template:record(waveform, "$(P)$(R)$(AXIS):Data$(DATA_IND)") {
NDAttrPlotData.template:record(stringin, "$(P)$(R)$(AXIS):DataLabel$(DATA_IND)") {
NDAttrPlotData.template:record(ao, "$(P)$(R)$(AXIS):DataSelect$(DATA_IND)") {
NDAttrPlotData.template:record(ai, "$(P)$(R)$(AXIS):DataSelect$(DATA_IND)_RBV") {
NDProcess.template:    field(DOL2, "$(P)$(R)TIFF:PortName_RBV")
NDProcess.template:    field(LNK3, "$(P)$(R)TIFF:ArrayCallbacks PP MS")
NDProcess.template:    field(LNK4, "$(P)$(R)TIFF:ReadFile PP MS")
NDProcess.template:    field(DOL2, "$(P)$(R)TIFF:PortName_RBV")
NDProcess.template:    field(LNK3, "$(P)$(R)TIFF:ArrayCallbacks PP MS")
NDProcess.template:    field(LNK4, "$(P)$(R)TIFF:ReadFile PP MS")
NDROIStat8.template:substitute  "ROI=1:"
NDROIStat8.template:substitute  "ROI=2:"
NDROIStat8.template:substitute  "ROI=3:"
NDROIStat8.template:substitute  "ROI=4:"
NDROIStat8.template:substitute  "ROI=5:"
NDROIStat8.template:substitute  "ROI=6:"
NDROIStat8.template:substitute  "ROI=7:"
NDROIStat8.template:substitute  "ROI=8:"
```
I have removed the search results where the ":" was not part of the record name, i.e. info tags, etc.

These are the .req files that hardcode a ":"
```
corvette:ADCore/ADApp/Db>grep : *.req | grep -v #
NDStats_settings.req:file "NDTimeSeries_settings.req", P=$(P), R=$(R)TS:
```
Presumably you would also like the .template files to be changed?

hinxx, 2018-09-24:
I missed the templates that you found.

If it is too much of a hassle to deal with at this moment, I can rely on my local changes to get the job done and this can wait until better time.

FYI, I have changed the NDStats_settings.req to hold:

```file "NDTimeSeries_settings.req", P=$(P), R=$(S)```

and my auto_settings.req has:

```file "NDStats_settings.req",         P=$(P),  R=CH$(N)-ST-,      S=CH$(N)-TS-```