areaDetector/ADCore #349: HDF5 Plugin to support compressed NDArrays

ulrikpedersen, 2018-11-16:
With the recent work from @brunoseivam on #332 it is now possible to compress NDArrays with a range of compression algorithms. The HDF5 file writer should be adapted to support this. This can be done using the HDF5 ["Direct Chunk Write"](https://confluence.hdfgroup.org/x/WNnoAg) feature where a single chunk can be written from a pre-compressed buffer and circumvent the HDF5 pipeline. Even with relatively small compression factors this could significantly increase file writer performance.

There are a number of things to consider:

When to create the dataset?
-----------------------------------

* When writing a pre-compressed buffer using Direct Chunk Write, the compression filter ID and associated parameters need to be set already when creating the dataset. Currently this information is only available once the first frame arrive in the HDF5 plugin - which could be too late as files are often created earlier.

* The chunking of the HDF5 dataset must be configured so that a full NDArray fits in a single chunk. Again, that is a requirement of Direct Chunk Write. The chunking configuration is set on dataset creation time.

These two issues could be solved in a couple of different ways:

* When using "Lazy Open" the HDF5 file is not created until the first frame arrives - so it can be interrogated for compression settings before creating the HDF5 dataset. The downside is that creating the file can be time consuming and is often best done prior to starting acquisition to avoid dropping the first few frames.
* When **not** using "Lazy Open" we could have parameters on the HDF5 plugin that allow pre-defining what compression settings we expect NDArrays to arrive with. We can use these to create the HDF5 dataset - but will then need to validate compression parameters on every incoming frame (we probably need to do that anyway because we can't change compression mid-way through writing a dataset).

HDF5 support for JPEG filter?
-------------------------------------

The Blosc library comes with a registered HDF5 filter plugin (ID: 32001). However, it is not clear (to me at least) if there is an equivalent to the JPEG compression - a number of JPEG filters are listed, but would they work with the particular JPEG implementation in libjpeg that NDCodec uses?  https://support.hdfgroup.org/services/filters.html

GDYendell, 2019-02-19:
Is this complete as of #365 @ulrikpedersen ?

ulrikpedersen, 2019-02-26:
Implemented in #365 

Thanks @GDYendell

closed 2019-02-26 by ulrikpedersen