areaDetector/ADCore #350: Problems with compressed data in NTNDArrays

MarkRivers, 2018-11-25:
I am working on getting blosc and jpeg decompression of NTNDArrays from pvAccess working in the ImageJ plugin EPICS_NTNDA_Viewer.java.

I am using Java JNA to call the C code for the decompressors.

It is generally working fine for integer data types (int8, uint8, int16, uint16, int32, uint32)

I am having problems with float and double.   I believe the problem is that the blosc compressed buffer size is not the same as the size of the NTNDArray.value field times the field size.

When I am reading compressed Float32 arrays I see the following warnings in my code:
```
Warning: compressedSize=153185 != length*4=153184
Warning: compressedSize=153193 != length*4=153192
Warning: compressedSize=153193 != length*4=153192
Warning: compressedSize=153201 != length*4=153200
Warning: compressedSize=153201 != length*4=153200
Warning: compressedSize=153186 != length*4=153184
```
Note that the compressed size is often not a multiple of 4.  The Array.compressed size is not equal to Array.value.length*4.  When I decompress the blosc buffer I don’t get an error, but only the beginning 25-35% of the image has the correct values, the end of the image is all 0.

When reading Float64 arrays I see this.  The compressed size is not equal to NTNDArray.value.length*8.  In this case jblosc sometimes (but not always) returns an error.  But the decompressed array is never correct, the beginning is OK, but the end is all zeros.
```
Warning: compressedSize=300213 != length*8=300208
Warning: compressedSize=300307 != length*8=300304
Warning: compressedSize=300314 != length*8=300312
Warning: compressedSize=300319 != length*8=300312
jBlosc.decompress returned status=-1
Warning: compressedSize=300305 != length*8=300304
Warning: compressedSize=300321 != length*8=300320
Warning: compressedSize=300315 != length*8=300312
Warning: compressedSize=300319 != length*8=300312
jBlosc.decompress returned status=-1
Warning: compressedSize=300289 != length*8=300288
Warning: compressedSize=300318 != length*8=300312
jBlosc.decompress returned status=-1
```
It seems like there may be a bug in ndndArrayConverter.cpp, and it is not setting Array.value.length to be the correct size and is thus dropping a few bytes?  This is the code, which I am not completely understanding.
```
template <typename arrayType, typename srcDataType>
void NTNDArrayConverter::fromValue (NDArray *src)
{
    typedef typename arrayType::value_type arrayValType;

    string unionField(string(ScalarTypeFunc::name(arrayType::typeCode)) +
            string("Value"));

    NDArrayInfo_t arrayInfo;
    src->getInfo(&arrayInfo);

    int64 compressedSize = src->compressedSize;
    int64 uncompressedSize = arrayInfo.totalBytes;

    m_array->getCompressedDataSize()->put(compressedSize);
    m_array->getUncompressedDataSize()->put(uncompressedSize);

    // The uncompressed data type would be lost when converting to NTNDArray,
    // so we must store it somewhere. codec.parameters seems like a good place.
    PVScalarPtr uncompressedType(PVDC->createPVScalar(pvInt));
    uncompressedType->putFrom<int32>(NDDataTypeToScalar[src->dataType]);

    PVStructurePtr codec(m_array->getCodec());
    codec->getSubField<PVUnion>("parameters")->set(uncompressedType);
    codec->getSubField<PVString>("name")->put(src->codec);

    size_t count = src->codec.empty() ? arrayInfo.nElements : compressedSize;

    src->reserve();
    shared_vector<arrayValType> temp((srcDataType*)src->pData,
            freeNDArray<srcDataType>(src), 0, count);

    PVUnionPtr dest = m_array->getValue();
    dest->select<arrayType>(unionField)->replace(freeze(temp));
    dest->postPut();
}
```
I don't understand the comment:
```
// The uncompressed data type would be lost when converting to NTNDArray,
    // so we must store it somewhere. codec.parameters seems like a good place.
    PVScalarPtr uncompressedType(PVDC->createPVScalar(pvInt));
    uncompressedType->putFrom<int32>(NDDataTypeToScalar[src->dataType]);
```

This implies to me that the compressed data type and the uncompressed data type are different.  But this is not what I see, the data type on compressed NTNDArrays is the same as the uncompressed version.  Was the original plan to change the data type to uint8 for compressed data?  Is this still what should be done?

There is something else I don't understand.  This is pvget of an NTNDArray which is double[4,4].  
```
corvette:/usr/local/ImageJ>pvget -r "field()" 13SIM1:Pva1:Image
13SIM1:Pva1:Image
epics:nt/NTNDArray:1.0
    union value
        double[]  [2.71683e-312,1.3793e-312,8.70018e-313,18632.8,-2.64229e-308,4.61227e-82,7.86811e-62,-9.69781e+87,18631,18632,18633,18634,18632,18633,18634,18635,7.11455e-322,8.15208e-322,1.21071e-316,4.17292e-316,4.17344e-316,3.45846e-323,0.1,6.94118e-310,4.17342e-316,4.24399e-314,2.89339e-316,6.94106e-310,6.94106e-310,6.94106e-310,0,0,-nan,6.94111e-310,6.94118e-310,6.94106e-310,0,4.19956e-322,2.07508e-322,2.07508e-322,0,3.68604e+228,3.58168e+246,1.50947e+161,1.94671e+227,3.17109e+180,1.46016e-319,8.15208e-322,1.21071e-316,4.17292e-316,4.17344e-316,3.45846e-323,0.1,6.94118e-310,4.17342e-316,4.24399e-314,2.89339e-316,6.94106e-310,6.94106e-310,6.94106e-310,0,0,-nan,6.94111e-310,6.94118e-310]
    codec_t codec
        string name blosc
        any parameters
            int  10
    long compressedSize 65
    long uncompressedSize 128
    dimension_t[] dimension
        dimension_t
            int size 4
            int offset 0
            int fullSize 4
            int binning 1
            boolean reverse false
        dimension_t
            int size 4
            int offset 0
            int fullSize 4
            int binning 1
            boolean reverse false
```
The uncompressedSize makes sense: 4 * 4 * 8 = 128.  The compressedSize is 65, which is in bytes.  But then why is pvget printing out 65 doubles?  

ImageJ reports
```
Warning: compressedSize=65 != length*8=64
```
So it thinks there the length of Array.value is 8 doubles, so a total of 64 bytes.  This is one less than the uncompressed size.

MarkRivers, 2018-11-26:
I realized that some of the problem was in my ImageJ plugin.  I was reading compressedSize/elementSize (e.g.1,2,4,or 8) elements from the NTNDArray.value.  That is not reading enough data if compressedSize is not a multiple of elementSize.  I fixed that by always reading one additional element.

There is still a problem decoding Float32 and Float64 arrays.  For Float32 the Snappy filter works fine if Shuffle=None, so it is sort of working.  The other filters work fine to a certain point in the image, and then the rest of the image is zeros.  There are no longer any errors from the jblosc decompressor.

I wonder if the problem occurs when the float arrays contain a NaN value?

Would it be better to always set the NTNDArray to be PVByteArray when the data is compressed?  It certainly would be more straightforward to handle decompressing in Java which doesn't allow casting pointers like C++.

I tried to change ntndArrayConverter.cpp to do that, but my C++/pvData skills are not up to the task.

brunoseivam, 2018-11-26:
> This implies to me that the compressed data type and the uncompressed data type are different. But this is not what I see, the data type on compressed NTNDArrays is the same as the uncompressed [...]

Yes, compressed data should be in bytes. See my point when I opened the original PR:

> Regarding the last point, there is a small issue that I'd want to discuss. While compressed NTNDArrays will have ubyte type, there is no natural field to hold the original (uncompressed) data type. This information is important on the decompression side. I solved this issue by using the codec.parameters field, which is of type "any" (variant union). I simply shoved a PVInt32 there and set its value to the NDDataType_t value of the original array.

So, apparently, NDPva is not setting the value field correctly, as you observed. As you noted, it should be always PVByteArray (or PVUByteArray, I don't know if signedness matters in this context).

On the decompression side, you should read `codec.parameters` back to get the original data type.

I'm looking into it.

brunoseivam, 2018-11-26:
Hi Mark,

Please try with branch 'fix-350'.

PR #351 (2018-11-26): Fixes #350
Fixes #350
commit 87c7d32b (2018-11-26): Merge pull request #351 from areaDetector/fix-350

Fixes #350
closed 2018-11-26 by MarkRivers