areaDetector/ADCore #360: NDPluginCodec to support bitshuffle/lz4 compression

MarkRivers, 2018-12-06:
It would be good to add lz4 and bitshuffle/lz4 compression (non-Blosc versions) to NDPluginCodec.  One use case would be to have ADEiger support directly producing compressed NDArrays, without that driver having to decompress them. 

The ImageJ EPICS_NTNDA_Viewer would also need to support this compression.

Those compressed arrays could then be fed directly to NDPluginPva and then to viewers.  They could also be fed directly to NDFileHDF5 once it supports Direct Chunk Write.  The arrays would not need to be first decompressed by ADEiger and then re-compressed by NDPluginPva and NDFileHDF5.

The decompression for other plugins (NDPluginStats, etc.) could be handled in NDPluginCodec, which will speed up the ADEiger driver, and spread the load over more cores.  It might run at a reduced rate.

MarkRivers, 2018-12-07:
I have now added bitshuffle to ADSupport, and changed NDPluginCodec to support two new compressors: LZ4 (plain lz4) and BSLZ4 (bitshuffle/lz4).

I can set Codec1 to compress with either of these and Codec2 to decompress and it works fine, I get back the original array.

However,  the compression factor is not what I expect.  When using BSLZ4 the compression factor is always less than plain LZ4.  This is not what I see when I use the equivalent Blosc compression.

Here is a table that compares the measured compression factor when using the SimDetector Peaks mode and compressing UInt8, UInt16, and UInt32 data.  The Blosc tests all used Blosc Compression Level=9 (highest).

Data type | Blosc LZ4 Shuffle=None | Blosc LZ4 Shuffle=Bit | LZ4 | BSLZ4
----------  |  --------------------------| ------------------------|------|-------
UInt8       | 170.11                             | 170.11                         | 171.28 | 70.00
UInt16     | 192.13                            | 203.41                          | 193.5   | 96.25
UInt32    | 215.35                             | 225.32                          | 217.40 | 122.77

So when enabling Shuffle=Bit with Blosc it always improves the compression ratio for 16-bit and 32-bit data, and has no effect for 8-bit data.  Plain LZ4 has very similar compression to Blosc LZ4 with Shuffle=None.  However, BSLZ4 (bitshuffle) always has a **worse** compression than without bitshuffle.  This does not make sense to me.

Anyone have an idea what is wrong?

MarkRivers, 2018-12-09:
The bitshuffle code is now on the master branch.

brunoseivam, 2018-12-09:
> Anyone have an idea what is wrong?

My guess would be that the actual shuffling algorithm is different for each case. It may also depend on the data itself. Maybe actual Eiger data would compress better?

MarkRivers, 2018-12-13:
>Maybe actual Eiger data would compress better?

Bruno is correct.  Running with the Eiger I have found the following:
- lz4 compresses better than bitshuffle/lz4 when the number of non-zero pixels is very small.
- As the number of counts in the image increases bitshuffle/lz4 compresses better than lz4.
- The compressed size for both lz4 and bitshuffle/lz4 from the Eiger is the same (or a least very close) to the compressed size that NDPluginCodec produces with the same image and the same compressor.

I will report some quantitative tests in a future comment.

MarkRivers, 2018-12-13:
This is a comparison the various compressors on data from the Eiger 500K.  The Eiger was measuring a radioactive source that gave  ~1 count/s/pixel in the center ~10% of the detector, and 0.1 count/s/pixel averaged over the entire detector. 

The LZ4 or BSLZ4 compression was done on the Eiger.  These data were decompressed by Codec1 on the IOC, giving the compression factors for those codecs, That decompressed data was then recompressed by Codec2.  When using BSLZ4 on the Eiger and in Codec2 the compression factors were exactly the same, verifying that the same algorithm is being used.  When using LZ4 on the Eiger and in Codec2 the compression factors were better in NDPluginCodec than the Eiger LZ4 compression.  For example, at 100 sec the Eiger LZ4 compression factor was 1.60, while NDPluginCodec was 2.85.   This may be because NDPluginCodec is using a more recent version of lz4 (7.5.1) than the Eiger is? 

Codec2 was also used to measured the Blosc compressors.  The Blosc compressors were run with Compression Level=9.  The LZ4 (fast) and ZSTD (slow) Blosc codecs were tested.  They were each tested with no shuffle, Bit shuffle, and Byte shuffle.

**Compression factor (uncompressed size/compressed size) vs exposure time for several codecs**

Codec  | 0.01 sec | 0.1 sec | 1.0 sec | 10 sec | 100 sec
------- |  ---------- | ------- | ------ | ------- | -------
LZ4                      | 146.5 |  56.3    | 9.5 | 2.8  | 1.6
BSLZ4                   | 109. 5 | 70.1   | 29.5 | 13.8 | 7,5
Blosc LZ4            | 146.4  | 58.7   | 11.6 | 4.2 | 2.8
Blosc LZ4 Bit       |  144,4 | 64.8   | 17.6 | 7.6 | 4.7
Blosc LZ4 Byte    | 149,0  | 90.6   | 34.2 | 14.2 | 7.5
Blosc ZSTD         | 400.2  | 99.6   | 41.7 | 13.9 | 6.6
Blosc ZSTD Bit    | 400.2  | 169.6  | 43.1 | 17.0 | 7.4
Blosc ZSTD Byte | 227.8  | 176.2  | 56.5 | 18.3 | 8.3

When the counts are very low with many zeros the Blosc ZSTD compressor is significantly better.  But for more realistic counts the Eiger BSLZ4 is about the same, and is ~100 times faster.

xiaoqiangwang, 2018-12-14:
Thanks for the benchmarking. Any information about the speed? The blosc bit/byte shuffle algorithms can optionally use SIMD operations. See 
https://github.com/areaDetector/ADSupport/blob/master/supportApp/bloscSrc/Makefile#L60
https://github.com/areaDetector/ADSupport/blob/master/supportApp/bloscSrc/Makefile#L64

MarkRivers, 2019-03-19:
NDPluginCode now supports bitshuffle/lz4, so closing this.  The performance tables may be useful to extract for future reference.

closed 2019-03-19 by MarkRivers