areaDetector/ADCore #363: Can this PV be smaller?

prjemian, 2018-12-13:
In [NDFileHDF5.template : `XMLFileName`](https://github.com/areaDetector/ADCore/blob/master/ADApp/Db/NDFileHDF5.template#L676), does the NELM field need to be so large?  It is defined as a 1MB.  Perhaps a kB or two would be more than sufficient?

The MB size is causing problems when I try to record that PV value with PyEpics. See https://github.com/APS-2BM-MIC/ipython-user2bmb/issues/13

MarkRivers, 2018-12-13:
The PV was made that large to be able to directly write the XML layout using Channel Access.  I agree it is too large a default.  I have changed it to be controlled by the macro XMLSIZE with a default of 2048.  This default is set both in NDFileHDF.template (5ad2a688308f523b23ab9193a52c4f584e2d1b4c) and in EXAMPLE_commonPlugins.cmd (3765026bede4daecb6d9670b1f2981a4fad5bc24).  That first commit also changed ADDR and TIMEOUT to be macro parameters to be consistent with other plugin template files.

Users who require a larger value will need to change their commonPlugins.cmd or other startup script to set a larger value for XMLSIZE.

MarkRivers, 2018-12-13:
Closed via 5ad2a688308f523b23ab9193a52c4f584e2d1b4c and 3765026bede4daecb6d9670b1f2981a4fad5bc24.

closed 2018-12-13 by MarkRivers