areaDetector/ADCore #370: Problem with direct chunk write

MarkRivers, 2019-01-10:
I see a problem with direct chunk write, or more specifically with going back to non-chunk write mode.

The problem is easy to reproduce:
- Run the simDetector with 1024x1024, Float32 images
- Set HDF5 plugin to get data from SIM1, save a single HDF5 file.  It is 4MB, which is correct.
`-rw-rw-r--   1 epics     domain users  4.2M Jan 10 16:56 test_hdf5_1.h5`
- Set the CODEC1 plugin to blosc, lz4, byte shuffle.  The compression is ~4, so the compressed images are ~1MB.
- Set the HDF5 plugin to get its data from CODEC1, and save a single HDF5 file.  The size is ~1MB as expected.
`-rw-rw-r--   1 epics     domain users 1012K Jan 10 16:57 test_hdf5_3.h5`
- Now switch the HDF5 plugin back to get its data from SIM1 again.
- The first problem is that the HDF5 Compression_RBV is still blosc.  This is probably not what the user expects.  It was set to blosc because that was the previous input compressed array.  But now it is configured to do blosc compression inside the HDF5 plugin.  I'm not sure how to handle this.  We may need an "actual compression" PV.
- I manually set the HDF5 compression to None.  I expect the saved file to be 4MB again.
- However, it is not, it is still ~1MB.
`-rw-rw-r--   1 epics     domain users  1.1M Jan 10 16:58 test_hdf5_9.h5`
- I'm not sure what it wrong?  Why does it still seem to be compressed even though I set Compression=None.  Is it really doing the compression inside the HDF5 code, or is it still thinking the NDArray itself is pre-compressed?
- It seems like I need to restart the IOC to disable the compression?

ulrikpedersen, 2019-01-11:
@GDYendell: could you take a look at this, please?

GDYendell, 2019-01-11:
@MarkRivers I don't see the same behaviour. If I capture a 1024x1024, float32, 1 second exposure frame compressed with NDCodec using blosc(LZ4, ByteShuffle). I get compression of ~14 and frames of ~ 325KB. If I point the HDF5 plugin at the SimDetector directly (and set compression None) I then get 4MB frames as expected. I can then swap back and forth between None and blosc and the file sizes are as expected.

I added some FLOW logging that give some idea of the codec definitions in the plugin and on the NDArrays. Does the `Setting blosc compression filter` message get printed when you have set compression None and the file still looks compressed?

MarkRivers, 2019-01-11:
I just tested again.  This time I configured the simDetector with Linear Ramp mode rather than Sine mode and I was getting compressing ~80.  When I save HDF5 directly from SIM1 it works.  I enabled TraceFlow.

```
2019/01/11 08:59:17.000 NDFileHDF5:writeInt32: calling base class function=28, value=1 old=0
2019/01/11 08:59:17.000 NDFileHDF5::openFile Filename: /home/epics/scratch/test_hdf5_uncompressed_8.h5
2019/01/11 08:59:17.001 NDFileHDF5::openFile attribute list copied. num pArray attributes = 41 local copy = 37
2019/01/11 08:59:17.001 NDFileHDF5::configureDims  NDArray:   {  1024, 1024, }
2019/01/11 08:59:17.001 NDFileHDF5::configureDims dimension report:
 framesize   1024,  1024,
 chunkdims   1024,  1024,
   maxdims   1024,  1024,
      dims   1024,  1024,
    offset      0,     0,
   virtual      0,
2019/01/11 08:59:17.001 NDFileHDF5::createNewFile Calculated istorek=0
2019/01/11 08:59:17.001 NDFileHDF5::createFileLayout Creating dataspace with given dimensions
2019/01/11 08:59:17.001 NDFileHDF5::createFileLayout Configuring chunking
2019/01/11 08:59:17.001 NDFileHDF5::createFileLayout Setting fillvalue
2019/01/11 08:59:17.001 NDFileHDF5::createFileLayout Empty layout file, use default layout
2019/01/11 08:59:17.002 NDFileHDF5::createXMLFileLayout Root tree: < Group: '' groups=1 dsets=0 attr=0>
                                  < Group: '/entry' groups=2 dsets=0 attr=1>
                                  < Group: '/entry/data' groups=0 dsets=0 attr=1>
                                  < Group: '/entry/instrument' groups=3 dsets=0 attr=1>
                                    < Group: '/entry/instrument/NDAttributes' groups=0 dsets=0 attr=1>
                                    < Group: '/entry/instrument/detector' groups=1 dsets=1 attr=1>
                                      < Group: '/entry/instrument/detector/NDAttributes' groups=0 dsets=1 attr=1>
                                    < Group: '/entry/instrument/performance' groups=0 dsets=1 attr=0>
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrStr name=NX_class value=NXentry
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrStr name=NX_class value=NXdata
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrStr name=NX_class value=NXinstrument
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrStr name=NX_class value=NXcollection
2019/01/11 08:59:17.002 NDFileHDF5::createDatasetDetector Setting cache size=0 slots=2
2019/01/11 08:59:17.002 NDFileHDF5::createDatasetDetector Creating first empty dataset called "data"
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrInt32 name=NDArrayDimBinning value=1,1,
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrInt32 array found, size: 2
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrInt32 name=NDArrayDimOffset value=0,0,
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrInt32 array found, size: 2
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrInt32 name=NDArrayDimReverse value=0,0,
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrInt32 array found, size: 2
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrInt32 name=NDArrayNumDims value=2
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrStr name=NX_class value=SDS
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrInt32 name=signal value=1
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrStr name=NX_class value=NXdetector
2019/01/11 08:59:17.002 NDFileHDF5::writeH5attrStr name=NX_class value=NXcollection
2019/01/11 08:59:17.002 NDFileHDF5::createXMLFileLayout Default dataset name: /entry/instrument/detector/data
2019/01/11 08:59:17.002 NDFileHDF5::createAttributeDataset Creating attribute datasets. extradims=0 attribute count=41
2019/01/11 08:59:17.007 NDFileHDF5::writeFile getting attribute list
2019/01/11 08:59:17.007 NDFileHDF5::writeFile copying attribute list
2019/01/11 08:59:17.007 NDFileHDF5Dataset::writeFile: set_extent dims={1024,1024,0}
2019/01/11 08:59:17.007 Dataset chunk dimensions [1024, 1024, 0] do not match Array dimensions [1024, 1024]
2019/01/11 08:59:17.007 NDFileHDF5Dataset::writeFile NDArray not correctly chunked. Using standard write
2019/01/11 08:59:17.009 NDFileHDF5::writeFile wrote frame. dt=0.00202s (T=0.00811s)
2019/01/11 08:59:17.009 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'NDArrayUniqueId'
2019/01/11 08:59:17.009 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'NDArrayTimeStamp'
2019/01/11 08:59:17.009 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'NDArrayEpicsTSSec'
2019/01/11 08:59:17.009 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'NDArrayEpicsTSnSec'
2019/01/11 08:59:17.009 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'ColorMode'
2019/01/11 08:59:17.009 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'AcquireTime'
2019/01/11 08:59:17.009 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'RingCurrent'
2019/01/11 08:59:17.009 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'RingCurrent_EGU'
2019/01/11 08:59:17.009 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'ID_Energy'
2019/01/11 08:59:17.010 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'ID_Energy_EGU'
2019/01/11 08:59:17.010 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'ImageCounter'
2019/01/11 08:59:17.010 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'MaxSizeX'
2019/01/11 08:59:17.010 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'MaxSizeY'
[... cut ...]
2019/01/11 08:59:30.185 Dataset chunk dimensions [1024, 1024, 0] do not match Array dimensions [1024, 1024]
2019/01/11 08:59:30.185 NDFileHDF5Dataset::writeFile ERROR Unable to write pre-compressed data - mismatched chunk definition
2019/01/11 08:59:30.185 NDFileHDF5::writeFile ERROR: could not write to dataset. Aborting
2019/01/11 08:59:30.186 NDPluginFile::writeFileBase Error writing file, status=3
```

It looks like this is the problem:
> 2019/01/11 08:59:30.185 Dataset chunk dimensions [1024, 1024, 0] do not match Array dimensions [1024, 1024]

MarkRivers, 2019-01-11:
I am running the master branch of ADSupport, ADCore, and ADSimDetector.  I built with HDF5_EXTERNAL=NO so it is using the HDF5 library in ADSupport.  Can you try this configuration?

If I switch CODEC1 to Compressor=None then the HDF5 plugins saves OK.  When I set it to Blosc I now always get the above error.

GDYendell, 2019-01-11:
> `2019/01/11 08:59:30.185 Dataset chunk dimensions [1024, 1024, 0] do not match Array dimensions [1024, 1024]`

Does it work if you explicitly define the chunk dimensions as `[1024, 1024, 1]`?

MarkRivers, 2019-01-11:
When I changed to Stream mode with NumCapture=1 then the error went away.  It was happening in Single mode.  When I went back to Single mode it continued to work.  This was with LazyOpen=No.

However, I continue to have the original problem.  My "uncompressed" files are not the correct size, they are the compressed size.

When I save an uncompressed array (HDF5 source=SIM1) with TraceFlow on I see the following:
```

2019/01/11 09:27:43.316 NDFileHDF5:writeInt32: function=28, value=1 old=0
2019/01/11 09:27:43.316 NDFileHDF5:writeInt32: calling base class function=28, value=1 old=0
2019/01/11 09:27:43.316 NDFileHDF5::openFile Filename: /home/epics/scratch/test_hdf5_uncompressed_40.h5
2019/01/11 09:27:43.316 NDFileHDF5::openFile attribute list copied. num pArray attributes = 41 local copy = 37
2019/01/11 09:27:43.316 NDFileHDF5::configureDims  NDArray:   {  1024, 1024, }
2019/01/11 09:27:43.316 NDFileHDF5::configureDims dimension report:
 framesize   1024,  1024,
 chunkdims   1024,  1024,
   maxdims   1024,  1024,
      dims   1024,  1024,
    offset      0,     0,
   virtual      0,
2019/01/11 09:27:43.316 NDFileHDF5::createNewFile Calculated istorek=0
2019/01/11 09:27:43.317 NDFileHDF5::createFileLayout Creating dataspace with given dimensions
2019/01/11 09:27:43.317 NDFileHDF5::createFileLayout Configuring chunking
2019/01/11 09:27:43.317 NDFileHDF5::createFileLayout Setting fillvalue
2019/01/11 09:27:43.317 NDFileHDF5::createFileLayout Empty layout file, use default layout
2019/01/11 09:27:43.317 NDFileHDF5::createXMLFileLayout Root tree: < Group: '' groups=1 dsets=0 attr=0>
                                  < Group: '/entry' groups=2 dsets=0 attr=1>
                                  < Group: '/entry/data' groups=0 dsets=0 attr=1>
                                  < Group: '/entry/instrument' groups=3 dsets=0 attr=1>
                                    < Group: '/entry/instrument/NDAttributes' groups=0 dsets=0 attr=1>
                                    < Group: '/entry/instrument/detector' groups=1 dsets=1 attr=1>
                                      < Group: '/entry/instrument/detector/NDAttributes' groups=0 dsets=1 attr=1>
                                    < Group: '/entry/instrument/performance' groups=0 dsets=1 attr=0>
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrStr name=NX_class value=NXentry
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrStr name=NX_class value=NXdata
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrStr name=NX_class value=NXinstrument
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrStr name=NX_class value=NXcollection
2019/01/11 09:27:43.317 NDFileHDF5::createDatasetDetector Setting cache size=4194304 slots=53
2019/01/11 09:27:43.317 NDFileHDF5::createDatasetDetector Creating first empty dataset called "data"
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrInt32 name=NDArrayDimBinning value=1,1,
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrInt32 array found, size: 2
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrInt32 name=NDArrayDimOffset value=0,0,
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrInt32 array found, size: 2
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrInt32 name=NDArrayDimReverse value=0,0,
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrInt32 array found, size: 2
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrInt32 name=NDArrayNumDims value=2
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrStr name=NX_class value=SDS
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrInt32 name=signal value=1
2019/01/11 09:27:43.317 NDFileHDF5::writeH5attrStr name=NX_class value=NXdetector
2019/01/11 09:27:43.318 NDFileHDF5::writeH5attrStr name=NX_class value=NXcollection
2019/01/11 09:27:43.318 NDFileHDF5::createXMLFileLayout Default dataset name: /entry/instrument/detector/data
2019/01/11 09:27:43.318 NDFileHDF5::createAttributeDataset Creating attribute datasets. extradims=0 attribute count=41
2019/01/11 09:27:43.322 NDFileHDF5::writeFile getting attribute list
2019/01/11 09:27:43.322 NDFileHDF5::writeFile copying attribute list
2019/01/11 09:27:43.322 NDFileHDF5Dataset::writeFile: set_extent dims={1024,1024,0}
2019/01/11 09:27:43.322 NDFileHDF5Dataset::writeFile NDArray correctly chunked. Using direct chunk write
2019/01/11 09:27:43.324 NDFileHDF5::writeFile wrote frame. dt=0.00115s (T=0.00757s)
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'NDArrayUniqueId'
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'NDArrayTimeStamp'
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'NDArrayEpicsTSSec'
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'NDArrayEpicsTSnSec'
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'ColorMode'
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'AcquireTime'
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'RingCurrent'
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'RingCurrent_EGU'
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'ID_Energy'
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'ID_Energy_EGU'
2019/01/11 09:27:43.324 NDFileHDF5::closeAttributeDataset: closing attribute dataset 'ImageCounter'
[... cut ...]
2019/01/11 09:27:43.326 NDFileHDF5::closeFile closing HDF datatype 216172782113789615
2019/01/11 09:27:43.326 NDFileHDF5::closeFile closing groups
2019/01/11 09:27:43.327 NDFileHDF5::closeFile file closed! runtime=0.011 s overall acquisition performance=3008.54 Mbit/s
```
Note that it says it is using direct chunk write, which does not seem correct.

This is a 1024x1024 Float32 array, so the file should be 4MB.  But it is only 0.5 MB.
`-rw-rw-r--   1 epics     domain users     580007 Jan 11 09:27 test_hdf5_uncompressed_40.h5`

GDYendell, 2019-01-11:
That is what I would expect. It can use direct chunk write uncompressed frames and there is a performance gain to be had. But, obviously the file is wrong.

If you run `h5dump -Hp <file>` does it show a compression factor and filter definition? E.g.: 
```
...
STORAGE_LAYOUT {
               CHUNKED ( 1, 1024, 1024 )
               SIZE 290695 (14.429:1 COMPRESSION)
            }
            FILTERS {
               USER_DEFINED_FILTER {
                  FILTER_ID 32001
                  COMMENT blosc
                  PARAMS { 2 2 4 4194304 5 0 1 }
               }
            }
...
```

MarkRivers, 2019-01-11:
I shows no filter:
```
HDF5 "test_hdf5_uncompressed_40.h5" {
GROUP "/" {
   GROUP "entry" {
      ATTRIBUTE "NX_class" {
         DATATYPE  H5T_STRING {
            STRSIZE 7;
            STRPAD H5T_STR_NULLTERM;
            CSET H5T_CSET_ASCII;
            CTYPE H5T_C_S1;
         }
         DATASPACE  SCALAR
      }
      GROUP "data" {
         ATTRIBUTE "NX_class" {
            DATATYPE  H5T_STRING {
               STRSIZE 6;
               STRPAD H5T_STR_NULLTERM;
               CSET H5T_CSET_ASCII;
               CTYPE H5T_C_S1;
            }
            DATASPACE  SCALAR
         }
         DATASET "data" {
            DATATYPE  H5T_IEEE_F32LE
            DATASPACE  SIMPLE { ( 1024, 1024 ) / ( 1024, 1024 ) }
            STORAGE_LAYOUT {
               CHUNKED ( 1024, 1024 )
               SIZE 451863
            }
            FILTERS {
               NONE
            }
            FILLVALUE {
               FILL_TIME H5D_FILL_TIME_IFSET
               VALUE  0
            }
            ALLOCATION_TIME {
               H5D_ALLOC_TIME_INCR
            }
```

Note that the size 451863 which is incorrect.

GDYendell, 2019-01-11:
It looks like it has written a compressed block without setting the filter. I saw this while I was developing. It's possible the NDCodec is providing compressed data without setting the codec on the NDArray.

GDYendell, 2019-01-11:
If I build against the ADSupport-compiled HDF5 1.10.1, I get a segfault trying to close the file when I run the HDF5 plugin with blosc compression and point it at simDetector. I have to run it with no compression or point it at Codec. It works OK if I build against our external HDF5 1.10.2 (I had other problems with 1.10.1 unfortunately, so that's the closest comparison).

MarkRivers, 2019-01-11:
> It looks like it has written a compressed chunk without setting the filter. I saw this while I was developing when the NDPluginCodec was providing compressed data without setting the codec on the NDArray.

I am getting the data directly from SIM1, not from the CODEC.  There is no Codec set in the NDArray.  

It looks to me like the HSD5 plugin may be using the NDArray.compressedSize even when the codec is not set.  compressedSize should be ignored if the codec is null.

GDYendell, 2019-01-14:
NDArray states that compressedSize should be equal to dataSize if it is uncompressed (https://github.com/areaDetector/ADCore/blob/master/ADApp/ADSrc/NDArray.h#L129). Is that not the case? What is the difference between dataSize and info.totalBytes as you use in your change?

If I print the compressedSize in the direct write if block, it is as expected (first through codec and then direct to simDetector without compression):

```
   // Write the data to the hyperslab.
   if (H5_VERSION_GE(1, 8, 11) && verifyChunking(pArray) == asynSuccess) {
     // The chunking and compression settings match - use direct chunk write
    asynPrint(this->pAsynUser_, ASYN_TRACE_FLOW,
               "%s::%s NDArray correctly chunked. Using direct chunk write\n",
               fileName, functionName);
+    asynPrint(this->pAsynUser_, ASYN_TRACE_ERROR,
+              "compressedSize: %d\n", (int) pArray->compressedSize);
     #if H5_VERSION_GE(1, 10, 3)
     hdfstatus = H5Dwrite_chunk(this->dataset_, H5P_DEFAULT, 0x0,
                                this->offset_, pArray->compressedSize, pArray->pData);
```

```
2019/01/14 09:13:29.682 NDFileHDF5::configureCompression Setting blosc compression filter level=5, shuffle=0, compressor=0
2019/01/14 09:13:29.683 NDFileHDF5Dataset::writeFile NDArray correctly chunked. Using direct chunk write
2019/01/14 09:13:29.683 compressedSize: 456894

2019/01/14 09:13:40.670 NDFileHDF5Dataset::writeFile NDArray correctly chunked. Using direct chunk write
2019/01/14 09:13:40.670 compressedSize: 4194304
```
Do you not get the same behaviour? Maybe it works when it first boots, but if you play around with the settings something goes wrong?

MarkRivers, 2019-01-14:
> What is the difference between dataSize and info.totalBytes as you use in your change?

dataSize is the actual amount of memory allocated for *pData,  which may be more than  required to hold the array.  This will happen if an NDArray from the pool is re-used, and allocated size happens to be more than the required size.  arrayInfo.totalBytes is the amount that the array actually uses, rather than the allocated size.

NDArray.compressedSize was added recently by Bruno, and it appears that the comment is not correct.  compressedSize appears not to be updated when an array is re-allocated from the pool.  compressedSize would not be a good name for that variable for a uncompressed array anyway. 

Perhaps we should change to dataSize (allocated size of pData), arraySize (actual size used by the current NDArray of given dimensions and datatype, and compressedSize (compressed size if codec is not empty).

MarkRivers, 2019-01-14:
I looked at NDArrayPool.cpp and indeed compresedSize is set to dataSize only when the pData is first allocated.  compressedSize is not update when an array is allocated from the pool and its arrayInfo.totalBytes is less than dataSize.  At that point compressedSize should be set to totalBytes.

For now to be safe we need to use arrayInfo.totalBytes as the actual size of the data used by the currentl NDArray.

GDYendell, 2019-01-14:
> compressedSize appears not to be updated when an array is re-allocated from the pool

That explains things. I am currently running a script where it randomly swaps between simDetector and Codec. Do you have to run for a while before it goes wrong?

GDYendell, 2019-01-15:
Just for sanity I wanted to try to reproduce the compressedSize bug. I have seen the problem. I ran my script for a while and kept varying the size on the simDetector.

I have a fix for direct chunk write in Single Mode. I can open a PR for this.

I think this just leaves the problem with swapping back to uncompressed arrays after running compressed arrays and compression still being enabled. Have you decided on the best solution for this, @MarkRivers ?

MarkRivers, 2019-01-15:
> I have a fix for direct chunk write in Single Mode. I can open a PR for this.

Sounds good.

I think we can live with the current state of leaving compression on.  I don't want to add another PV for "actual compression" because the HDF5 plugin is already complex enough.

PR #372 (2019-01-26): Direct chunk write 2
Fixes #370
closed 2019-01-26 by MarkRivers