areaDetector/ADCore #379: XML2 requirement in common Makefiles

MJGaughran, 2019-02-21:
The ADCore commonLibraryMakefile and commonDriverMakefile now require XML2. As this is in the common Makefiles, all other area detector modules now require xml2.

For Windows dynamic builds in particular, modules have to include both ADSupport and ADCore in a configure/RELEASE file. I would assume for Linux where xml2 is not a system lib, all modules will have to specify XML2_LIB in CONFIG_SITE, although I have not tested this.

Is there any other configuration that I should have changed for either ADCore or ADSupport?

Would it be possible to change this to explicitly raise an error in the ADSrc and pluginSrc Makefiles, such as with:
```
ifneq ($(WITH_XML2),YES)
$(error Module requires XML2)
endif
```

MarkRivers, 2019-02-21:
> The ADCore commonLibraryMakefile and commonDriverMakefile now require XML2. As this is in the common Makefiles, all other area detector modules now require xml2.

Yes, and this is documented in areaDetector/configure/EXAMPLE_CONFIG_SITE.local

```
EXAMPLE_CONFIG_SITE.local:# As of ADCORE R3-0 libxml2 is required for ADCore so WITH_XML2 is not supported
EXAMPLE_CONFIG_SITE.local:# XML2 is required for ADCore so there is no WITH_XML2.
```

> For Windows dynamic builds in particular, modules have to include both ADSupport and ADCore in a configure/RELEASE file. 

Yes, and this is done in areaDetector/configure/EXAMPLE_RELEASE* files
```
corvette:~/devel/areaDetector/configure>grep ADSupport EXAMPLE*
EXAMPLE_RELEASE.local:ADSUPPORT=$(AREA_DETECTOR)/ADSupport
EXAMPLE_RELEASE.local.APS_Jenkins:ADSUPPORT=$(AREA_DETECTOR)/ADSupport
EXAMPLE_RELEASE_LIBS.local:# ADSupport is needed by ADCore plugins
EXAMPLE_RELEASE_LIBS.local:ADSUPPORT=$(AREA_DETECTOR)/ADSupport
EXAMPLE_RELEASE_PRODS.local:# ADSupport is needed by ADCore plugins
EXAMPLE_RELEASE_PRODS.local:ADSUPPORT=$(AREA_DETECTOR)/ADSupport
```
> I would assume for Linux where xml2 is not a system lib, all modules will have to specify XML2_LIB in CONFIG_SITE, although I have not tested this.

If it is not a system lib and you are not using the version in ADSupport then yes, you must specify XML2_LIB.  But generally I would recommend using the version in ADSupport if not using a system lib.

> Is there any other configuration that I should have changed for either ADCore or ADSupport?

Not that I know of.

> Would it be possible to change this to explicitly raise an error in the ADSrc and pluginSrc Makefiles, such as with:

```
ifneq ($(WITH_XML2),YES)
$(error Module requires XML2)
endif
```

No, because WITH_XML2 is no longer used at all, as documented in EXAMPLE_CONFIG_SITE.local, so we cannot test its value.

MJGaughran, 2019-03-01:
I now understand that the way we structure our area detector modules is a little different at Diamond. The CONFIG_SITE files are not shared between all modules.

I don't think there's a convenient alternative to your configuration, so I'm closing this issue.

closed 2019-03-01 by MJGaughran