areaDetector/ADCore #386: Why are the FFT dimensions half of those of the original image?

bsobhani, 2019-03-08:
The following lines are in NDFFTPlugin.cpp: 

pPvt->nFreqX = pPvt->nTimeX / 2;
pPvt->nFreqY = pPvt->nTimeY / 2;

This results in FFTAbs (the matrix that is displayed in CSS) being 1/4 of the size of the original image. I thought a 2D fft should have the same dimensions as its input.

MarkRivers, 2019-03-08:
> I thought a 2D fft should have the same dimensions as its input.

When taking the FFT of an N element array the result is 2 arrays each of size N/2, the real part and the imaginary part.  The factor of 2 reflects the Nyquist limit, there are only N/2 frequencies.  The absolute value is sqrt(real^2 + imaginary^2), which is only dimensions N/2.  The same is true in 2-D.

bsobhani, 2019-03-08:
Ok, I just tried taking some sample ffts on wolfram alpha and it looks like elements in the latter half of the array are redundant.

closed 2019-03-08 by bsobhani