areaDetector/ADCore #388: NDAttribute timeseries as circular buffer

hinxx, 2019-03-22:
Currently it is not possible to have NDAttribute plugin operate circular buffer of timeseries points. I guess this issue is similar to #327.
Would #333 be the way to go for this one, too? I can try to provide a patch.

MarkRivers, 2019-03-22:
I think the same approach we did with NDPluginStats probably makes sense. I’m happy for you to make a pull request.

hinxx, 2019-03-22:
I have a dirty patch,...

Current implementation of timeseries in NDAttribute performs callbacks per asyn port address, each containing a 1-D NDArray. If I want to keep this behavior I think I need a NDTimeSeries plugin instance for each NDAttributeN template instance. Also the NDArrays I allocate are single point, 1-D arrays. This is what I have right now.

I like how NDStats does it with 1-D NDArray holding multiple value from the same time point. I think NDAttrabute could do the same for every attribute it tracks, but I'm not sure of the impact of such change on current users. My use case would favor this solution.

@MarkRivers, what do you think?

hinxx, 2019-03-22:
Pull request #389 contains changes to achieve behavior as described in the first paragraphs of the previous comment.

MarkRivers, 2019-03-28:
I merged the pull request without reading your comment carefully enough.  

I agree that it should be changed to pass all of the values in a single NDArray, like the Stats plugin does.  That way only a single time series plugin is plugin is needed

I don't see how this would adversely impact current users, since it currently erases them all at the same time, they are not independent.

I have made the change an pushed to Github.  I have also
- Changed EXAMPLE_commonPlugins.cmd and EXAMPLE_commonPlugins_settings.req to load and save the TimeSeries plugin.
- Changed the medm screens and autoconverted opi, ui, and edl screens to use the TimeSeries plugin.
- Added a note on the changes to RELEASE.md for R3-5.

hinxx, 2019-03-28:
I see the difference, looks better than what I had. Also thanks for fixing missing bits in pull request!

MarkRivers, 2019-03-28:
This issue is resolved by #389

closed 2019-03-28 by MarkRivers