areaDetector/ADCore #394: lz4 compression not supported in HDF5 plugin

MarkRivers, 2019-04-02:
NDPluginCodec can compress NDArrays with pure lz4 compression (i.e. without bitshuffle or Blosc).  However, the HDF5 plugin does not support this, and trying to save such arrays crashes the IOC.

I will try to add support for lz4 but may need some help from @xiaoqiangwang .

xiaoqiangwang, 2019-04-02:
Here is an LZ4 filter implementation 
https://github.com/nexusformat/HDF5-External-Filter-Plugins/blob/master/LZ4/src/H5Zlz4.c

It should be straight forward to integrate the source.

MarkRivers, 2019-04-02:
Yes, I had found that and am using it as a starting point.  Unfortunately it does not have a header file, so it seems it was intended only for using dynamic loading.  I am modifying to allow static loading as well.

MarkRivers, 2019-04-02:
I have created pull requests in ADCore and ADSupport to implement the lz4 filter with HDF5.

MarkRivers, 2019-04-03:
What testing have others done to see if HDF5 files written with all of the different compressions can actually be read successfully?  This would include files that are compressed by the HDF5 library (i.e. uncompressed arrays received by NDFileHDF5) as well as pre-compressed arrays written with Direct Chunk Write.

How does one configure the dynamic filter loading so that h5dump will find the filter and use it?

I am worried that, at least for lz4, files written with the new H5Zlz4.c will be different from files written with Direct Chunk Write.  H5Zlz4.c breaks up the writes into blocks of no more than 1 or 2 GB with some additional header bytes, while files written with Direct Chunk Write will lack these extra header bytes?  

We need a way to test that files we write are actually readable.

GDYendell, 2019-04-03:
I tested blosc (NDCodec and hdf5 compressed) compression when adding Direct Write Chunk. I didn't play with level or the shuffle types.

To view with h5dump you need to export the path to hdf5 filter libraries beforehand, e.g.:

```
export HDF5_PLUGIN_PATH=/dls_sw/prod/tools/RHEL6-x86_64/hdf5filters/0-5-1/prefix/hdf5_1.10/h5plugin
```

where that folder contains

`libh5bshuf.so   libh5lz4.so     libH5Zblosc.so
`

@ulrikpedersen has put together a collection of these [here](https://github.com/dls-controls/hdf5filters) for use at DLS with the sources and some docs.

MarkRivers, 2019-04-03:
How are the names of those shareable library files determined?

It seems like it would be nice to build those filter libraries in ADSupport, since we have all of the necessary source files.  We can build them for Windows, Linux, etc.

MarkRivers, 2019-04-03:
I have read the HDF5 document about dynamic filters here:
https://support.hdfgroup.org/HDF5/doc/Advanced/DynamicallyLoadedFilters/HDF5DynamicallyLoadedFilters.pdf

I cannot find any description of what the library files should be named, so perhaps it does not matter?  Does it just look for all of the shareable libraries in HDF5_PLUGIN_PATH and see what filter each supports?

If that is the case, can we just use the libhdf5.so file that we build in ADSupport, since it contains the definitions for all of the dynamic filters?  I tried that to run h5dump on a Blosc compressed file and it did not work.  I am confused about what is different about that library from the ones that Ulrik created, since it should have the same symbols?

GDYendell, 2019-04-03:
I think it does not matter what the .so is called, it will just load everything in the directory it is given and search for `H5PLget_plugin_type` methods, as in 4.3 of that link.

MarkRivers, 2019-04-03:
> it will just load everything in the directory it is given and search for H5PLget_plugin_type methods, as in 4.3 of that link.

OK, I understand.  ADSupport/supportApp/hdf5Src contains the code for the Blosc, bitshuffle, and lz4 HDF5 filters.  Of these, only lz4 that I added yesterday contains the H5PLget_plugin_type method.  But that means that the resulting libhdf5.so file should be able to dynamically decompress lz4 compressed files.  In fact it works!  This is the h5dump output when HDF5_PLUGIN_PATH points to ADSupport/lib/linux-x86_64.

I don't think this solution is going to work for adding other filters however, since H5PLget_plugin_type is a global symbol and we will get link errors if multiple source files define it.  I need to find a way to make separate libraries for each of the filters.  But I think it will be convenient for most users to have those libraries built as part of ADSupport.

```
HDF5 "/home/epics/scratch/test_hdf5_86.h5" {
GROUP "/" {
   GROUP "entry" {
      GROUP "data" {
         DATASET "data" {
            DATATYPE  H5T_STD_I32LE
            DATASPACE  SIMPLE { ( 1, 1024, 1024 ) / ( 1, 1024, 1024 ) }
            STORAGE_LAYOUT {
               CHUNKED ( 1, 1024, 1024 )
               SIZE 58499 (71.699:1 COMPRESSION)
            }
            FILTERS {
               USER_DEFINED_FILTER {
                  FILTER_ID 32004
                  COMMENT HDF5 lz4 filter; see http://www.hdfgroup.org/services/contributions.html
                  PARAMS { 0 0 }
               }
            }
            FILLVALUE {
               FILL_TIME H5D_FILL_TIME_IFSET
               VALUE  0
            }
            ALLOCATION_TIME {
               H5D_ALLOC_TIME_INCR
            }
            DATA {
            (0,0,0): 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
            (0,0,16): 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
            (0,0,30): 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45,
            (0,0,44): 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59,
            (0,0,58): 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73,
            (0,0,72): 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87,
            (0,0,86): 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100,
            (0,0,99): 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111,
            (0,0,110): 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122,
            (0,0,121): 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133,
            (0,0,132): 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144,
            (0,0,143): 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155,
```

MarkRivers, 2019-04-08:
I have merged the lz4_filter branches in both ADCore and ADSupport.  

ADCore can write blosc, lz4, and bslz4 files wtih either normal HDF5 library operations or direct chunk write.

These files can be read correctly using the dynamically loadable plugin libraries built in ADSupport.  Tested on both Windows and Linux.

MarkRivers, 2019-04-08:
Closing this issue.

closed 2019-04-08 by MarkRivers