areaDetector/ADCore #396: Segafault in NDAttribute plugin

hinxx, 2019-04-04:
With the latest code in master I can make simulator IOC to segfault if I use a single attribute, and 8 as seen in the example.

```
epics> auto_settings.sav: 83 of 83 PV's connected
[New Thread 0x7fff8e0e5700 (LWP 22945)]

Thread 196 "ATTR1_Plugin_1" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff8d107700 (LWP 22810)]
0x00007ffff739ee86 in paramVal::getStatus (this=0x580) at ../../asyn/asynPortDriver/paramVal.cpp:78
78	    return status_;
(gdb) bt
#0  0x00007ffff739ee86 in paramVal::getStatus (this=0x580) at ../../asyn/asynPortDriver/paramVal.cpp:78
#1  0x00007ffff73a0e75 in paramList::getStatus (this=0x7ffff6319b98 <main_arena+120>, index=43, status=0x7fff8d106b30) at ../../asyn/asynPortDriver/asynPortDriver.cpp:355
#2  0x00007ffff73a2a2d in asynPortDriver::getParamStatus (this=0x10b5930, list=1, index=43, paramStatus=0x7fff8d106b30) at ../../asyn/asynPortDriver/asynPortDriver.cpp:1016
#3  0x00007ffff73a6687 in asynPortDriver::doCallbacksGenericPointer (this=0x10b5930, genericPointer=0x7fff4c000950, reason=43, address=1) at ../../asyn/asynPortDriver/asynPortDriver.cpp:2715
#4  0x00007ffff78d6f5f in NDPluginAttribute::doTimeSeriesCallbacks (this=0x10b5930, pArray=0xe8cac0) at ../NDPluginAttribute.cpp:115
#5  0x00007ffff78d6e19 in NDPluginAttribute::processCallbacks (this=0x10b5930, pArray=0xe8cac0) at ../NDPluginAttribute.cpp:96
#6  0x00007ffff78d2016 in NDPluginDriver::processTask (this=0x10b5930) at ../NDPluginDriver.cpp:524
#7  0x00007ffff78d3832 in NDPluginDriver::run (this=0x10b5930) at ../NDPluginDriver.cpp:936
#8  0x00007ffff636a4e4 in epicsThreadCallEntryPoint (pPvt=0x10b1380) at ../../../src/libCom/osi/epicsThread.cpp:83
#9  0x00007ffff6373225 in start_routine (arg=0x10b0cf0) at ../../../src/libCom/osi/os/posix/osdThread.c:403
#10 0x00007ffff3b686ba in start_thread (arg=0x7fff8d107700) at pthread_create.c:333
#11 0x00007ffff605c41d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb) 

```

hinxx, 2019-04-04:
I think the problem is that TS plugin wants to access to NDArray in addr=1 and is is not there since the number of maximum addresses is 1, too. It should be 2.

MarkRivers, 2019-04-04:
I don't understand why you say the maximum number of addresses is 1.  It should be maxAttributes.

This is the beginning of the NDPluginAttribute constructor:
```
NDPluginAttribute::NDPluginAttribute(const char *portName, int queueSize, int blockingCallbacks,
                                     const char *NDArrayPort, int NDArrayAddr, int maxAttributes,
                                     int maxBuffers, size_t maxMemory,
                                     int priority, int stackSize)
    /* Invoke the base class constructor */
    : NDPluginDriver(portName, queueSize, blockingCallbacks,
                   NDArrayPort, NDArrayAddr, maxAttributes, maxBuffers, maxMemory,
                   asynInt32ArrayMask | asynFloat64Mask | asynFloat64ArrayMask | asynGenericPointerMask,
                   asynInt32ArrayMask | asynFloat64Mask | asynFloat64ArrayMask | asynGenericPointerMask,
                   ASYN_MULTIDEVICE, 1, priority, stackSize, 1)
```
So it calls the constructor for NDPluginDriver with maxAttributes as the 6'th argument.  This is the constructor for NDPluginDriver:
```
NDPluginDriver::NDPluginDriver(const char *portName, int queueSize, int blockingCallbacks,
                               const char *NDArrayPort, int NDArrayAddr, int maxAddr,
                               int maxBuffers, size_t maxMemory, int interfaceMask, int interruptMask,
                               int asynFlags, int autoConnect, int priority, int stackSize, int maxThreads,
                               bool compressionAware)

    : asynNDArrayDriver(portName, maxAddr, maxBuffers, maxMemory,
          interfaceMask | asynInt32Mask | asynFloat64Mask | asynOctetMask | asynInt32ArrayMask | asynDrvUserMask,
          interruptMask | asynInt32Mask | asynFloat64Mask | asynOctetMask | asynInt32ArrayMask,
          asynFlags, autoConnect, priority, stackSize),
```
So the 6'th argument is maxAddr which is passes to asynNDArrayDriver, which it then passes to asynPortDriver constructor.

When I run asynReport on ATTR1 I see that it indeed has 8 addresses for the simDetector when using EXAMPLE_commonPlugins.cmd
```epics> asynReport 1 ATTR1
ATTR1 multiDevice:Yes canBlock:No autoConnect:Yes
    enabled:Yes connected:Yes numberConnects 1
    nDevices 8 nQueued 0 blocked:No
    asynManagerLock:No synchronousLock:No
    exceptionActive:No exceptionUsers 0 exceptionNotifys 0
    traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
    addr 0 autoConnect Yes enabled Yes connected Yes exceptionActive No
        exceptionActive No exceptionUsers 1 exceptionNotifys 0
        blocked No
        traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
    addr 1 autoConnect Yes enabled Yes connected Yes exceptionActive No
        exceptionActive No exceptionUsers 0 exceptionNotifys 0
        blocked No
        traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
    addr 2 autoConnect Yes enabled Yes connected Yes exceptionActive No
        exceptionActive No exceptionUsers 0 exceptionNotifys 0
        blocked No
        traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
    addr 3 autoConnect Yes enabled Yes connected Yes exceptionActive No
        exceptionActive No exceptionUsers 0 exceptionNotifys 0
        blocked No
        traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
    addr 4 autoConnect Yes enabled Yes connected Yes exceptionActive No
        exceptionActive No exceptionUsers 0 exceptionNotifys 0
        blocked No
        traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
    addr 5 autoConnect Yes enabled Yes connected Yes exceptionActive No
        exceptionActive No exceptionUsers 0 exceptionNotifys 0
        blocked No
        traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
    addr 6 autoConnect Yes enabled Yes connected Yes exceptionActive No
        exceptionActive No exceptionUsers 0 exceptionNotifys 0
        blocked No
        traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
    addr 7 autoConnect Yes enabled Yes connected Yes exceptionActive No
        exceptionActive No exceptionUsers 0 exceptionNotifys 0
        blocked No
        traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
```

MarkRivers, 2019-04-04:
I should add that it does not crash for me, I can plot all 8 attributes with no problem.

MarkRivers, 2019-04-04:
OK, I think I see what you mean if I set maxAttributes=1.  Then only address 0 exists, not 1.  I changed my commonPlugins.cmd do set maxAttributes=1.  I then get the following errors:
```
2019/04/04 08:33:30.445 asynPortDriver:getParamAlarmSeverity: port=ATTR1 error setting parameter 43 in list 1, bad index
2019/04/04 08:33:30.455 asynPortDriver:getParamStatus: port=ATTR1 error setting parameter 43 in list 1, bad index
2019/04/04 08:33:30.455 asynPortDriver:getParamAlarmStatus: port=ATTR1 error setting parameter 43 in list 1, bad index
2019/04/04 08:33:30.455 asynPortDriver:getParamAlarmSeverity: port=ATTR1 error setting parameter 43 in list 1, bad index
2019/04/04 08:33:30.465 asynPortDriver:getParamStatus: port=ATTR1 error setting parameter 43 in list 1, bad index
```
I does not crash for me.  This may be because of some fix to asynPortDriver to prevent the access violation.  What version of asyn are you using?  I am using the master branch, which is the same as R4-35.

I think your original message says that it was also crashing when you used 8 attributes.  I don't understand that.  Is is true?

I have pushed a fix to master that forces maxAddr to be at least 2.  With that fix setting maxAttributes=1 in commonPlugins.cmd no longer generates error messages and works OK.

hinxx, 2019-04-05:
> OK, I think I see what you mean if I set maxAttributes=1. 

Yes, that's what I meant.

>  What version of asyn are you using? I am using the master branch, which is the same as R4-35.

I'm still at R4-33. I'll upgrade to R4-35.

> I think your original message says that it was also crashing when you used 8 attributes. I don't understand that. Is is true?

No that was not meant, and not happening. 8 attributes work just fine.

> I have pushed a fix to master that forces maxAddr to be at least 2. 

Thank you!

MarkRivers, 2019-04-08:
I believe this is fixed so I am closing the issue.  @hinxx if you find a problem please re-open the issue.

closed 2019-04-08 by MarkRivers