areaDetector/ADCore #398: HDF5 plugin: failed to create dataset

ulrikpedersen, 2019-04-09:
As reported by @hinxx in https://github.com/areaDetector/ADCore/pull/397#commitcomment-33082264:

Messages `Failed to create dataset: timestamp. Continuing to next.` when using default built-in XML layout should not happen.

MarkRivers, 2019-04-09:
I did a number of tests with the default built-in XML layout yesterday and did not see those messages.  Under what circumstances did they occur?

hinxx, 2019-04-09:
> Under what circumstances did they occur?

I've seen them if I just replaced the `asynPrint(this->pasynUserSelf, ASYN_TRACE_WARNING,` in the next line with `printf(`.

MarkRivers, 2019-04-09:
I just enabled ASYN_TRACE_WARNING in the HDF5 plugin.  Indeed I see this message once for each file created:
```
2019/04/09 13:56:16.512 NDFileHDF5::createTree Failed to create dataset: timestamp. Continuing to next.
```
@ulrikpedersen I see comments in NDFileHDF5.cpp from you about creating this dataset, which is in the performance group?  Can you take a look to see what we are getting this error?

ajgdls, 2019-05-14:
The warning here is being printed due to the fact that the performance/timestamp dataset is a bit hardcoded into the system.  It shows up in the structure but specifically does not get created at this point in the code.  There is a comment that states only detector and constant datasets are created, which is why the warning is raised.
The timestamp dataset is then created further on and still gets saved into the file as expected.  I think the solution is to check specifically for the performance/timestamp dataset and do not attempt to create it at this point, which will stop the warning from being generated.  I'll raise a PR for people to comment on.

MarkRivers, 2019-05-19:
Closed via #408

closed 2019-05-19 by MarkRivers