areaDetector/ADCore #401: Problem with chunking on NDArrays with >2 dimensions

MarkRivers, 2019-04-20:
I just added a JPEG filter for HDF5.  It worked fine for 2-D mono images.  However, it was crashing when trying to compress 3-D arrays with colorMode=RGB1.  These arrays have dimensions [3,1024,1024], for example.

The problem was that the jpeg compression filter was being called with nbytes=3072, rather than the full NDArray which is 3 MB.  The reason is that NDFileHDF5 does not behave sensibly in chosing the chunking when pArrays->ndims > 2.  It was defaulting to [1024, 3, 1] for Rows, Columns, and Frames per chunk.  This resulted in only 3KB per chunk, rather than 3MB.  I was able to make it work by setting FramesPerChunk to 1024.  Even though I was only running in Single Mode that passed the full NDArray to the compression filter.  [1024, 3, 1024] also works in Stream mode.

Something is wrong in the chunking logic.

MarkRivers, 2019-04-20:
I have committed the new JPEG filter code to the master branch in both ADSupport and ADCore.  It basically works with the exception of needing to set the chunking in a non-intuitive way.

MarkRivers, 2019-04-20:
The JPEG filter only works with 8-bit images with NDColorModeMono (2-D) or NDColorModeRGB1 (3-D where first dimension is 3).

MarkRivers, 2019-04-21:
If I have an RGB image [3,1024,1024] I can write it with the JPEG compression done in the HDF5 plugin if I set NumRowsChunks=1024, NumColsChunks=3, NumFramesChunks=1024.  However, if I try to use Direct Chunk Write by sending it JPEG compressed arrays from NDPluginCodec it fails with the following message in Stream mode:
```
2019/04/21 08:31:48.075 NDFileHDF5Dataset::writeFile ERROR Unable to write pre-compressed data - mismatched chunk definition
2019/04/21 08:31:48.075 NDFileHDF5::writeFile ERROR: could not write to dataset. Aborting
2019/04/21 08:31:48.094 NDPluginFile::writeFileBase Error writing file, status=3
```
If I try to use DirectChunkWrite in Single mode and just save 1 precompressed NDArray I get an access violation.
```
(gdb) bt
#0  0x00007efc49f39d5b in std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::string const&) () from /lib64/libstdc++.so.6
#1  0x000000000064047a in NDFileHDF5::createHardLinks (this=this@entry=0x22b1950, root=0x7efc1c19ec30) at ../NDFileHDF5.cpp:727
#2  0x0000000000640571 in NDFileHDF5::createHardLinks (this=this@entry=0x22b1950, root=0x7efc1c19d880) at ../NDFileHDF5.cpp:741
#3  0x00000000006405d1 in NDFileHDF5::createHardLinks (this=this@entry=0x22b1950, root=0x7efc1c19d790) at ../NDFileHDF5.cpp:720
#4  0x000000000064e6c1 in NDFileHDF5::openFile (this=0x22b1950, fileName=0x7efc45f826a0 "/home/epics/scratch/test_hdf5_normal_jpeg_rgb1_49.h5", openMode=<optimized out>, pArray=
    0x7efb8c001420) at ../NDFileHDF5.cpp:316
#5  0x000000000066c45d in NDPluginFile::openFileBase (this=this@entry=0x22b1950, openMode=openMode@entry=2, pArray=pArray@entry=0x7efb8c001420) at ../NDPluginFile.cpp:73
#6  0x000000000066d014 in NDPluginFile::writeFileBase (this=this@entry=0x22b1950) at ../NDPluginFile.cpp:254
#7  0x000000000066dc56 in NDPluginFile::writeInt32 (this=this@entry=0x22b1950, pasynUser=pasynUser@entry=0x374ada8, value=value@entry=1) at ../NDPluginFile.cpp:818
#8  0x000000000064496e in NDFileHDF5::writeInt32 (this=0x22b1950, pasynUser=0x374ada8, value=1) at ../NDFileHDF5.cpp:2031
#9  0x000000000114d3c2 in writeInt32 (drvPvt=0x22b1950, pasynUser=0x374ada8, value=1) at ../../asyn/asynPortDriver/asynPortDriver.cpp:1715
#10 0x000000000118dfa5 in processCallbackOutput (pasynUser=0x374ada8) at ../devBusyAsyn.c:315
#11 0x0000000001144763 in portThread (pport=0x22b2310) at ../../asyn/asynDriver/asynManager.c:902
#12 0x00000000012aad2c in start_routine (arg=0x2276fe0) at ../../src/osi/os/posix/osdThread.c:403
#13 0x00007efc4a7dbe25 in start_thread () from /lib64/libpthread.so.0
#14 0x00007efc49694bad in clone () from /lib64/libc.so.6
(gdb)
```

It seems to me that the chunking definitions is not general enough. It is oriented around 2-D data (user defined chunking only supports NumRowsChunks, NumColsChunks, and NumFramesChunks.  The HDF5 plugin should be fully N-dimensional, since it can be used for any type of NDArray, including color images, spectroscopy data from the Xspress3, etc.  I think the following may make more sense:
- Flag ChunkingAuto=Yes/No.  If Yes then it uses the actual rank and dimensions of the NDArray to define the chunking.  If No then it uses the user-defined chunking which should be renamed to Dim0ChunkSize, Dim1ChunkSize. Dim2ChunkSize, etc.

I find the following code on line 192 of NDFileHDF5Dataset.cpp:
```
  // Check dimensions of data
  if (pArray->ndims > 2) {
    asynPrint(this->pAsynUser_, ASYN_TRACE_FLOW,
              "Data must be 1D or 2D to use direct chunk write\n");
    return asynError;
  }
```
Why was this restriction imposed?  It prevents me from using Direct Chunk Write with RGB data.  By having that message be ASYN_TRACE_FLOW it also makes it unlikely to be seen by the user.  If I enable ASYN_TRACE_FLOW and try to Stream 10 NDArrays that are precompressed I get the following output:
```
2019/04/21 08:44:52.783 asynManager::portThread port=FileHDF1 callback
2019/04/21 08:44:52.783 NDFileHDF5:writeInt32: function=35, value=1 old=0
2019/04/21 08:44:52.783 NDFileHDF5:writeInt32: calling base class function=35, value=1 old=0
2019/04/21 08:44:52.783 NDFileHDF5::openFile Filename: /home/epics/scratch/test_hdf5_normal_jpeg_rgb1_49.h5
2019/04/21 08:44:52.785 NDFileHDF5::openFile attribute list copied. num pArray attributes = 22 local copy = 18
2019/04/21 08:44:52.785 NDFileHDF5::configureDims  NDArray:   {     3, 1024, 1024, }
2019/04/21 08:44:52.785 NDFileHDF5::configureDims dimension report:
 framesize      1,  1024,  1024,     3,
 chunkdims      1,  1024,  1024,     3,
   maxdims      1,  1024,  1024,     3,
      dims      1,  1024,  1024,     3,
    offset      0,     0,     0,     0,
   virtual      1,
2019/04/21 08:44:52.785 NDFileHDF5::createNewFile Calculated istorek=0
2019/04/21 08:44:52.786 NDFileHDF5::createFileLayout Creating dataspace with given dimensions
2019/04/21 08:44:52.786 NDFileHDF5::createFileLayout Configuring chunking
2019/04/21 08:44:52.786 NDFileHDF5::configureCompression Overriding compression settings from pre-compressed NDArray
2019/04/21 08:44:52.786 NDFileHDF5::createFileLayout Setting fillvalue
2019/04/21 08:44:52.787 NDFileHDF5::createFileLayout Empty layout file, use default layout
2019/04/21 08:44:52.788 NDFileHDF5::createXMLFileLayout Root tree: < Group: '' groups=1 dsets=0 attr=0>
                                  < Group: '/entry' groups=2 dsets=0 attr=1>
                                  < Group: '/entry/data' groups=0 dsets=0 attr=1>
[... cut ...]
2019/04/21 08:56:14.363 NDFileHDF5Dataset::writeFile ERROR Unable to write pre-compressed data - mismatched chunk definition
2019/04/21 08:56:14.363 NDFileHDF5::writeFile ERROR: could not write to dataset. Aborting
2019/04/21 08:56:14.364 NDPluginFile::writeFileBase Error writing file, status=3
2019/04/21 08:56:15.336 FileHDF1 addr -1 queueRequest priority 0 not lockHolder
```
In this case I do get the error about data must be 1D or 2D, and it does not crash.  Why does it crash with JPEG and not with Blosc?

MarkRivers, 2019-04-22:
I have added JPEGQuality to NDFileHDF5 and updated .template, .req, and all OPI screens.  Everything is pushed to master branch on Github if others want to test.

GDYendell, 2019-04-23:
The reason I limited it to 2D data in the first instance was because the `user_chunking` is 3D. It could work for higher dimensions in lazy open if we take the dimensions from the `NDArray` and assume full chunking. In the same way `createFileLayout` calls `configureCompression` with the `NDArray` to override any settings. To work without lazy open the user needs to be able to define the higher level dimensions. Could it be taken from the ExtraDims?

GDYendell, 2019-04-23:
I'm not sure I understand why this works.

> If I have an RGB image [3,1024,1024] I can write it with the JPEG compression done in the HDF5 plugin if I set NumRowsChunks=1024, NumColsChunks=3, NumFramesChunks=1024.

I would think the chunking should be [3, 1024, 1024, 1]. What is the h5dump output for the dataset?

MarkRivers, 2019-04-23:
> I would think the chunking should be [3, 1024, 1024, 1]. I guess it is because it is single mode. What is the h5dump output for the dataset?

No, I think NumColChunks is the first dimension of the array.  

Here I have set NumColChunks, NumRowChunks, and NumFramesChunks all to 1024 before saving a file.  One of them will be shortened to 3 when the file is saved.
```
corvette:~/devel/areaDetector>caget 13SIM1:HDF1:ArraySize0_RBV 13SIM1:HDF1:ArraySize1_RBV 13SIM1:HDF1:ArraySize2_RBV 13SIM1:HDF1:NumRowChunks_RBV 13SIM1:HDF1:NumColChunks_RBV 13SIM1:HDF1:NumFramesChunks_RBV
13SIM1:HDF1:ArraySize0_RBV     3
13SIM1:HDF1:ArraySize1_RBV     1024
13SIM1:HDF1:ArraySize2_RBV     1024
13SIM1:HDF1:NumRowChunks_RBV   1024
13SIM1:HDF1:NumColChunks_RBV   1024
13SIM1:HDF1:NumFramesChunks_RBV 1024
```
Now I save a file in Stream mode, NumCapture=10.  I then read the chunking RBV values again:
```
corvette:~/devel/areaDetector>caget 13SIM1:HDF1:ArraySize0_RBV 13SIM1:HDF1:ArraySize1_RBV 13SIM1:HDF1:ArraySize2_RBV 13SIM1:HDF1:NumRowChunks_RBV 13SIM1:HDF1:NumColChunks_RBV 13SIM1:HDF1:NumFramesChunks_RBV
13SIM1:HDF1:ArraySize0_RBV     3
13SIM1:HDF1:ArraySize1_RBV     1024
13SIM1:HDF1:ArraySize2_RBV     1024
13SIM1:HDF1:NumRowChunks_RBV   1024
13SIM1:HDF1:NumColChunks_RBV   3
13SIM1:HDF1:NumFramesChunks_RBV 1024
```
This is what h5dump says about the file:
```
corvette:~/devel/areaDetector>h5dump --properties ~/scratch/test_hdf5_normal_jpeg_rgb1_84.h5
HDF5 "/home/epics/scratch/test_hdf5_normal_jpeg_rgb1_84.h5" {
GROUP "/" {
   GROUP "entry" {
      ATTRIBUTE "NX_class" {
         DATATYPE  H5T_STRING {
            STRSIZE 7;
            STRPAD H5T_STR_NULLTERM;
            CSET H5T_CSET_ASCII;
            CTYPE H5T_C_S1;
         }
         DATASPACE  SCALAR
         DATA {
         (0): "NXentry"
         }
      }
      GROUP "data" {
         ATTRIBUTE "NX_class" {
            DATATYPE  H5T_STRING {
               STRSIZE 6;
               STRPAD H5T_STR_NULLTERM;
               CSET H5T_CSET_ASCII;
               CTYPE H5T_C_S1;
            }
            DATASPACE  SCALAR
            DATA {
            (0): "NXdata"
            }
         }
         DATASET "data" {
            DATATYPE  H5T_STD_U8LE
            DATASPACE  SIMPLE { ( 10, 1024, 1024, 3 ) / ( 10, 1024, 1024, 3 ) }
            STORAGE_LAYOUT {
               CHUNKED ( 1, 1024, 1024, 3 )
               SIZE 2209392 (14.238:1 COMPRESSION)
            }
            FILTERS {
               USER_DEFINED_FILTER {
                  FILTER_ID 32019
                  COMMENT jpeg; see https://github.com/UChicago-CARS/jpegHDF5
                  PARAMS { 88 1024 1024 1 }
               }
            }
            FILLVALUE {
               FILL_TIME H5D_FILL_TIME_IFSET
               VALUE  0
            }
            ALLOCATION_TIME {
               H5D_ALLOC_TIME_INCR
            }
            DATA {
            (0,0,0,0): 244, 234, 225,
            (0,0,1,0): 246, 237, 228,
            (0,0,2,0): 247, 238, 229,
            (0,0,3,0): 238, 229, 222,
            (0,0,4,0): 252, 245, 237,
            (0,0,5,0): 249, 242, 236,
            (0,0,6,0): 246, 241, 237,
            (0,0,7,0): 251, 248, 243,
```

MarkRivers, 2019-04-23:
I just made the following temporary change to NDFileHDF5Dataset::verifyChunking.  It basically completely disables verifyChunking.
```
 asynStatus NDFileHDF5Dataset::verifyChunking(NDArray *pArray)
 {
   // If compression is enabled, it must match our configuration
   if (!pArray->codec.empty() || !this->codec.empty()) {
     if (pArray->codec != this->codec) {
       asynPrint(this->pAsynUser_, ASYN_TRACE_FLOW,
                 "Dataset codec '%s' [%d, %d, %d] does not match NDArray codec '%s' [%d, %d, %d]\n",
                 this->codec.name.c_str(), this->codec.level, this->codec.shuffle, this->codec.compressor,
                 pArray->codec.name.c_str(), pArray->codec.level, pArray->codec.shuffle, pArray->codec.compressor);
       return asynError;
     }
   }
+return asynSuccess;
   // Check dimensions of data
   if (pArray->ndims > 2) {
     asynPrint(this->pAsynUser_, ASYN_TRACE_FLOW,
               "Data must be 1D or 2D to use direct chunk write\n");
     return asynError;
   }
   bool mismatch = false;
```
Once I do that then I am able to Stream precompressed [3,1024,1024] NDArrays written with Direct Chunk Write with no problem.

Here is the h5dump output on such a file.
```
corvette:~/devel/areaDetector>h5dump --properties ~/scratch/test_hdf5_direct_chunk_jpeg_rgb1_90.h5 | more
HDF5 "/home/epics/scratch/test_hdf5_direct_chunk_jpeg_rgb1_90.h5" {
GROUP "/" {
   GROUP "entry" {
      ATTRIBUTE "NX_class" {
         DATATYPE  H5T_STRING {
            STRSIZE 7;
            STRPAD H5T_STR_NULLTERM;
            CSET H5T_CSET_ASCII;
            CTYPE H5T_C_S1;
         }
         DATASPACE  SCALAR
         DATA {
         (0): "NXentry"
         }
      }
      GROUP "data" {
         ATTRIBUTE "NX_class" {
            DATATYPE  H5T_STRING {
               STRSIZE 6;
               STRPAD H5T_STR_NULLTERM;
               CSET H5T_CSET_ASCII;
               CTYPE H5T_C_S1;
            }
            DATASPACE  SCALAR
            DATA {
            (0): "NXdata"
            }
         }
         DATASET "data" {
            DATATYPE  H5T_STD_U8LE
            DATASPACE  SIMPLE { ( 10, 1024, 1024, 3 ) / ( 10, 1024, 1024, 3 ) }
            STORAGE_LAYOUT {
               CHUNKED ( 1, 1024, 1024, 3 )
               SIZE 3492550 (9.007:1 COMPRESSION)
            }
            FILTERS {
               USER_DEFINED_FILTER {
                  FILTER_ID 32019
                  COMMENT jpeg; see https://github.com/UChicago-CARS/jpegHDF5
                  PARAMS { 88 1024 1024 1 }
               }
            }
            FILLVALUE {
               FILL_TIME H5D_FILL_TIME_IFSET
               VALUE  0
            }
            ALLOCATION_TIME {
               H5D_ALLOC_TIME_INCR
            }
            DATA {
            (0,0,0,0): 238, 222, 206,
            (0,0,1,0): 240, 224, 208,
            (0,0,2,0): 242, 226, 211,
            (0,0,3,0): 242, 228, 215,
            (0,0,4,0): 243, 229, 218,
```

verifyChunking() is making assumptions about the rank of the NDArray that it should not be making.  It does not allow NDArrays with more than 2 dimensions.  This is not correct, NDFileHDF1 should correctly handle NDArrays of any rank.  Streaming RGB1 compressed video is definitely high on our user requirements list, but there are other types of detectors that may produce NDArrays with rank>2.

GDYendell, 2019-04-23:
Effectively NumRowsChunks, NumColsChunks, NumFramesChunks mean different things in Single Mode. You can use the Frames as a real dimension to the image data. When I added this it seemed like the existing API already restricted it to 3D (N 2D images) because there are only 3 chunk parameters.

> `CHUNKED ( 1, 1024, 1024, 3 )`

How does it get these chunking parameters? Have you got 1024, 1024, 3 as Frames, Cols, Rows and then the 1 is added by the driver? Will it just keep adding 1s to the outer chunk dimension to match the NDArray rank + 1? What if the image dimensions were [1024, 1024, 3, 255] (e.g. RGBA)?

MarkRivers, 2019-04-23:
> When I added this it seemed like the existing API already restricted it to 3D (N 2D images) because there are only 3 chunk parameters.

I think there are only 3 user chunk parameters, but it actually sets the chunking to the rank of the array or the rank + 1.

> How does it get these chunking parameters? Have you got 1024, 1024, 3 as Frames, Cols, Rows and then the 1 is added by the driver?

Yes, I think so. 

> What if the image dimensions were [1024, 1024, 3, 255] (e.g. RGBA)?

Actually I think RGBA is [1024, 1024, 4].  It just adds one extra value for each pixel.

GDYendell, 2019-04-23:
Bad example. I don't know of anything specific, but if there is a 4D image this wouldn't work because it would need to define 4 specific values, not just add an extra 1 to the outer dimension. You can cheat and get it to work for 3D because you just want an extra 1 (unless you want to chunk by multiple frames), but any more than that and you need more inputs. There has to be some kind of limit to the dimensionality and currently it is 2 (or arguably 3).

MarkRivers, 2019-04-23:
I have found another behavior of the chunking in NDFileHDF5 that I think is simply incorrect.  The problem is that it will automatically decrease the size of NumRowChunks or NumColChunks, but it never automatically increases it.  This leads to highly inefficient chunking unless the user manually fixes the problem.  Users should *not* have to worry about chunking by default.

Here is a simple example.  simDetector producing 1024x1024 mono frames.  Set chunking to [1024,1024, 1]. Save one frame in single mode.  Run h5dump:
```
corvette:~/scratch5dump --properties test_hdf5_mono_single_1024_before_94.h5 | more
HDF5 "test_hdf5_mono_single_1024_before_94.h5" {
GROUP "/" {
   GROUP "entry" {
      ATTRIBUTE "NX_class" {
         DATATYPE  H5T_STRING {
            STRSIZE 7;
            STRPAD H5T_STR_NULLTERM;
            CSET H5T_CSET_ASCII;
            CTYPE H5T_C_S1;
         }
         DATASPACE  SCALAR
         DATA {
         (0): "NXentry"
         }
      }
      GROUP "data" {
         ATTRIBUTE "NX_class" {
            DATATYPE  H5T_STRING {
               STRSIZE 6;
               STRPAD H5T_STR_NULLTERM;
               CSET H5T_CSET_ASCII;
               CTYPE H5T_C_S1;
            }
            DATASPACE  SCALAR
            DATA {
            (0): "NXdata"
            }
         }
         DATASET "data" {
            DATATYPE  H5T_STD_U8LE
            DATASPACE  SIMPLE { ( 1024, 1024 ) / ( 1024, 1024 ) }
            STORAGE_LAYOUT {
               CHUNKED ( 1024, 1024 )
               SIZE 1048576
            }
            FILTERS {
               NONE
            }
            FILLVALUE {
               FILL_TIME H5D_FILL_TIME_IFSET
               VALUE  0
            }
            ALLOCATION_TIME {
```
So the chunking is [1024,1024] as expected.

Now change binning to 2x2 in driver, so frames are 512x512. 
```
corvette:~/scratch>h5dump --properties test_hdf5_mono_single_512_95.h5 | more
HDF5 "test_hdf5_mono_single_512_95.h5" {
GROUP "/" {
   GROUP "entry" {
      ATTRIBUTE "NX_class" {
         DATATYPE  H5T_STRING {
            STRSIZE 7;
            STRPAD H5T_STR_NULLTERM;
            CSET H5T_CSET_ASCII;
            CTYPE H5T_C_S1;
         }
         DATASPACE  SCALAR
         DATA {
         (0): "NXentry"
         }
      }
      GROUP "data" {
         ATTRIBUTE "NX_class" {
            DATATYPE  H5T_STRING {
               STRSIZE 6;
               STRPAD H5T_STR_NULLTERM;
               CSET H5T_CSET_ASCII;
               CTYPE H5T_C_S1;
            }
            DATASPACE  SCALAR
            DATA {
            (0): "NXdata"
            }
         }
         DATASET "data" {
            DATATYPE  H5T_STD_U8LE
            DATASPACE  SIMPLE { ( 512, 512 ) / ( 512, 512 ) }
            STORAGE_LAYOUT {
               CHUNKED ( 512, 512 )
               SIZE 262144
            }
            FILTERS {
               NONE
            }
            FILLVALUE {
               FILL_TIME H5D_FILL_TIME_IFSET
               VALUE  0
            }
            ALLOCATION_TIME {
```
The chunking was automatically changed to [512,512] as expected.

Now change the binning back to 1x1.
```
corvette:~/scratch>h5dump --properties test_hdf5_mono_single_1024_after_9
test_hdf5_mono_single_1024_after_93.h5  test_hdf5_mono_single_1024_after_96.h5
corvette:~/scratch>h5dump --properties test_hdf5_mono_single_1024_after_96.h5 | more
HDF5 "test_hdf5_mono_single_1024_after_96.h5" {
GROUP "/" {
   GROUP "entry" {
      ATTRIBUTE "NX_class" {
         DATATYPE  H5T_STRING {
            STRSIZE 7;
            STRPAD H5T_STR_NULLTERM;
            CSET H5T_CSET_ASCII;
            CTYPE H5T_C_S1;
         }
         DATASPACE  SCALAR
         DATA {
         (0): "NXentry"
         }
      }
      GROUP "data" {
         ATTRIBUTE "NX_class" {
            DATATYPE  H5T_STRING {
               STRSIZE 6;
               STRPAD H5T_STR_NULLTERM;
               CSET H5T_CSET_ASCII;
               CTYPE H5T_C_S1;
            }
            DATASPACE  SCALAR
            DATA {
            (0): "NXdata"
            }
         }
         DATASET "data" {
            DATATYPE  H5T_STD_U8LE
            DATASPACE  SIMPLE { ( 1024, 1024 ) / ( 1024, 1024 ) }
            STORAGE_LAYOUT {
               CHUNKED ( 512, 512 )
               SIZE 1048576
            }
            FILTERS {
               NONE
            }
            FILLVALUE {
               FILL_TIME H5D_FILL_TIME_IFSET
               VALUE  0
            }
            ALLOCATION_TIME {
```
The chunking is stuck at [512,512] which is inefficient.  If we had been switching between RGB1 and Mono the situation would be much worse, because NumColChunks would be stuck at 3 rather than 1024!

I think there should be a mode switch ChunkingMode=Auto/Manual.  In Auto mode the chunking is automatically set to a single complete NDArray, i.e. NumFramesChunks=1, and for each dimension up to rank-1 the chunking should be the size of the NDArray in that dimension.  This is probably the desired behavior 99% of the time?

MarkRivers, 2019-04-23:
> There has to be some kind of limit to the dimensionality and currently it is 2 (or arguably 3).

Why does there have to be a limit to the dimensionality?  See my comment above.  I think there should an Auto chunking mode that always sets the chunking to a single complete NDArray.  For manual chunking there should be NumFramesChunks, NumColChunks (or Dim0Chunk), NumRowChunks (or Dim1Chunk), Dim2Chunk, ... Dim9Chunk,  since ND_ARRAY_MAX_DIMS=10.

GDYendell, 2019-04-23:
I agree. Increasing the manual chunking parameters to 10 would cover most use cases and making them numbered instead of implying an incorrect meaning is much clearer. Auto chunking (to higher _and_ lower values) would be nice, but presumably that would only work with lazy open, as we need to call set_chunk before creating the dataset.

MarkRivers, 2019-04-23:
> Auto chunking (to higher and lower values) would be nice, but presumably that would only work with lazy open, as we need to call set_chunk before creating the dataset.

I don't think it requires lazy open, it just requires that the previous frame that the plugin received has the same structure as the ones it will receive when acquisition starts.  That is already required when lazy open is not used, independent of chunking, because the dimensions of the array and the attributes must not change between file creation and receipt of the arrays.

MarkRivers, 2019-05-11:
Closed via #404

closed 2019-05-11 by MarkRivers