areaDetector/ADCore #403: Allow HDF5 plugin to append new NDArrays to existing HDF5 archive

KedoKudo, 2019-04-24:
The current HDF5 plugin in the ADCore does not seem to support __appending__ new group/dataset to existing HDF5 archive as it will always overwrite the whole file if the file exists.  Given that there are some experiments with multiple scans that are better separated into different groups, it would be really helpful to add a new option in the HDF plugin to allow adding new group/dataset to existing HDF5 archive.

MarkRivers, 2019-04-24:
That would be a major change.  I would be happy to look at a pull request if you want to work on it.

GDYendell, 2019-04-24:
Is it possible to achieve what you want with a VDS? You can write separate files and then combine them so that it looks it was all written to the same file.

https://support.hdfgroup.org/HDF5/docNewFeatures/NewFeaturesVirtualDatasetDocs.html

KedoKudo, 2019-04-24:
That is the workaround solution we are considering to use for the short term.  However, having a single HDF5 archive as the direct output from the area detector is much better when it comes to managing the experiment data.