areaDetector/ADCore #419: ADCore Template includes not working

toomanycats, 2019-09-17:
Hello, 
I've recently built synApps 6.1 with EPICS R3.15.4 and AD 3.7 for CentOS 7 64 bit.
I'm porting over an older IOC written for AD2.0 and EPICS R3.14.12.

I'm hitting a run time error where the templates in ADCore have an include that's not being found on the path. I can hack it with 
    addpath "/usr/local/epics/R3.15.4/modules/soft/synApps_6_1/support/areaDetector    -R3-7/ADCore/ADApp/Db"

at the top of files such as `ADCore/ADApp/Db/ADBase.template`.

I'm assuming there is a better way and I'm just not seeing it.
Cheers

MarkRivers, 2019-09-18:
What tool are you using, msi or dbLoadRecords? The example IOCs all set the DB_FILE_PATH environment variable.

toomanycats, 2019-09-18:
Hi Mark, I'm using dbLoadRecords. I went looking for the examples but must have missed them.
Grepping for DB_FILE_PATH in the source tree isn't turning up anything.

MarkRivers, 2019-09-18:
Sorry, I was doing that from memory.

The correct environment variable is EPICS_DB_INCLUDE_PATH. Look at st_base.cmd in ADSimdetector.

toomanycats, 2019-09-18:
thanks for getting back to me...doh the readme had that in ADCore, which read at least once...

MarkRivers, 2019-09-18:
Is it working now and we can close this issue?

closed 2019-09-20 by toomanycats