areaDetector/ADCore #420: Debug assertion in NDFileHDFLayoutXML.cpp, when built with debug DLL configuration

pheest, 2019-10-23:
This problem appears to be caused by the lines
// Also a problem in Visual Studio.  The warning is:
// warning LNK4217: locally defined symbol xmlFree imported in function "private: int __cdecl hdf5::LayoutXML::parse_root(void)" 
#ifdef _MSC_VER
  #define xmlFree free
#endif

I think the assertion signifies that memory is allocated in one DLL (through xmlMalloc), but is freed in another (through free directly).

I commented out the preprocessor override, and the issue went away.
I believe it can be removed.

I can't say I've noticed the locally defined symbol warning.

MarkRivers, 2019-10-23:
What have you tested building?
- VS version?
- Static/dynamic?
- Debug/Release?

I was definitely getting those warnings. I believe they occurred with VS2015, Static Release build.

pheest, 2019-10-25:
Hi Mark,
I'm using VS2017.
I've built debug and release /x86 dynamic.
[areaDetector.win32-x86.log](https://github.com/areaDetector/ADCore/files/3771921/areaDetector.win32-x86.log)
[areaDetector.win32-x86-debug.log](https://github.com/areaDetector/ADCore/files/3771922/areaDetector.win32-x86-debug.log)

I don't see any "locally defined symbol" warnings here.

pheest, 2019-10-28:
OK, the linker warning is present in static builds.
The fix is to add something like:
ifneq ($(SHARED_LIBRARIES),YES)
  USR_CFLAGS_WIN32 += -DLIBXML_STATIC
  USR_CXXFLAGS_WIN32 += -DLIBXML_STATIC
endif
to the pluginSrc makefile.

pheest, 2019-11-18:
My pull request includes this.
But I left the useage for migw 'as was' since I don't have the capability for a build within that environment.
I would encorage you to verify whether adding -DLIBXML_STATIC correctly fixes the issue for that environment also.

BenBradnick, 2020-04-16:
The removal of:
```
#ifdef _MSC_VER
  #define xmlFree free
#endif
```
results in the following linker error when we try and link against the library in VS2015: `NDPlugin.lib(NDFileHDF5LayoutXML.obj) : error LNK2001: unresolved external symbol xmlFree`

MarkRivers, 2020-04-16:
@BenBradnick what architecture are you building for when you see that error message?

BenBradnick, 2020-04-16:
@MarkRivers this is for windows-x64.

MarkRivers, 2020-04-16:
What version of ADCore?

MarkRivers, 2020-04-16:
I just built the latest versions of ADCore and ADSimDetector for windows-x64 on Visual Studio 2015 with no problems.
```
Finished generating code
"Installing created executable ../../../bin/windows-x64/simDetectorApp.exe"
make[5]: Leaving directory 'J:/epics/devel/areaDetector-3-9/ADSimDetector/iocs/simDetectorIOC/simDetectorApp/src/O.windows-x64'
make[4]: Leaving directory 'J:/epics/devel/areaDetector-3-9/ADSimDetector/iocs/simDetectorIOC/simDetectorApp/src'
make[3]: Leaving directory 'J:/epics/devel/areaDetector-3-9/ADSimDetector/iocs/simDetectorIOC/simDetectorApp'
```

BenBradnick, 2020-04-16:
I am also using ADCore 3-9 (albeit the Diamond-equivalent branch). ADCore builds fine but an IOC built using ADPcoWin does not work. I will try and build the master release and see if it's our branch causing the problem

MarkRivers, 2020-04-16:
I suggest first building ADSimDetector with your current branch.  If that works then the problem must be ADPcoWin.

BenBradnick, 2020-04-16:
Hi @MarkRivers - I have tried building a Windows ADSimDetector using our branch and I get the same problem. This is using ADSupport 1-9 with XML2_EXTERNAL = NO in CONFIG_SITE. Linux-x86_64 is building but we are using an external XML2 library in that case.
```
make[2]: Entering directory 'H:/Development/local/support/ADSimDetector/iocs/winSimDetector/winSimDetectorApp/src/O.windows-x64'
link -nologo  -LTCG -incremental:no -opt:ref -release  -MACHINE:X64              -out:winSimDetector.exe        winSimDetector_registerRecordDeviceDri
ver.obj winSimDetectorMain.obj     W:\\prod\\R3.14.12.7\\support\\calc\\3-7-3\\lib\\windows-x64\\calc.lib  ..\\..\\..\\lib\\windows-x64\\simDetector.l
ib  W:\\prod\\R3.14.12.7\\support\\ADCore\\3-9dls2\\lib\\windows-x64\\NDPlugin.lib  W:\\prod\\R3.14.12.7\\support\\ADCore\\3-9dls2\\lib\\windows-x64\\
ADBase.lib  W:\\prod\\R3.14.12.7\\support\\ADSupport\\1-9dls2\\lib\\windows-x64\\tiff.lib  W:\\prod\\R3.14.12.7\\support\\ADSupport\\1-9dls2\\lib\\win
dows-x64\\jpeg.lib  W:\\prod\\R3.14.12.7\\support\\ADSupport\\1-9dls2\\lib\\windows-x64\\blosc.lib  W:\\prod\\R3.14.12.7\\support\\ADSupport\\1-9dls2\
\lib\\windows-x64\\xml2.lib  W:\\prod\\R3.14.12.7\\support\\ADSupport\\1-9dls2\\lib\\windows-x64\\hdf5_hl.lib  W:\\prod\\R3.14.12.7\\support\\ADSuppor
t\\1-9dls2\\lib\\windows-x64\\hdf5.lib  W:\\prod\\R3.14.12.7\\support\\ADSupport\\1-9dls2\\lib\\windows-x64\\szip.lib  W:\\prod\\R3.14.12.7\\support\\
ADSupport\\1-9dls2\\lib\\windows-x64\\zlib.lib  W:\\prod\\R3.14.12.7\\support\\busy\\1-7-2dls3\\lib\\windows-x64\\busy.lib  W:\\prod\\R3.14.12.7\\supp
ort\\asyn\\4-39\\lib\\windows-x64\\asyn.lib  W:\\epics\\R3.14.12.7\\base\\lib\\windows-x64\\recIoc.lib  W:\\epics\\R3.14.12.7\\base\\lib\\windows-x64\
\softDevIoc.lib  W:\\epics\\R3.14.12.7\\base\\lib\\windows-x64\\miscIoc.lib  W:\\epics\\R3.14.12.7\\base\\lib\\windows-x64\\rsrvIoc.lib  W:\\epics\\R3
.14.12.7\\base\\lib\\windows-x64\\dbtoolsIoc.lib  W:\\epics\\R3.14.12.7\\base\\lib\\windows-x64\\asIoc.lib  W:\\epics\\R3.14.12.7\\base\\lib\\windows-
x64\\dbIoc.lib  W:\\epics\\R3.14.12.7\\base\\lib\\windows-x64\\registryIoc.lib  W:\\epics\\R3.14.12.7\\base\\lib\\windows-x64\\dbStaticIoc.lib  W:\\ep
ics\\R3.14.12.7\\base\\lib\\windows-x64\\ca.lib  W:\\epics\\R3.14.12.7\\base\\lib\\windows-x64\\Com.lib  Gdi32.lib Oleaut32.lib ws2_32.lib advapi32.li
b user32.lib kernel32.lib winmm.lib
NDPlugin.lib(NDFileHDF5LayoutXML.obj) : error LNK2001: unresolved external symbol xmlFree
```

MarkRivers, 2020-04-16:
I updated your comment to add line breaks after each library to make it easier to read. 
This is the complete output of the link command on my system, also with added line breaks:
```
link -nologo  -LTCG -incremental:no -opt:ref -release  -MACHINE:X64              
-out:simDetectorApp.exe    
simDetectorApp_registerRecordDeviceDriver.obj 
simDetectorAppMain.obj     
../../../../../lib/windows-x64/simDetector.lib  
J:/epics/devel/areaDetector-3-9/ADCore/lib/windows-x64/NDPlugin.lib 
J:/epics/devel/areaDetector-3-9/ADCore/lib/windows-x64/ADBase.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/qsrv.lib  
J:/epics/devel/areaDetector-3-9/ADCore/lib/windows-x64/ntndArrayConverter.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/nt.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/pvDatabase.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/pvAccessIOC.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/pvAccessCA.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/pvAccess.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/pvData.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/netCDF.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/Magick++.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/coders.lib 
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/Magick.lib 
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/jbig.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/jp2.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/bzlib.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/png.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/webp.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/lcms.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/ttf.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/wmf.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/filters.lib 
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/tiff.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/xml2.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/NeXus.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/hdf5.lib 
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/hdf5_hl.lib 
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/bitshuffle.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/blosc.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/szip.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/zlib.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/jpeg.lib  
J:/epics/devel/asyn-4-39/lib/windows-x64/asyn.lib  
J:/epics/devel/autosave-5-10/lib/windows-x64/autosave.lib  
J:/epics/devel/busy-1-7-2/lib/windows-x64/busy.lib  
J:/epics/devel/calc-3-7-3/lib/windows-x64/calc.lib  
J:/epics/devel/devIocStats-3-1-16/lib/windows-x64/devIocStats.lib 
J:/epics/devel/sscan-2-11-3/lib/windows-x64/sscan.lib  
J:/epics/devel/seq-2-2-5/lib/windows-x64/seq.lib  
J:/epics/devel/seq-2-2-5/lib/windows-x64/pv.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/dbRecStd.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/dbCore.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/ca.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64/Com.lib 
netapi32.lib gdi32.lib oleaut32.lib psapi.lib
Generating code
Finished generating code
"Installing created executable ../../../bin/windows-x64/simDetectorApp.exe"
make[1]: Leaving directory 'J:/epics/devel/areaDetector-3-9/ADSimDetector/iocs/simDetectorIOC/simDetectorApp/src/O.windows-x64'
```
Mine has more libraries because I built with GraphicsMagick, netCDF, and pvAccess.  But otherwise it looks very similar to yours, and I don't get an error while you do.  I don't understand this.

MarkRivers, 2020-04-16:
ADCore/ADApp/pluginSrc/Makefile had these lines added in R3-9:
```
ifneq ($(SHARED_LIBRARIES), YES)
   USR_CFLAGS_WIN32 += -DLIBXML_STATIC
   USR_CXXFLAGS_WIN32 += -DLIBXML_STATIC
endif
```
When you build with the windows-x64 architecture on base 3.14.12.7 is SHARED_LIBRARIES defined to be YES?  If not that could cause the problem.  

If you rebuild NDFileHDF5LayoutXML.cpp do you see the flag -DLIBXML_STATIC?

pheest, 2020-04-17:
Hi Mark, Ben,

I’ve built areaDetector end-to-end for the windows-x64-debug configuration, and using specifically base 3.14.12.7.
I haven’t been able to reproduce this issue.

I have the build log file, if this would help.

LIBXML_STATIC is not defined.

I am using VS2017 (not 15) though, to be fair.

All the best,

Peter.

BenBradnick, 2020-04-17:
Thanks for your help! So the problem is that we are building ADSupport dynamically and ADCore statically:

- ADSupport with SHARED_LIBRARIES = YES
- ADCore with SHARED_LIBRARIES = NO

This means that ADCore is adding the -DLIBXML_STATIC flag based on the assumption that both are built in the same way, whereas our XML library in ADSupport is dynamic, not static.

BenBradnick, 2020-04-17:
@MarkRivers should this flag be changed to something that indicates whether ADSupport is being built statically or not rather than assuming the same build configuration as ADCore?

MarkRivers, 2020-04-17:
I think it might be better to track down what is so different about ``xmlFree`` and try to fix it in ADSupport.  The libraries in ADSupport define thousands of functions that get linked correctly when you build ADSupport dynamically and ADCore statically. Only ``xmlFree`` has this problem. 

I looked at the code in xml2Src quickly and I cannot even find where ``xmlFree()`` is defined to be ``free()``.  It is rather opaque to me.

pheest, 2020-04-17:
Hi Mark,

I will have a look into this in due course.

I think I understand the reason why this particular function is a trouble, but I need to verify.

I would see it as odd - but conceptually valid – to mix the two build models.

All the best,

Peter.

MarkRivers, 2020-04-17:
@pheest thanks, I was hoping you would volunteer for this!

BenBradnick, 2020-04-17:
Thank you @pheest. I believe the definition occurs in `globals.c` in `xml2Src`.

I'm not sure why we mix different types of builds either.

pheest, 2020-04-22:
Hi Mark, Ben,

Summary of build issues that I’ve looked at:

  1.  Within xml2Src/makefile:

  ifeq ($(SHARED_LIBRARIES),YES)

    USR_CFLAGS += -DPIC -DNOLIBTOOL

  endif

and within xml2Src/libxml.h

#if !defined(PIC) && !defined(NOLIBTOOL)

#  define LIBXML_STATIC

#endif

I think this can be (more simply and clearly) translated as:

  ifneq ($(SHARED_LIBRARIES),YES)

    USR_CFLAGS += -DLIBXML_STATIC

  endif

PIC is used (Linux only) within elfgcchack.h and libxml.h. I’m not quite sure what the purpose is.

NOLIBTOOL is not used anywhere, that I can see.

  1.  In PluginSrc/makfile:

ifneq ($(SHARED_LIBRARIES), YES)

   USR_CFLAGS_WIN32 += -DLIBXML_STATIC

   USR_CXXFLAGS_WIN32 += -DLIBXML_STATIC

endif

Is clearly an assumption that the build models of ADCore and ADSupport are the same.

If this isn’t specified, then xmlFree triggers Microsoft LNK4217 ‘locally defined symbol’ warning.

Because the pre-processor will have (incorrectly) added __declspec(dllimport) to the function name.

I think also can be removed:

  ifeq ($(SHARED_LIBRARIES),NO)

    # This flag is used to indicate that the TIFF library was built statically

    USR_CXXFLAGS_WIN32 += -DLIBTIFF_STATIC

  endif

Because I don’t think LIBTIFF_STATIC is used anywhere, that can see.

  1.  Within PluginSrc/NDFileHDF5LayoutXML.cpp, I think

xmlFree() can be replaced with xmlGetGlobalState()->xmlFree()

In which case the above LNK4217 warning goes away.

I have exercised the SimDetector with iocHDF5Test in order to verify that the code in question was called, and operated normally.

Do you want this change to be made?

Hope this helps,

Peter.

BenBradnick, 2020-04-24:
I can also get rid of my LNK2001 error by doing the same as you:
```
xmlFree() can be replaced with xmlGetGlobalState()->xmlFree()
```

It also works when I add the originally removed code:
```
#ifdef _MSC_VER
  #define xmlFree free
#endif
```

I'm not sure which is preferred.

pheest, 2020-04-29:
Hi Ben,

I believe calling xmlGetGlobalState()->xmlFree() is likely to be the more correct solution approach.

I originally removed   #define xmlFree free because this usage caused a debug assertion (in a DLL build).
I believe the assertion signified that memory (xml2) allocated within one library was being freed within another(ADCore)  – this could represent a memory leak to xml2.

I’m not wholly sure about this – the best reference on the subject I’ve been able to locate is:
https://devblogs.microsoft.com/oldnewthing/20161209-00/?p=94905

Cheers,

Peter.

BenBradnick, 2020-05-06:
It sounds like your solution fixes both issues so if @MarkRivers is happy this change could be made?

BenBradnick, 2020-07-09:
Hi @MarkRivers, @pheest - I'm just pinging you both again to see if this change is okay to be made?

Also, should the change (xmlFree() can be replaced with xmlGetGlobalState()->xmlFree()) be made directly in the code or with a define? I'm happy to create the pull request, there's just also the following define that will need to be considered in NDFileHDF5LayoutXMP.cpp (whether this will need to be kept and use an else block instead):

```
#ifdef __MINGW32__
  #define xmlFree free
#endif
```

pheest, 2020-08-07:
I believe “xmlFree() can be replaced with xmlGetGlobalState()->xmlFree()” Is the best approach to a tricky issue.
No downsides noted.

I’m interested in Mark’s views on my other comments on simplification of the build pre-processor usage.
These are not required to make the above change – I just couldn’t follow the pre-processor logic, without simplifying it first.

MarkRivers, 2020-09-19:
Closed via #453

closed 2020-09-19 by MarkRivers