areaDetector/ADCore #421: <PV_name> is scalar in file, but array in ioc

hinxx, 2019-11-07:
With ADCore master and autosave master code from git, I'm getting these messages at IOC startup:

```
*** LAB:AMC-005-3:CH1-PROC1-NDAttributesFile is scalar in file, but array in ioc.
*** LAB:AMC-005-3:CH1-PROC1-NDAttributesMacros is scalar in file, but array in ioc.
*** LAB:AMC-005-3:CH1-TIFF1-FilePath is scalar in file, but array in ioc.
*** LAB:AMC-005-3:CH1-TIFF1-FileName is scalar in file, but array in ioc.
*** LAB:AMC-005-3:CH1-TIFF1-FileTemplate is scalar in file, but array in ioc.
*** LAB:AMC-005-3:CH1-TIFF1-NDAttributesFile is scalar in file, but array in ioc.
*** LAB:AMC-005-3:CH1-TIFF1-NDAttributesMacros is scalar in file, but array in ioc.
*** LAB:AMC-005-3:CH1-TRC1-NDAttributesFile is scalar in file, but array in ioc.
*** LAB:AMC-005-3:CH1-TRC1-NDAttributesMacros is scalar in file, but array in ioc.
...
```

They appear after the autosave file was written and IOC restarted (2nd and later starts), for all plugins loaded.

I might be doing something wrong, but I haven't noticed these messages earlier, but I am using latest sources in both ADCore and autosave, as of today.

MarkRivers, 2019-11-07:
I cannot reproduce this problem.  I am also running the master branch of autosave, ADCore, and ADSimDetector.

What does your TIFF1-FilePath entry look like in auto_settings.sav.  Mine looks like this:
```
13SIM1:TIFF1:FilePath @array@ { "47" "104" "111" "109" "101" "47" "101" "112" "105" "99" "115" "47" "115" "99" "114" "97" "116" "99" "104" "0" }
```
My Proc1:TIFF:FilePath looks like this because it has not been set.  It might look like a scaler in the file and an array in the IOC, but I don't get a warning about that.
```
13SIM1:Proc1:TIFF:FilePath 0
```

hinxx, 2019-11-08:
I have not initialized it either, value in save file is:
```
LAB:AMC-005-3:CH1-TIFF1-FilePath 0
```
If I set it to '/tmp', this is saved:
```
LAB:AMC-005-3:CH1-TIFF1-FilePath @array@ { "47" "116" "109" "112" }
```
No complaints at IOC start when TIFF1-FilePath is set.

The difference seems to come from the fact that I'm using `fdbrestoreX("auto_settings.sav")` instead of `set_pass0_restoreFile("auto_settings.sav")` and `set_pass1_restoreFile("auto_settings.sav")`.

I need to have the PV value that is being restored actually being applied to HW as well at IOC boot. I might be using the restore feature wrong, but it seems to do the job (apart from seeing these warnings about scalars/arrays).

Thank you for the time, I should probably address this with autosave folks.

MarkRivers, 2019-11-08:
> I need to have the PV value that is being restored actually being applied to HW as well at IOC boot.

set_pass1_restoreFile("auto_settings.sav") will restore to hardware as long as the driver does not return asynSuccess when that parameter is read before iocInit().  If it does return asynSuccess then the record will use that value rather then the one from autosave (bumpless reboot).

hinxx, 2019-11-14:
Thanks for the additional input @MarkRivers ! I'll got about and try that out.

closed 2019-11-14 by hinxx