areaDetector/ADCore #427: NDPluginCircularBuff operation wrt pre-trigger NDArrays

hinxx, 2020-01-21:
If I understand the NDPluginCircularBuff operation correctly (and the settings for pre-trigger and post-trigger NDArray count), after trigger occurs, all the pre-trigger collected NDArrays should be sent to clients first, then post-trigger NDArrays are sent afterwards. For example, for pre-trigger count 5 and post-trigger count 10, CA client should see 15 NDArrays. For some reason I do not see all the pre-trigger NDArrays, only few, while post-trigger ones are sent out.

Here is an example of two waveforms being monitored, source of `TR1` StdArrays plugin is driver, source of `TR2` StdArrays plugin is circular buffer `CB1` CircularBuffer plugin. Source of `CB1` plugin is driver.

```
$ camonitor -# 5 -tsI LAB:AMC-004-4:I1-XP-TR2-ArrayData LAB:AMC-004-4:I1-XP-TR1-ArrayData LAB:AMC-004-4:I1-XP-CB1-CurrentQty_RBV LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV LAB:AMC-004-4:I1-XP-CB1-Trigger_RBV
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 14043 62557 10922 52429 0  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 7562 63015 10922 10922 57974  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 7562 10922 62557 21845 54613  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 1927 57344 14043 54613 12603  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 57974 5174 43690 16384 61440  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 14043 26810 54613 43690 63015  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071437 5 4681 60855 60855 0 2520  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 52933 0 4681 4681 32768  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 2184 46811 10922 0 57344  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 61895 18204 10922 54613 10922  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 0 25486 56913 50641 32768  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 16384 6553 63351 62557 50641  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 63351 0 54613 45875 58982  
LAB:AMC-004-4:I1-XP-CB1-Trigger_RBV                  +4.283611 1 Triggered STATE MINOR
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 50641 25486 56599 14043 4681  
LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV                  +4.285092 1 1  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +4.142917 5 16384 6553 63351 62557 50641  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.000000 5 16384 6553 63351 62557 50641  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.000000 5 16384 6553 63351 62557 50641  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.000000 5 16384 6553 63351 62557 50641  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 50641 0 5461 61440 37809  
LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV                  +0.072785 1 2  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.214287 5 50641 0 5461 61440 37809  
LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV                  +0.071233 1 3  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.071429 5 18724 58982 54613 10922 10922  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 18724 58982 54613 10922 10922  
LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV                  +0.071453 1 4  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.071429 5 5782 4681 57344 51492 9362  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 5782 4681 57344 51492 9362  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 3641 3641 10922 52429 2184  
LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV                  +0.070580 1 5  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.071429 5 3641 3641 10922 52429 2184  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 54613 60855 8192 0 57344  
LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV                  +0.072824 1 6  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.071429 5 54613 60855 8192 0 57344  
LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV                  +0.068593 1 7  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 0 8192 2979 57344 32768  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.071429 5 0 8192 2979 57344 32768  
LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV                  +0.072632 1 8  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.071438 5 0 60362 50641 47891 10922  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071438 5 0 60362 50641 47891 10922  
LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV                  +0.071370 1 9  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 14043 57344 10922 10922 4096  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.071429 5 14043 57344 10922 10922 4096  
LAB:AMC-004-4:I1-XP-CB1-PostTriggerQty_RBV                  +0.069834 1 0  
LAB:AMC-004-4:I1-XP-CB1-Trigger_RBV                  +0.642995 1 Done  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.071429 5 4681 7562 12603 54613 0  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 4681 7562 12603 54613 0  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 19661 56173 60855 0 21845  
LAB:AMC-004-4:I1-XP-CB1-CurrentQty_RBV                  +4.644258 1 1  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 57974 2979 0 14894 8192  
LAB:AMC-004-4:I1-XP-CB1-CurrentQty_RBV                  +0.070993 1 2  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 4681 60855 16384 54613 18204  
LAB:AMC-004-4:I1-XP-CB1-CurrentQty_RBV                  +0.072436 1 3  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 8936 54613 62557 54613 27306  
LAB:AMC-004-4:I1-XP-CB1-CurrentQty_RBV                  +0.070297 1 4  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 50641 51492 21845 37809 49152  
LAB:AMC-004-4:I1-XP-CB1-CurrentQty_RBV                  +0.073715 1 5  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 38726 57344 12603 5461 57344  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 8192 1724 18204 52429 2520  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 58982 19661 0 18204 49152  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 58982 0 56599 0 62557  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 51492 19661 13107 0 7562  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 4681 54613 0 6553 2184  
[... 14 lines cut ...]
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071438 5 14043 7562 60855 57344 5461  
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 10922 10922 51492 10922 8192  
```

Some explanation of the setup is due.
I'm running continuous acquisition at 14 Hz, acquiring 1D arrays of digitized data on several asyn addresses.
I have setup the `ArrayData` records to use driver timestamp by setting TSE=-2. With `camonitor` looking at the difference of updates (14 Hz ~ 0.071 ms) I can see how the waveforms are updated in time. The `TR1` is updated for every NDArray showing consistent time stamp difference. Circular buffer plugin is set to automatically restart (PresetTriggerCount = 0). Pre-trigger count is set to 5, post-trigger count is set to 10. I'm showing only 5 elements of NDArray for brewity.

After starting the circular buffer is collects 5 NDArrays and then start wrapping as new ones arrive. Upon hitting the `Trigger` button I can see the `TR2` updates coming to client using `camonitor`.

In my observation there seem to be only one NDArray sent that was in the ring buffer before the Trigger was issued. This is the line
```
LAB:AMC-004-4:I1-XP-TR1-ArrayData                  +0.071429 5 16384 6553 63351 62557 50641  
```
It gets sent out on `TR2` multiple times:
```
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +4.142917 5 16384 6553 63351 62557 50641  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.000000 5 16384 6553 63351 62557 50641  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.000000 5 16384 6553 63351 62557 50641  
LAB:AMC-004-4:I1-XP-TR2-ArrayData                  +0.000000 5 16384 6553 63351 62557 50641  
```

Questions:
Where are 5 pre-trigger NDArrays that were collected in the ring buffer?
Why are there 4 updates with the same NDArray?

MarkRivers, 2020-01-21:
I am trying to reproduce this with the simDetector.  I set the AcquirePeriod to 0.071 seconds so I also get 14 frames/s like you.

I have setup the plugins as follows:

simDetector -> CircularBuffer1 -> Stats1

I set the CircularBuffer plugins for PreCount=5, PostCount=10, PresetTriggerCount=0.

I set up camonitor to monitor the UniqueId_RBV and TimeStamp_RBV records from the Stats1 plugin.  This will tell me the UniqueID and TimeStamp of each NDArray that the Stats1 plugin received.

```
corvette:~>camonitor -tc -f3 13SIM1:Stats1:UniqueId_RBV 13SIM1:Stats1:TimeStamp_RBV
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:44.162887) 2574
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:44.163034) 948463955.408

(CircularBuffer Trigger button pressed here)

13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:49.988601) 5306
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:49.988633) 948464149.617
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.001579) 5307
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.001605) 948464149.688
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.014406) 5308
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.014433) 948464149.759
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.025730) 5309
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.025757) 948464149.830
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.036470) 5310
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.036495) 948464149.901
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.045148) 5311
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.045175) 948464149.972
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.056063) 5312
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.056087) 948464150.044
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.126176) 5313
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.126206) 948464150.115
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.199267) 5314
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.199292) 948464150.186
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.270734) 5315
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.270766) 948464150.257
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.341051) 5316
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.341086) 948464150.328
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.412554) 5317
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.412598) 948464150.399
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.481956) 5318
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.481997) 948464150.470
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.554352) 5319
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.554382) 948464150.541
13SIM1:Stats1:UniqueId_RBV     (2020-01-21 08:15:50.627594) 5320
13SIM1:Stats1:TimeStamp_RBV    (2020-01-21 08:15:50.627629) 948464150.612
```

So I am seeing all 15 NDArrays.  The UniqueId is increasing by 1 each time, with no repeats.  The timestamps are 0.071 seconds apart, as expected.

So I am not able to reproduce your problem with this setup.  Can you repeat your experiment with Stats1 and monitoring the 2 PVs I did to see what you get?

I think I may know what your problem is.  NDPluginStdArrays is being called very fast for pre-trigger NDArrays.  Notice that the camonitor timestamps from -tc are only 10-13 ms apart for the first 5 arrays. Unfortunately there is no queuing in Channel Access, and the record processing cannot keep up with the rate at which the callbacks are being done.  The data is thus overwritten and you get the same data over Channel Access 5 times.

I believe you can fix this by enabling ring buffers in the your waveform record for NDPluginStdArrays.  They are disabled by default.

I wonder what happens if you use NDPluginPva rather than NDPluginStdArrays? Does it do queuing, if not by default then is there a way to enable it?

MarkRivers, 2020-01-21:
I tried to reproduce @hinxx  results more closely.  I used this plugin chain:

simDetector -> CircularBuffer1 -> StdArrays

I then monitored ArrayData like he did.
```
[epics@gse-marccd2 ~]$ camonitor -tc -#5 13SIM1:image1:ArrayData
13SIM1:image1:ArrayData (2020-01-21 11:27:39.804918) 5 52 53 54 55 56

13SIM1:image1:ArrayData (2020-01-21 11:27:42.654654) 5 88 89 90 91 92
13SIM1:image1:ArrayData (2020-01-21 11:27:42.655837) 5 89 90 91 92 93
13SIM1:image1:ArrayData (2020-01-21 11:27:42.657121) 5 90 91 92 93 94
13SIM1:image1:ArrayData (2020-01-21 11:27:42.658377) 5 91 92 93 94 95
13SIM1:image1:ArrayData (2020-01-21 11:27:42.659620) 5 92 93 94 95 96
13SIM1:image1:ArrayData (2020-01-21 11:27:42.660875) 5 93 94 95 96 97
13SIM1:image1:ArrayData (2020-01-21 11:27:42.725464) 5 94 95 96 97 98
13SIM1:image1:ArrayData (2020-01-21 11:27:42.796643) 5 95 96 97 98 99
13SIM1:image1:ArrayData (2020-01-21 11:27:42.867814) 5 96 97 98 99 100
13SIM1:image1:ArrayData (2020-01-21 11:27:42.939311) 5 97 98 99 100 101
13SIM1:image1:ArrayData (2020-01-21 11:27:43.010217) 5 98 99 100 101 102
13SIM1:image1:ArrayData (2020-01-21 11:27:43.081339) 5 99 100 101 102 103
13SIM1:image1:ArrayData (2020-01-21 11:27:43.152289) 5 100 101 102 103 104
13SIM1:image1:ArrayData (2020-01-21 11:27:43.223528) 5 101 102 103 104 105
13SIM1:image1:ArrayData (2020-01-21 11:27:43.294494) 5 102 103 104 105 106
```
Note that the first 5 monitors are about 1.2 ms apart.  But the data is not all the same, the counts increase by 1 at each pixel for each callback, which is what is expected with the simDetector as I have configured it.  So I am not seeing duplicate values for the first 5 like @hinxx is.

hinxx, 2020-01-22:
>  Can you repeat your experiment with Stats1 and monitoring the 2 PVs I did to see what you get?

I see the same as you do. All updates are there, with all the pre-trigger updates.

>  The data is thus overwritten and you get the same data over Channel Access 5 times.

Seems to be the case. If I monitor `TR2-UniqueId_RBV` it gets all 5 pre-trigger updates, with expected unique IDs.

> I then monitored ArrayData like he did.

How large is your waveform? My waveforms are about 17 000 points, float32 data type.

> I believe you can fix this by enabling ring buffers in the your waveform record for NDPluginStdArrays.

Is that on the waveform record itself? I tried searching for it on https://wiki-ext.aps.anl.gov/epics/index.php/RRM_3-14_Waveform but can't see it.

MarkRivers, 2020-01-22:
> How large is your waveform? My waveforms are about 17 000 points, float32 data type.

My waveform record is 1MB (1024x1024 NDArray of UInt8 data type).  So it is more than 10 times larger than yours.

>>I believe you can fix this by enabling ring buffers in the your waveform record for NDPluginStdArrays.

>Is that on the waveform record itself?

No, it is in the asyn device support.  The documentation is here:
https://epics-modules.github.io/master/asyn/R4-38/asynDriver.html#DeviceCallbackBuffers

If you add this info tag to the waveform record then it will buffer 10 callbacks:

    info(asyn:FIFO, "10")

MarkRivers, 2020-01-22:
> It might be worth mentioning that this is on master branches of epics-base 7, asyn and adcore, all checked out ~10 days ago.

I think this is just how EPICS works.  It does not not provide queuing for arrays.  There is a race condition between the speed of driver callbacks and the speed of record processing.

Try the info(asyn:FIFO,10) and see if that fixes it.

hinxx, 2020-01-22:
> Try the info(asyn:FIFO,10) and see if that fixes it.

I added `info(asyn:FIFO,10)` to `$(P)$(R)ArrayData` record in `NDStdArrays.template` and now it works as expected; no duplicated or dropped arrays!

Thank you @MarkRivers.

I noticed that the value of the asyn:FIFO field depends on how many pre-trigger NDArrays I want (i.e. 5, 10, 100) and needs to be larger than pre-trigger value.

MarkRivers, 2020-01-22:
> I noticed that the value of the asyn:FIFO field depends on how many pre-trigger NDArrays I want (i.e. 5, 10, 100) and needs to be larger than pre-trigger value.

Yes, that is the expected behavior.

hinxx, 2020-01-23:
@MarkRivers thank you for solving this issue for me!

closed 2020-01-23 by hinxx