areaDetector/ADCore #437: record(longin, "$(P)$(R)DroppedOutputArrays_RBV")

kgofron, 2020-04-16:


kgofron, 2020-04-16:
We are running into limit of long PVs in the areaDetector. It seems that limit of 60 char is exceeded for some of the PVs. Can shortening the postfix part of the areaDetector PVs be possible.?

The 1st one that showed problem
https://github.com/areaDetector/ADCore/blob/master/ADApp/Db/NDPluginBase.template#L375
```
record(longin, "$(P)$(R)DroppedOutputArrays_RBV")
```
```
# Create a TIFF file plugin to read dark and flatfield images into the processing plugin

NDFileTIFFConfigure("PROC1TIFF", 20, 0, "cam01", 0)

dbLoadRecords("NDFileTIFF.template",  "P=XF:05IDD-BI:1{Mscp:1-Cam:1},R=Proc1:TIFF:,PORT=PROC1TIFF,ADDR=0,TIMEOUT=1,NDARRAY_PORT=cam01")

Can't create record "XF:05IDD-BI:1{Mscp:1-Cam:1}Proc1:TIFF:DroppedOutputArrays_RBV" of type "longin"

Error at or before ")" in path "/epics/prod/Deb7/production/areaDetector/ADCore/db"  file "NDPluginBase.template" line 375

in path "/epics/prod/Deb7/production/areaDetector/ADCore/db"  file "NDFileTIFF.template" line 9
```

MarkRivers, 2020-04-16:
I would turn this around and say that can't you make your $(P) prefix shorter than 27 characters?

kgofron, 2020-04-16:
Let me put another spin, whether increasing EPICS PV length beyond 60 be possible?

Otherwise, If we have to, we will have to modify existing iocs and limit Prefix up to  27 characters, Only now during remote upgrades we are catching these few PVs.

mp49, 2020-04-16:
I saw the same issue in the past, and I started using a convention for plugin PV names that helps keep the overall name short:

P1 - 1st processing plugin (rather than :Proc1:)
S1 - 1st stats plugin
A1 - 1st array plugin
etc.

MarkRivers, 2020-08-10:
> Let me put another spin, whether increasing EPICS PV length beyond 60 be possible?

See this tech-talk thread for a discussion of that issue.
https://epics.anl.gov/tech-talk/2017/msg01217.php

My conclusions:
- Backwards compatibility of areaDetector precludes changing PV names in the AD template files.  You are free to change the prefixes in commonPlugins.cmd for your site as suggested by @mp49.
- It is unlikely that EPICS base will increase the 60 character limit any time soon for the reasons discussed in that tech-talk thread.

closed 2020-08-10 by MarkRivers