areaDetector/ADCore #445: Missing includes of shareLib.h for EPICS 7.0.4

anjohnson, 2020-06-04:
EPICS 7.0.4 changes the macros that our internal libraries use to mark symbols for import/export, and as a result most core API header files no longer `#include <shareLib.h>`. That means external applications can't rely on the epicsShare* macros to have been defined by upstream APIs, their header files need to include that header themselves if they use those macros.

I've just built an old version of ADCore needed for the APS Controls DAQ systems, and had to add the `#include` statements found in the diffs below to get it to build. You might have fixed a few of these already, and there may be other headers that need to do something similar, so this is just to warn you and get you started on what needs doing to support 7.0.4. There are no implications for backwards-compatibility in adding this include, although there's a slight chance that doing so might tickle some hidden decoration issues on Microsoft builds (which I haven't tried doing). Asyn 4-34 built with no problems.

Sorry, GitHub won't let me attach these as a .patch file, but since the paths in my version might not match those in this git repo anyway I think this is the easiest way to show my changes:

```diff
diff --git a/ADCore/ADApp/ADSrc/ADDriver.h b/ADCore/ADApp/ADSrc/ADDriver.h
index 2a4eec3..304e34c 100755
--- a/ADCore/ADApp/ADSrc/ADDriver.h
+++ b/ADCore/ADApp/ADSrc/ADDriver.h
@@ -4,6 +4,7 @@
 #include <epicsTypes.h>
 #include <epicsMessageQueue.h>
 #include <epicsTime.h>
+#include <shareLib.h>
 #include <asynStandardInterfaces.h>
 
 #include "asynNDArrayDriver.h"
diff --git a/ADCore/ADApp/ADSrc/NDArray.h b/ADCore/ADApp/ADSrc/NDArray.h
index 92189d8..d6f2e12 100644
--- a/ADCore/ADApp/ADSrc/NDArray.h
+++ b/ADCore/ADApp/ADSrc/NDArray.h
@@ -15,6 +15,7 @@
 #include <epicsMutex.h>
 #include <epicsTime.h>
 #include <stdio.h>
+#include <shareLib.h>
 
 #include "NDAttribute.h"
 #include "NDAttributeList.h"
diff --git a/ADCore/ADApp/ADSrc/NDAttribute.h b/ADCore/ADApp/ADSrc/NDAttribute.h
index a356131..0f28d8f 100644
--- a/ADCore/ADApp/ADSrc/NDAttribute.h
+++ b/ADCore/ADApp/ADSrc/NDAttribute.h
@@ -16,6 +16,7 @@
 
 #include <ellLib.h>
 #include <epicsTypes.h>
+#include <shareLib.h>
 
 /** Success return code  */
 #define ND_SUCCESS 0
diff --git a/ADCore/ADApp/ADSrc/NDAttributeList.h b/ADCore/ADApp/ADSrc/NDAttributeList.h
index a8f6ccb..c036b82 100644
--- a/ADCore/ADApp/ADSrc/NDAttributeList.h
+++ b/ADCore/ADApp/ADSrc/NDAttributeList.h
@@ -12,6 +12,7 @@
 #include <stdio.h>
 #include <ellLib.h>
 #include <epicsMutex.h>
+#include <shareLib.h>
  
 #include "NDAttribute.h"
 
diff --git a/ADCore/ADApp/ADSrc/asynNDArrayDriver.h b/ADCore/ADApp/ADSrc/asynNDArrayDriver.h
index 9ba270a..e942b83 100755
--- a/ADCore/ADApp/ADSrc/asynNDArrayDriver.h
+++ b/ADCore/ADApp/ADSrc/asynNDArrayDriver.h
@@ -1,6 +1,7 @@
 #ifndef asynNDArrayDriver_H
 #define asynNDArrayDriver_H
 
+#include <shareLib.h>
 #include "asynPortDriver.h"
 #include "NDArray.h"
 #include "ADCoreVersion.h"
diff --git a/NDDriverStdArrays/NDDriverStdArraysApp/src/NDDriverStdArrays.h b/NDDriverStdArrays/NDDriverStdArraysApp/src/NDDriverStdArrays.h
index 82aea5d..5ec1786 100644
--- a/NDDriverStdArrays/NDDriverStdArraysApp/src/NDDriverStdArrays.h
+++ b/NDDriverStdArrays/NDDriverStdArraysApp/src/NDDriverStdArrays.h
@@ -13,6 +13,7 @@
  */
 
 
+#include <shareLib.h>
 #include "ADDriver.h"
 
 #define DRIVER_VERSION      1
diff --git a/pvaDriver/pvaDriverApp/src/pvaDriver.h b/pvaDriver/pvaDriverApp/src/pvaDriver.h
index eba3b03..d440102 100644
--- a/pvaDriver/pvaDriverApp/src/pvaDriver.h
+++ b/pvaDriver/pvaDriverApp/src/pvaDriver.h
@@ -1,3 +1,4 @@
+#include <shareLib.h>
 #include <pv/clientFactory.h>
 #include <pv/pvAccess.h>
 #include <pv/ntndarray.h>
```

MarkRivers, 2020-06-04:
I am already building ADCore for base 7.0.4 and did not need to make any changes to do so.

git grep shows the following files in ADCore contain shareLib.h.  It is fewer than you listed.
```
corvette:~/support/areaDetector/ADCore>git grep shareLib.h
ADApp/ADSrc/CCDMultiTrack.h:#include <shareLib.h>
ADApp/ADSrc/PVAttribute.cpp:#include <shareLib.h>
ADApp/ADSrc/asynNDArrayDriver.cpp:#include <shareLib.h>
ADApp/ADSrc/functAttribute.cpp:#include <shareLib.h>
ADApp/ADSrc/paramAttribute.cpp:#include <shareLib.h>
```

I don't think this is actually an issue.

anjohnson, 2020-06-04:
Okay, glad to hear that. I did say I was building an older version (a cut-down copy of AreaDetector 3.2) because that's what the our DAQ software still needs to build against.

closed 2020-06-04 by anjohnson