areaDetector/ADCore #455: Clean up epicsExport.h vs epicsExportSharedSymbols

MarkRivers, 2020-08-10:
The way we are currently handling epicsExportSharedSymbols is not ideal.  

It requires, for example, that any .cpp file in ADSrc that includes NDArray.h must also include all of the external header files that NDArray.h includes (e.g. epicsMutex.h, epicsTime.h) before the .cpp file includes epicsExport.h or defines epicsExportSharedSymbols.  This is hard to maintain, because if a new external file is added to NDArray.h then it must also be added to all the .cpp files in this library/dll that directly or indirectly include NDArray.h.

Note that if NDArray.h is included in a .cpp file that is in a separate library (for example NDPluginDriver.cpp) then it is not necessary to include those other external headers.  It only applies to .cpp files that will be in the same library/dll as the functions defined in that .h file.

We should probably switch to the more verbose mechanism of undefining epicsExportSharedSymbols in the header file when including external header files.  That makes it easier to maintain the .cpp files, because they do not also need to include those external header files.

xiaoqiangwang, 2020-08-11:
Since asynPortDriver is the base class of many, asynPortDriver.h is also one included, sometimes implicitly. The following is to exclude it from symbol export.
```
$ cat exclude_asynPortDriver_export.patch       
diff --git a/ADApp/pluginSrc/NDFileHDF5.cpp b/ADApp/pluginSrc/NDFileHDF5.cpp
index 1add341..7308015 100644                                               
--- a/ADApp/pluginSrc/NDFileHDF5.cpp                                        
+++ b/ADApp/pluginSrc/NDFileHDF5.cpp                                        
@@ -34,6 +34,7 @@                                                           
 #include <osiSock.h>                                                       
                                                                            
 #include <asynDriver.h>                                                    
+#include <asynPortDriver.h>                                                
                                                                            
 #include <epicsExport.h>                                                   
 #include "NDFileHDF5.h"                                                    
diff --git a/ADApp/pluginSrc/NDFileJPEG.cpp b/ADApp/pluginSrc/NDFileJPEG.cpp
index 5b7e6b1..f03e798 100644                                               
--- a/ADApp/pluginSrc/NDFileJPEG.cpp                                        
+++ b/ADApp/pluginSrc/NDFileJPEG.cpp                                        
@@ -17,6 +17,7 @@                                                           
 #include <iocsh.h>                                                         
                                                                            
 #include <asynDriver.h>                                                    
+#include <asynPortDriver.h>                                                
                                                                            
 #include <epicsExport.h>                                                   
 #include "NDPluginFile.h"                                                  
diff --git a/ADApp/pluginSrc/NDFileMagick.cpp b/ADApp/pluginSrc/NDFileMagick.cpp
index c3f37bd..b77ace5 100644                                                   
--- a/ADApp/pluginSrc/NDFileMagick.cpp                                          
+++ b/ADApp/pluginSrc/NDFileMagick.cpp                                          
@@ -17,6 +17,7 @@                                                               
 #include <iocsh.h>                                                             
                                                                                
 #include <asynDriver.h>                                                        
+#include <asynPortDriver.h>                                                    
                                                                                
 #include <epicsExport.h>                                                       
 #include "NDPluginFile.h"                                                      
diff --git a/ADApp/pluginSrc/NDFileNetCDF.cpp b/ADApp/pluginSrc/NDFileNetCDF.cpp
index 9e7be93..a365e5e 100644                                                   
--- a/ADApp/pluginSrc/NDFileNetCDF.cpp                                          
+++ b/ADApp/pluginSrc/NDFileNetCDF.cpp                                          
@@ -18,6 +18,7 @@                                                               
 #include <iocsh.h>                                                             
                                                                                
 #include <asynDriver.h>                                                        
+#include <asynPortDriver.h>                                                    
                                                                                
 #include <epicsExport.h>                                                       
 #include "NDPluginFile.h"                                                      
diff --git a/ADApp/pluginSrc/NDFileNexus.cpp b/ADApp/pluginSrc/NDFileNexus.cpp  
index 9699416..801b11a 100644                                                   
--- a/ADApp/pluginSrc/NDFileNexus.cpp                                           
+++ b/ADApp/pluginSrc/NDFileNexus.cpp                                           
@@ -22,6 +22,7 @@                                                               
 #include <string.h>                                                            
                                                                                
 #include <asynDriver.h>                                                        
+#include <asynPortDriver.h>                                                    
                                                                                
 #include <epicsExport.h>                                                       
 #include "NDPluginFile.h"                                                      
diff --git a/ADApp/pluginSrc/NDFileNull.cpp b/ADApp/pluginSrc/NDFileNull.cpp    
index 80a8dcd..794858c 100644                                                   
--- a/ADApp/pluginSrc/NDFileNull.cpp                                            
+++ b/ADApp/pluginSrc/NDFileNull.cpp                                            
@@ -18,6 +18,7 @@                                                               
 #include <iocsh.h>                                                             
[... cut ...]
+++ b/ADApp/pluginSrc/NDPluginTransform.cpp
@@ -24,6 +24,7 @@
 #include <iocsh.h>

 #include <asynDriver.h>
+#include <asynPortDriver.h>

 #include <epicsExport.h>
 #include "NDPluginDriver.h"
diff --git a/ADApp/pluginSrc/NDPosPlugin.cpp b/ADApp/pluginSrc/NDPosPlugin.cpp
index 1dbe780..dce65b2 100644
--- a/ADApp/pluginSrc/NDPosPlugin.cpp
+++ b/ADApp/pluginSrc/NDPosPlugin.cpp
@@ -18,6 +18,7 @@
 #include <epicsThread.h>

 #include <asynDriver.h>
+#include <asynPortDriver.h>

 #include <epicsExport.h>
 #include "NDPluginDriver.h"
```

MarkRivers, 2020-08-11:
I plan to discuss this issue with Andrew Johnson and get his advice.  I see that in EPICS base 7.0.4 there are 2 approaches.

libCom and ca no longer use epicsExportSharedSymbols.  

In 3.15 public functions were declared like this:
```
epicsShareFunc void epicsShareAPI epicsEnvSet (const char *name, const char *value);
epicsShareFunc short epicsShareAPI ca_field_type (chid chan); 
```

In 7.0.4 they are declared like this:
```
LIBCOM_API void epicsStdCall epicsEnvSet (const char *name, const char *value);
LIBCA_API short epicsStdCall ca_field_type (chid chan); 
```

In 7.0.4 pvAccess uses this mechanism in monitor.h, for example:
```
#ifdef epicsExportSharedSymbols
#   define monitorEpicsExportSharedSymbols
#   undef epicsExportSharedSymbols
#endif

#include <epicsMutex.h>
#include <pv/status.h>
#include <pv/pvData.h>
#include <pv/sharedPtr.h>
#include <pv/bitSet.h>
#include <pv/createRequest.h>

#ifdef monitorEpicsExportSharedSymbols
#   define epicsExportSharedSymbols
#       undef monitorEpicsExportSharedSymbols
#endif
```

beldish, 2020-08-20:
I see you are working on this. 

I get these errors while compiling:

/usr/bin/g++ -D_GNU_SOURCE -D_DEFAULT_SOURCE -D_X86_64_ -DUNIX -Dlinux -O3 -Wall -DPREFIX_MAGICK_SYMBOLS -DHAVE_JPEG -DHAVE_BLOSC -DHAVE_BITSHUFFLE -DH5_NO_DEPRECATED_SYMBOLS -DH5Gopen_vers=2 -mtune=generic -m64 -fPIC -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include -I/home/mbeldis/work/EPICS/support/synApps_6_1/support/asyn-R4-36/include -I/home/mbeldis/work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADSupport/include/os/Linux -I/home/mbeldis/work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADSupport/include -I/home/mbeldis/work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore/include -I/home/mbeldis/work/EPICS/epics-base/include/compiler/gcc -I/home/mbeldis/work/EPICS/epics-base/include/os/Linux -I/home/mbeldis/work/EPICS/epics-base/include -c ../NDArrayRing.cpp
In file included from ../../../include/NDArray.h:20:0,
from ../NDArrayRing.h:4,
from ../NDArrayRing.cpp:19:
../../../include/NDAttribute.h:87:23: error: variable ‘epicsShareClass NDAttribute’ has initializer but incomplete type
class epicsShareClass NDAttribute {
^
../../../include/NDAttribute.h:87:23: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
../../../include/NDAttribute.h:88:1: error: expected primary-expression before ‘public’
public:
^
../../../include/NDAttribute.h:88:1: error: expected ‘}’ before ‘public’
../../../include/NDAttribute.h:88:1: error: expected ‘,’ or ‘;’ before ‘public’
../../../include/NDAttribute.h:92:16: error: expected constructor, destructor, or type conversion before ‘(’ token
NDAttribute(NDAttribute& attribute);
^
../../../include/NDAttribute.h:94:26: error: declaration of ‘~NDAttribute’ as non-member
virtual ~NDAttribute();
^
../../../include/NDAttribute.h:95:13: error: ‘NDAttribute’ does not name a type
virtual NDAttribute* copy(NDAttribute *pAttribute);
^
../../../include/NDAttribute.h:96:33: error: ‘virtual’ outside class declaration
virtual const char *getName();
^
../../../include/NDAttribute.h:97:40: error: ‘virtual’ outside class declaration
virtual const char *getDescription();
^
../../../include/NDAttribute.h:98:35: error: ‘virtual’ outside class declaration
virtual const char *getSource();
^
../../../include/NDAttribute.h:99:66: error: ‘virtual’ outside class declaration
virtual const char *getSourceInfo(NDAttrSource_t *pSourceType);
^
../../../include/NDAttribute.h:100:42: error: ‘virtual’ outside class declaration
virtual NDAttrDataType_t getDataType();
^
../../../include/NDAttribute.h:101:76: error: ‘virtual’ outside class declaration
virtual int getValueInfo(NDAttrDataType_t *pDataType, size_t *pDataSize);
^
../../../include/NDAttribute.h:102:84: error: ‘virtual’ outside class declaration
virtual int getValue(NDAttrDataType_t dataType, void *pValue, size_t dataSize=0);
^
../../../include/NDAttribute.h:103:44: error: ‘virtual’ outside class declaration
virtual int getValue(std::string& value);
^
../../../include/NDAttribute.h:104:54: error: ‘virtual’ outside class declaration
virtual int setDataType(NDAttrDataType_t dataType);
^
../../../include/NDAttribute.h:105:44: error: ‘virtual’ outside class declaration
virtual int setValue(const void *pValue);
^
../../../include/NDAttribute.h:106:44: error: ‘virtual’ outside class declaration
virtual int setValue(const std::string&);
^
../../../include/NDAttribute.h:107:29: error: ‘virtual’ outside class declaration
virtual int updateValue();
^
../../../include/NDAttribute.h:108:45: error: ‘virtual’ outside class declaration
virtual int report(FILE *fp, int details);
^
../../../include/NDAttribute.h:109:5: error: ‘friend’ used outside of class
friend class NDArray;
^
../../../include/NDAttribute.h:110:5: error: ‘friend’ used outside of class
friend class NDAttributeList;
^
../../../include/NDAttribute.h:113:1: error: expected unqualified-id before ‘private’
private:
^
../../../include/NDAttribute.h:124:1: error: expected declaration before ‘}’ token
};
^
../../../include/NDAttribute.h:93:24: warning: ‘const char* attrSourceString(NDAttrSource_t)’ declared ‘static’ but never defined [-Wunused-function]
static const char *attrSourceString(NDAttrSource_t type);
^
make[5]: *** [NDArrayRing.o] Error 1
make[5]: Leaving directory `/home/mbeldis/Work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore/ADApp/pluginSrc/O.linux-x86_64'
make[4]: *** [install.linux-x86_64] Error 2
make[4]: Leaving directory `/home/mbeldis/Work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore/ADApp/pluginSrc'
make[3]: *** [pluginSrc.install] Error 2
make[3]: Leaving directory `/home/mbeldis/Work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore/ADApp'
make[2]: *** [ADApp.install] Error 2
make[2]: Leaving directory `/home/mbeldis/Work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore'
make[1]: *** [/home/mbeldis/work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore.install] Error 2
make[1]: Leaving directory `/home/mbeldis/Work/EPICS/support/synApps_6_1/support/areaDetector-R3-7'
make: *** [/home/mbeldis/work/EPICS/support/synApps_6_1/support/areaDetector-R3-7.install] Error 2
[mbeldis@sl7 support]$ ^C
[mbeldis@sl7 support]$ 

OS  Scientific Linux 7.8. EPICS base 7. synApps_6_1. asyn-R4-36.

shareLib.h is not picked up during compilation.

In NDArrayRing.cpp  
#inlcude <asynDriver.h>
or 
#include <epicsExport.h>
or both are missing.

NDArray.h is included via NDArrayRing.h.

MarkRivers, 2020-08-31:
I have decided to use the XXX_API mechanism that is used in base 7.0.4.  This completely eliminates the need for shareLib.h, epicsExportSharedSymbols, epicsShareFunc, epicsShareClass, etc.  In my opinion it is much simpler and cleaner.  With the shareLib.h method one always needs to be careful not to "accidentally" export a function from another library by forgetting to include the other library header file before defining epicsExportSharedSymbols.

I have created a new Without_shareLib branch that implements these changes.  It builds fine on Linux.  On Windows there are link errors with the pvAccess library.

This is the link command (with added line breaks) and the error message:

```
J:\epics\devel\areaDetector-3-9\ADCore\ADApp\pluginSrc>make
make -C O.windows-x64 -f ../Makefile TOP=../../.. \
    T_A=windows-x64 install
make[1]: Entering directory 'J:/epics/devel/areaDetector-3-9/ADCore/ADApp/pluginSrc/O.windows-x64'
link -nologo -subsystem:windows -dll -LTCG -incremental:no -opt:ref -release  
-MACHINE:X64 -out:NDPlugin.dll -implib:NDPlugin.lib     
NDPluginDriver.obj throttler.obj NDPluginAttribute.obj NDPluginCircularBuff.obj NDArrayRing.obj 
NDPluginColorConvert.obj NDPluginFFT.obj fft.obj NDPluginGather.obj NDPluginOverlay.obj 
NDPluginOverlayTextFont.obj NDPluginProcess.obj NDPluginROI.obj NDPluginROIStat.obj 
NDPluginScatter.obj NDPluginStats.obj NDPluginStdArrays.obj NDPluginTimeSeries.obj 
NDPluginTransform.obj NDPluginAttrPlot.obj NDPluginCodec.obj NDPosPlugin.obj 
NDPosPluginFileReader.obj NDPluginFile.obj NDFileNull.obj NDFileMagick.obj NDFileHDF5.obj 
NDFileHDF5Dataset.obj NDFileHDF5AttributeDataset.obj NDFileHDF5LayoutXML.obj 
NDFileHDF5Layout.obj NDFileJPEG.obj NDFileNetCDF.obj NDFileNexus.obj NDFileTIFF.obj 
NDPluginPva.obj
../../../lib/windows-x64/ADBase.lib  
J:/epics/devel/asyn-4-40/lib/windows-x64/asyn.lib  
../../../lib/windows-x64/ntndArrayConverter.lib  
H:/epics-devel/base-7.0.4/lib/windows-x64/nt.lib  
H:/epics-devel/base-7.0.4/lib/windows-x64/pvDatabase.lib  
H:/epics-devel/base-7.0.4/lib/windows-x64/pvAccess.lib  
H:/epics-devel/base-7.0.4/lib/windows-x64/pvData.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/netCDF.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/Magick++.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/coders.lib
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/Magick.lib
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/jbig.lib
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/jp2.lib
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/bzlib.lib
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/png.lib
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/webp.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/lcms.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/ttf.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/wmf.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/filters.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/tiff.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/xml2.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/NeXus.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/hdf5.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/hdf5_hl.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/bitshuffle.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/blosc.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/szip.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/zlib.lib  
J:/epics/devel/areaDetector-3-9/ADSupport/lib/windows-x64/jpeg.lib  
H:/epics-devel/base-7.0.4/lib/windows-x64/dbRecStd.lib  
H:/epics-devel/base-7.0.4/lib/windows-x64/dbCore.lib  
H:/epics-devel/base-7.0.4/lib/windows-x64/ca.lib  
H:/epics-devel/base-7.0.4/lib/windows-x64/Com.lib 
ws2_32.lib user32.lib gdi32.lib oleaut32.lib
   Creating library NDPlugin.lib and object NDPlugin.exp
NDPluginPva.obj : error LNK2001: unresolved external symbol 
"public: static enum epics::pvData::ScalarType const epics::pvData::PVScalarValue<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > >::typeCode" (?typeCode@?$PVScalarValue@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@pvData@epics@@2W4ScalarType@23@B)
NDPluginPva.obj : error LNK2001: unresolved external symbol 
"public: static enum epics::pvData::ScalarType const epics::pvData::PVScalarValue<signed char>::typeCode" (?typeCode@?$PVScalarValue@C@pvData@epics@@2W4ScalarType@23@B)
NDPluginPva.obj : error LNK2001: unresolved external symbol 
"__int64 const epics::pvData::posixEpochAtEpicsEpoch" (?posixEpochAtEpicsEpoch@pvData@epics@@3_JB)
NDPluginPva.obj : error LNK2001: unresolved external symbol 
"public: static enum epics::pvData::ScalarType const epics::pvData::PVScalarValue<unsigned char>::typeCode" (?typeCode@?$PVScalarValue@E@pvData@epics@@2W4ScalarType@23@B)
NDPluginPva.obj : error LNK2001: unresolved external symbol "public: static enum epics::pvData::ScalarType const epics::pvData::PVScalarValue<char>::typeCode" (?typeCode@?$PVScalarValue@D@pvData@epics@@2W4ScalarType@23@B)
NDPlugin.dll : fatal error LNK1120: 5 unresolved externals
make[1]: *** [H:/epics-devel/base-7.0.4/configure/RULES_BUILD:298: NDPlugin.dll] Error 1120
make[1]: Leaving directory 'J:/epics/devel/areaDetector-3-9/ADCore/ADApp/pluginSrc/O.windows-x64'
make: *** [H:/epics-devel/base-7.0.4/configure/RULES_ARCHS:58: install.windows-x64] Error 2
```

MarkRivers, 2020-09-01:
This issue should be fixed in #457.

MarkRivers, 2020-09-19:
Closed via #457

closed 2020-09-19 by MarkRivers