areaDetector/ADCore #456: Compilation errors for NDArrayRing.cpp

beldish, 2020-08-20:
I get these errors while compiling:

/usr/bin/g++ -D_GNU_SOURCE -D_DEFAULT_SOURCE -D_X86_64_ -DUNIX -Dlinux -O3 -Wall -DPREFIX_MAGICK_SYMBOLS -DHAVE_JPEG -DHAVE_BLOSC -DHAVE_BITSHUFFLE -DH5_NO_DEPRECATED_SYMBOLS -DH5Gopen_vers=2 -mtune=generic -m64 -fPIC -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include -I/home/mbeldis/work/EPICS/support/synApps_6_1/support/asyn-R4-36/include -I/home/mbeldis/work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADSupport/include/os/Linux -I/home/mbeldis/work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADSupport/include -I/home/mbeldis/work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore/include -I/home/mbeldis/work/EPICS/epics-base/include/compiler/gcc -I/home/mbeldis/work/EPICS/epics-base/include/os/Linux -I/home/mbeldis/work/EPICS/epics-base/include -c ../NDArrayRing.cpp
In file included from ../../../include/NDArray.h:20:0,
from ../NDArrayRing.h:4,
from ../NDArrayRing.cpp:19:
../../../include/NDAttribute.h:87:23: error: variable ‘epicsShareClass NDAttribute’ has initializer but incomplete type
class epicsShareClass NDAttribute {
^
../../../include/NDAttribute.h:87:23: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
../../../include/NDAttribute.h:88:1: error: expected primary-expression before ‘public’
public:
^
../../../include/NDAttribute.h:88:1: error: expected ‘}’ before ‘public’
../../../include/NDAttribute.h:88:1: error: expected ‘,’ or ‘;’ before ‘public’
../../../include/NDAttribute.h:92:16: error: expected constructor, destructor, or type conversion before ‘(’ token
NDAttribute(NDAttribute& attribute);
^
../../../include/NDAttribute.h:94:26: error: declaration of ‘~NDAttribute’ as non-member
virtual ~NDAttribute();
^
../../../include/NDAttribute.h:95:13: error: ‘NDAttribute’ does not name a type
virtual NDAttribute* copy(NDAttribute *pAttribute);
^
../../../include/NDAttribute.h:96:33: error: ‘virtual’ outside class declaration
virtual const char *getName();
^
../../../include/NDAttribute.h:97:40: error: ‘virtual’ outside class declaration
virtual const char *getDescription();
^
../../../include/NDAttribute.h:98:35: error: ‘virtual’ outside class declaration
virtual const char *getSource();
^
../../../include/NDAttribute.h:99:66: error: ‘virtual’ outside class declaration
virtual const char *getSourceInfo(NDAttrSource_t *pSourceType);
^
../../../include/NDAttribute.h💯42: error: ‘virtual’ outside class declaration
virtual NDAttrDataType_t getDataType();
^
../../../include/NDAttribute.h:101:76: error: ‘virtual’ outside class declaration
virtual int getValueInfo(NDAttrDataType_t *pDataType, size_t *pDataSize);
^
../../../include/NDAttribute.h:102:84: error: ‘virtual’ outside class declaration
virtual int getValue(NDAttrDataType_t dataType, void *pValue, size_t dataSize=0);
^
../../../include/NDAttribute.h:103:44: error: ‘virtual’ outside class declaration
virtual int getValue(std::string& value);
^
../../../include/NDAttribute.h:104:54: error: ‘virtual’ outside class declaration
virtual int setDataType(NDAttrDataType_t dataType);
^
../../../include/NDAttribute.h:105:44: error: ‘virtual’ outside class declaration
virtual int setValue(const void *pValue);
^
../../../include/NDAttribute.h:106:44: error: ‘virtual’ outside class declaration
virtual int setValue(const std::string&);
^
../../../include/NDAttribute.h:107:29: error: ‘virtual’ outside class declaration
virtual int updateValue();
^
../../../include/NDAttribute.h:108:45: error: ‘virtual’ outside class declaration
virtual int report(FILE fp, int details);
^
../../../include/NDAttribute.h:109:5: error: ‘friend’ used outside of class
friend class NDArray;
^
../../../include/NDAttribute.h:110:5: error: ‘friend’ used outside of class
friend class NDAttributeList;
^
../../../include/NDAttribute.h:113:1: error: expected unqualified-id before ‘private’
private:
^
../../../include/NDAttribute.h:124:1: error: expected declaration before ‘}’ token
};
^
../../../include/NDAttribute.h:93:24: warning: ‘const char attrSourceString(NDAttrSource_t)’ declared ‘static’ but never defined [-Wunused-function]
static const char *attrSourceString(NDAttrSource_t type);
^
make[5]: *** [NDArrayRing.o] Error 1
make[5]: Leaving directory /home/mbeldis/Work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore/ADApp/pluginSrc/O.linux-x86_64' make[4]: *** [install.linux-x86_64] Error 2 make[4]: Leaving directory /home/mbeldis/Work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore/ADApp/pluginSrc'
make[3]: *** [pluginSrc.install] Error 2
make[3]: Leaving directory /home/mbeldis/Work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore/ADApp' make[2]: *** [ADApp.install] Error 2 make[2]: Leaving directory /home/mbeldis/Work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore'
make[1]: *** [/home/mbeldis/work/EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore.install] Error 2
make[1]: Leaving directory `/home/mbeldis/Work/EPICS/support/synApps_6_1/support/areaDetector-R3-7'
make: *** [/home/mbeldis/work/EPICS/support/synApps_6_1/support/areaDetector-R3-7.install] Error 2
[mbeldis@sl7 support]$ ^C
[mbeldis@sl7 support]$

OS Scientific Linux 7.8. EPICS base 7. synApps_6_1. asyn-R4-36.

shareLib.h is not picked up during compilation.

In NDArrayRing.cpp
#inlcude <asynDriver.h>
or
#include <epicsExport.h>
or both are missing.

NDArray.h is included via NDArrayRing.h.

MarkRivers, 2020-08-20:
What version of gcc are you running? The master branch of ADCore compiles with no error or warnings for me with gcc 4.8.5.

beldish, 2020-08-20:
[mbeldis@sl7 ~]$ gcc --version
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-39)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

[mbeldis@sl7 ~]$ g++ --version
g++ (GCC) 4.8.5 20150623 (Red Hat 4.8.5-39)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

If I run above failing command with -H switch, I see that shareLib.h is not included.

beldish, 2020-08-20:
On the second try, I downloaded assemble_synApps.sh from  https://github.com/EPICS-synApps/support.
Configured it and everything worked like a charm.

The version from  https://www.aps.anl.gov/BCDA/synApps/Where-to-find-it also can be compiled if one fixes xxx configure/RELEASE SUPPORT recursion and adds
#include <asynDriver.h>
and/or
#include <epicsExport.h>
to  NDArrayRing.cpp in .../EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore/ADApp/pluginSrc

These versions obviously differ as the github version required the installation of additional packages libnet-devel, libcap-devel and libpcap-devel. 
Closing this issue and thank you for your time.

closed 2020-08-20 by beldish

beldish, 2020-08-20:
NDAttribute.h is defined differently in these two repos.

The first is from github and the second from www.aps.anl.gov/BCDA/synApps/Where-to-find-it.
The first contains #include <asynDriver.h>

[mbeldis@sl7 EPICS]$ diff synApps/support/areaDetector-master/ADCore/include/NDAttribute.h  support/synApps_6_1/support/areaDetector-R3-7/ADCore/include/NDAttribute.h 
20,22d19
< /* asynDriver.h is needed only to define epicsInt64 on 3.14 */
< #include <asynDriver.h>
< 
38,39d34
<     NDInt64,    /**< Signed 64-bit integer */
<     NDUInt64,   /**< Unsigned 64-bit integer */
53,54d47
<     NDAttrInt64   = NDInt64,    /**< Signed 64-bit integer */
<     NDAttrUInt64  = NDUInt64,   /**< Unsigned 64-bit integer */
79,80d71
<     epicsInt64   i64;   /**< Signed 64-bit integer */
<     epicsUInt64  ui64;  /**< Unsigned 64-bit integer */
86c77
<   * This is needed for ellLists of C++ objects, for which making the first data element the ELLNODE
---
>   * This is needed for ellLists of C++ objects, for which making the first data element the ELLNODE 
99c90
<     NDAttribute(const char *pName, const char *pDescription,
---
>     NDAttribute(const char *pName, const char *pDescription,

MarkRivers, 2020-08-20:
GitHub is the primary repository. The synApps tar file is not a repository, it is a snapshot of the Github repository at some specific commit.

MarkRivers, 2020-08-20:
> These versions obviously differ as the github version required the installation of additional packages libnet-devel, libcap-devel and libpcap-devel.

libnet-devel etc. are only required for the Canberra support in the mca module. They can be disabled.