areaDetector/ADCore #464: GitHub Actions

GDYendell, 2020-12-01:
Has any consideration been made on moving CI to GitHub Actions? I have been working on the PVI PR and the travis were (and still are) incredibly slow to start, so I setup the Github Actions build and it seems to work really nicely. You can see them here: https://github.com/dls-controls/ADCore/actions

MarkRivers, 2020-12-01:
I think the first thing to do is to change ADCore to use ci-scripts.  That has been done for the asyn, busy, motor, modbus, ip, autosave, std, and mca modules.  That makes it easier to use a variety of CI, including Travis, Appveyor, and Github Actions.

It should be done for ADCore.  We should also add CI to each detector driver.  It will really help to have good caching support so that base, asyn, ADCore, ADSupport, etc. is not rebuilt each time a detector driver is rebuilt.

GDYendell, 2023-10-04:
Fixed by #483

closed 2023-10-04 by GDYendell