areaDetector/ADCore #468: writeNotify fails to notify for Acquire PV

mfurseman, 2021-02-08:
I have come across a strange problem while using [caproto](https://github.com/caproto/caproto) alongside areaDetector. This seems to be present on at least ADAndor and ADSim detectors.

When attempting to change the `$(P)$(R)Acquire` PV to the value of `1` with caprotos's `PV.write` method, which initiates a CA writeNotify, the callback notification is never sent from the areaDetector IOC.

I have been unable to trackdown the root cause of this problem but it seems that the callback to send the Notify event is created as Extra Labour for the CAS_Event thread, but in this particular case the event mutux is never granted to the thread and it [hangs](https://github.com/epics-base/epics-base/blob/30e1431fb458659baef752bcd4fa84484ed057f0/modules/database/src/ioc/db/dbEvent.c#L1011).

ulrikpedersen, 2021-02-08:
Hi @mfurseman good to see you around here 😄

When setting Acquire to 1 you start an acquisition but the callback shouldn't return until the acquisition is complete...

Docs about $(P)$(R)Acquire:
> Start (1) or stop (0) image acquisition. This record forward links to $(P)$(R)AcquireBusy which is an EPICS busy record that does not process its forward link until acquisition is complete. Clients should write 1 to the Acquire record to start acquisition, and wait for AcquireBusy to go to 0 to know that acquisition is complete. This can be done automatically with ca_put_callback.

So does your camera/ADSim complete the acquisition and set Acquire back down to 0? Does it work plain old `caput -c`?

MarkRivers, 2021-02-08:
If PV.write effectively does ca_put_callback then the callback will not occur until Acquire goes back to 0.  If you did a Continuous acquisition then it will never occur until you manually stop the camera.

You will see the same behavior with 
```
caput -c
```
but in that case there is a default 1 second timeout, so you will see a timeout error.  If you set a 10 second acquisition and do 
```
caput -c -w 15
```
then you should get a callback after 10 seconds, rather than a timeout.

mfurseman, 2021-02-09:
Hi @ulrikpedersen,

Nice to hear from you. We're implementing some EPICS solutions for diagnostics on the MAST-U Experiment.

Thank you for your insights, I suppose I've been looking for a complicated solution to a simple 'problem'! Indeed the callback is processed when acquisition stops.

closed 2021-02-09 by mfurseman