areaDetector/ADCore #471: New plugin for bad pixel correction

MarkRivers, 2021-04-21:
It would be very useful to have a new plugin for bad pixel correction.  To date this is handled in an ad-hoc manner in some specific drivers (e.g. ADPilatus).

Some vendor libraries support bad pixel maps (e.g. ADPilatus, ADPerkinElmer), which is fine.  But many vendors do not support this, so it would be useful to be able to do it in a plugin.

It would be good to have at least the following 3 methods of handling bad pixels:
- Replace the pixel value with that of a single nearby pixel.
- Replace the value with the median of N surrounding pixels.
- Set the pixel to a user-specified value, e.g. 0 or -1.

I think the plugin should handle:
-	All NDArray data types
-	2-D NDArrays and 3-D NDArrays with ColorMode=RGB1

Here are some cases that should be handled:
1. The neighboring pixels in the median filter happen to include other bad pixels.  They need to be excluded.
2. Only a portion of the detector is being read out.  
3. The detector is binned when being read out.
4. The detector readout is reversed in X or Y

The NDArray NDDimension structures contain all of the information required to handle cases 2-4:
```
/** Structure defining a dimension of an NDArray */
typedef struct NDDimension {
    size_t size;    /**< The number of elements in this dimension of the array */
    size_t offset;  /**< The offset relative to the origin of the original data source (detector, for example).
                      * If a selected region of the detector is being read, then this value may be > 0.
                      * The offset value is cumulative, so if a plugin such as NDPluginROI further selects
                      * a subregion, the offset is relative to the first element in the detector, and not
                      * to the first element of the region passed to NDPluginROI. */
    int binning;    /**< The binning (pixel summation, 1=no binning) relative to original data source (detector, for example)
                      * The offset value is cumulative, so if a plugin such as NDPluginROI performs binning,
                      * the binning is expressed relative to the pixels in the detector and not to the possibly
                      * binned pixels passed to NDPluginROI.*/
    int reverse;    /**< The orientation (0=normal, 1=reversed) relative to the original data source (detector, for example)
                      * This value is cumulative, so if a plugin such as NDPluginROI reverses the data, the value must
                      * reflect the orientation relative to the original detector, and not to the possibly
                      * reversed data passed to NDPluginROI. */
} NDDimension_t;
```
Some drivers probably don’t set those NDDimension fields as they should, so they will need to be fixed to handle those cases.

djvine, 2021-04-21:
How do you recommend to input the coordinates of pixels, rows, & columns to be corrected?

Text file? Binary array mask that's loaded similar to how you do flat and dark PROC plugin?

MarkRivers, 2021-04-21:
The bad pixel list could be contained in a file using JSON syntax:
```
{ "Bad pixels" :  
[
  {  "X" : 220,  "Y" : 150, "Median" : 3 },        
  {  "X" : 620,  "Y" : 750, "Set" : 0 },
  {  "X" : 820,  "Y" : 850, "Replace" : "821,850" }
]
}
```

djvine, 2021-04-22:
I couldn't find a parser for JSON in the standard library. 

How about just:
"X" : 220,  "Y" : 150, "Median" : 3       
"X" : 620,  "Y" : 750, "Set" : 0,
"X" : 820,  "Y" : 850, "Replace" : "821,850"

MarkRivers, 2021-04-22:
There is a JSON parser in ADEiger.

https://github.com/areaDetector/ADEiger/tree/master/eigerApp/frozenSrc

I was thinking of moving that to ADSupport.  The nice thing about using the parser is that it will accept white space, newlines, etc. and we don't need to worry about handling that.

djvine, 2021-04-23:
Do you assume any C++ standard? Can I use std::array?

MarkRivers, 2021-04-23:
We assume C++ 11.  You can use std::vector.

MarkRivers, 2021-04-28:
I have found a nice C++ JSON parser: 
https://github.com/nlohmann/json 

I am using that parser to read the bad pixel JSON file here:
https://github.com/areaDetector/ADCore/blob/dbd80d0d7bbedacb72f67fd696de35e7e2efee33/ADApp/pluginSrc/NDPluginBadPixel.cpp#L118

The bad pixel file is this:
```
{ "Bad pixels" :  
[
  {  "X" : 220,  "Y" : 150, "Median" : 3 },        
  {  "X" : 620,  "Y" : 750, "Set" : 0 },
  {  "X" : 820,  "Y" : 850, "Replace" : "821,850" }
]
}
```
It is simple to get the X and Y coordinates for each pixel because those keys are always present.  Where I am having trouble is figuring out how to write the code for keys that are not always present.

When the code referenced above parses the JSON string shown above I get the following output:
```
Bad pixel 0, X=220, Y=150
Median 3
Set null
Replace null
Bad pixel 1, X=620, Y=750
Median null
Replace null
Bad pixel 2, X=820, Y=850
Median null
Set null
Replace "821,850"
```
So std::cout shows that the value of badPixels[i]["Replace"] is null when I expect it to be.  My question is how do I test for that in my C++ code, since testing for !=0 is clearly not working, it always returns !=0 = true.

MarkRivers, 2021-04-28:
I have created a new branch, "bad_pixel_plugin" for this development.

MarkRivers, 2021-04-29:
The bad pixel file listed above has incorrect syntax.  The Replace value should be an array, like this:
```

{ "Bad pixels" :
[
  {  "X" : 220,  "Y" : 150, "Median" : 3 },
  {  "X" : 620,  "Y" : 750, "Set" : 0 },
  {  "X" : 820,  "Y" : 850, "Replace" : [821,850] }
]
}
```
NDPluginBadPixelFile is now parsing the JSON file correctly.

MarkRivers, 2021-04-29:
I have figured out how to use the JSON parser.  It is very nice, the code to read and parse the JSON file is very compact.

The JSON file now looks like this:
```
{ "Bad pixels" :
  [
    {"Pixel" : [220,150], "Median" : 1},
    {"Pixel" : [620,750], "Set" : 0},
    {"Pixel" : [820,850], "Replace" : [1,0]}
  ]
}
```
The value of Median is the size of the median filter, N.  The median filter is 2N+1, i.e. N=1 is a 3x3 filter, N=2 is a 5x5 filter.  This ensures that the filter dimension is always odd and hence centered on the bad pixel.

The value of Set is the value to set the bad pixel to.  It is stored as a double.  This allows it to be used with almost any data type without loss of precision.  The exception is NDArrays of type Int64 and UInt64 where a double cannot exactly represent some values.

The value of Replace is the relative coordinates of a nearby pixel to replace the bad pixel whose coordinates are [X,Y].  [1,0] uses the pixel at [X+1,Y], [0,-1] uses the pixel at [X, Y-1], etc.

The plugin has a single parameter which is the path to the JSON file.  After reading in the above JSON file asynReport on the plugin with details >0 shows the list of bad pixels:
```
epics> asynReport 1 BADPIX1
BADPIX1 multiDevice:Yes canBlock:No autoConnect:Yes
    enabled:Yes connected:Yes numberConnects 1
    nDevices 1 nQueued 0 blocked:No
    asynManagerLock:No synchronousLock:No
    exceptionActive:No exceptionUsers 0 exceptionNotifys 0
    traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
    addr 0 autoConnect Yes enabled Yes connected Yes exceptionActive No
        exceptionActive No exceptionUsers 1 exceptionNotifys 0
        blocked No
        traceMask:0x1 traceIOMask:0x0 traceInfoMask:0x1
Bad pixel 0, coords=[220,150], mode=Median, size=1
Bad pixel 1, coords=[620,750], mode=Set, value=0.000000
Bad pixel 2, coords=[820,850], mode=Replace, relative coordinates=[1,0]
Port: BADPIX1
  Timestamp: <undefined>
  Input EOS[0]:
  Output EOS[0]:
Parameter list 0
Number of parameters is: 81
Parameter 0 type=string, name=PORT_NAME_SELF, value=BADPIX1, status=0
Parameter 1 type=string, name=ADCORE_VERSION, value=3.10.0, status=0
Parameter 2 type=string, name=DRIVER_VERSION, value=3.10.0, status=0
...
```

MarkRivers, 2021-04-29:
OPI screens, template file, and autosave file are complete.  The plugin reads the JSON file, but does not yet actually do anything to fix the bad pixels in the processCallbacks method.

MarkRivers, 2021-05-01:
I have this plugin working.  I have merged the bad_pixel_plugin branch into master.

I ran a test with the ADSimDetector with these settings.  The simulated image is a 4x3 array of peaks peaks starting at [100,100] and 100 pixels apart in X and Y.  Each peak has a sigma width of 1 pixel and a maximum amplitude of 15000.  There is a background offset of 100.

[image]

[image]

NDPluginBadPixel was configured to read the new "bad_pixel_file.json" in ADSimDetector iocBoot/iocSimDetector directory.

[image]

That file contains the following:
```
{ "Bad pixels" :
  [
    {"Pixel" : [200,100], "Set" : 0},
    {"Pixel" : [300,100], "Set" : 1000},
    {"Pixel" : [400,100], "Set" : 2000},
    {"Pixel" : [200,200], "Replace" : [1,0]},
    {"Pixel" : [300,200], "Replace" : [2,0]},
    {"Pixel" : [400,200], "Replace" : [1,1]},
    {"Pixel" : [200,300], "Median" : 1},
    {"Pixel" : [300,300], "Median" : 2},
    {"Pixel" : [400,300], "Median" : 3}
  ]
}
```

This bad pixel file defines the following:
- Does not modify the first peak in each row.
- The first row uses Set mode with values of 0, 1000, and 2000.
- The second row of peaks uses Replace mode with [X+1,0], [X+2,0], and [X+1,Y+1]
- The third row of peaks uses Median with sizes of 1, 2, 3 which is median files of 3x3, 5x5, and 7x7 pixels.

This is an ImageJ screen shot of the resulting peaks, where the center pixel of each peak is modified according to the bad pixel file.  Each peak was verified to be quantitatively correct.

The function that reads the bad pixel file catches exceptions from the JSON parser and prints useful error messages to indicate where the problem is if the file is not valid JSON.

MarkRivers, 2021-05-01:
@djvine this is ready for you to test.

- It handles any NDArray data type.
- It should correctly handle bad pixels that are near the borders of the image, meaning it won't try to use a non-existent pixel.  However, this has not yet been tested.
  - In Replace mode it is up to the user to define a replacement pixel that is valid.  For example for a pixel on the right-most column of the detector [1,0] would result in a non-existent pixel so no replacement will be done.  [-1,0], [0,1], and [0,-1] would all be fine.
  - In Median mode the median will use fewer pixels if the bad pixel is too close to the border.  For example with a medianSize=1 the plugin normally uses 8 pixels in calculating the median (3x3 minus the bad pixel itself).  A bad pixel on any edge of the image will only use 6 pixels to calculate the median, and a bad pixel in one of the 4 corners of the image will only use 3 pixels.
- The plugin does not check that the replacement value computed is using another bad pixel.  It is up to the user to define the bad pixel mode and parameter so that this does not happen.
- It does not yet handle the following:
  - Color images
  - Only a portion of the detector is being read out.
  - The detector is binned when being read out.
  - The detector readout is reversed in X or Y
 
The plugin takes about 0.5 ms to execute with 1024x1024 8-bit images with 9 bad pixels.

djvine, 2021-05-01:
Ok I'll test and let you know.

I thought I would beat you to it but I fell behind a couple of days ago.

Thanks for doing this.

djvine, 2021-05-03:
Did you push your branch? I only see Master.

MarkRivers, 2021-05-03:
I said this in a comment on May 1.
>I have this plugin working. I have merged the bad_pixel_plugin branch into master.

So the master branch contains the new plugin.

xiaoqiangwang, 2021-05-03:
@MarkRivers Will this plugin handle pArray->ndims = 1?

MarkRivers, 2021-05-03:
I just pushed a new version.  It now handles reading out only a portion of the detector (e.g. non-zero MinX and MinY), and it handles binning (e.g. BinX and BinY != 1).

It does not handle reversing the axes (e.g. ReverseX or ReverseY != 0), and I do not plan to support that because it requires information on the actual detector size that is not currently available in the NDArray.

It only handles ColorMode=Mono.  I don't plan to implement other color modes in the near future.

I have added documentation for this plugin: https://areadetector.github.io/master/ADCore/NDPluginBadPixel.html

MarkRivers, 2021-05-03:
> @MarkRivers Will this plugin handle pArray->ndims = 1?

It should.  I need to test it, and if there are problems I will fix it.

djvine, 2021-05-03:
This statement:
The plugin does not check that the replacement value computed is using another bad pixel. It is up to the user to define the bad pixel mode and parameter so that this does not happen.

Means you can't use a median to fix a bad column. 

Two possible solutions:
* add a "bad column" category,
* check for adjacent bad pixels and handle them differently.

MarkRivers, 2021-05-03:
Another possible solution:
- Change the Median value to be a 2-D array
- "Median":[2,0] means use 5 pixels in X, and 1 pixel in Y.  Then it only does the median horizontally and can be used for bad columns.
- "Median":[0,1] means use 1 pixel in X and 3 pixels vertically.  This can be used for bad rows. 

This solution is easy to implement.  I'll do it now.

MarkRivers, 2021-05-03:
I have implemented that and tested with ADSimDetector.

MarkRivers, 2021-05-03:
I think the modification to have different X and Y sizes in the median is a good idea.  

But I would also like to be able to exclude other bad pixels automatically in Median mode.  To do that efficiently the bad pixel list needs to be sorted so that it is very fast to find whether there is a bad pixel at location [X,Y].   That suggests using an std::map rather than and std::vector for the list.  I am working on that.

MarkRivers, 2021-05-03:
I have improved the plugin so it avoids using other bad pixels in Median mode.   This was done by changing the bad pixel list to an std:set sorted on the bad pixel coordinates.  This makes it fast to search the list to see if a replacement pixel is also bad.  

In Replace mode it will print a warning message with ASYN_TRACE_WARNING if the replacement pixel is also bad.  In this case it will not do the replacement.

In Median mode it will also print a warning message with ASYN_TRACE_WARNING for each pixel in the median calculation that is also bad.  It will not use these bad pixels in the median calculation.

I have tested with the simDetector and it appears to be working.

MarkRivers, 2021-05-04:
@xiaoqiangwang  wrote:
> @MarkRivers Will this plugin handle pArray->ndims = 1?

Yes, it does work.  I have modified ADCSimDetector to be able to test it.  ADCSimDetector now exports 1-D NDArrays using asyn addr 1-8.  addr 0 is still the 2-D array. 

This is the screen shot of ADCSimDetector I used for testing.  It is generating 2000 point Float32 arrays.

[image]

This is the NDPluginBadPixel screen:

[image]

This is the bad_pixel_file.json in the iocADCSimDetector directory
```
{ "Bad pixels" :
  [
    {"Pixel" : [10,0], "Set" : 0},
    {"Pixel" : [20,0], "Set" : 1000},
    {"Pixel" : [30,0], "Set" : 2000},
    {"Pixel" : [40,0], "Replace" : [1,0]},
    {"Pixel" : [50,0], "Replace" : [2,0]},
    {"Pixel" : [60,0], "Replace" : [3,0]},
    {"Pixel" : [70,0], "Median" : [1,0]},
    {"Pixel" : [80,0], "Median" : [2,0]},
    {"Pixel" : [90,0], "Median" : [3,0]}
  ]
}
```
This is the NDPluginStdArrays plugin.  Note that it is getting its data from BADPIX1, the NDPluginBadPixel plugin, and is using addr=1 which is the sin wave data.

[image]

I used IDL to read the array using Channel Access and then look at the values in the regions of the bad pixels:
```
IDL> t = caget('13ADCSIM1:image1:ArrayData', v, max=100)
IDL> print, v[9:11]
      0.78194016       0.0000000      0.43372667
IDL> print, v[19:21]
     -0.43404078       1000.0000     -0.78172278
IDL> print, v[29:31]
   0.00017432633       2000.0000      0.97488910
IDL> print, v[39:41]
      0.43372667     -0.97496670     -0.97496670
IDL> print, v[78:82]
   0.00017432527      0.78194016      0.21695049      0.43372667     -0.43404078
```
The Set, Replace, and Median values in the array are all quantitatively correct.

MarkRivers, 2021-05-04:
All of the changes in ADCore, ADSimDetector, and ADCSimDetector are now pushed to the master branches on Github.

The documentation is updated to reflect these latest changes:
https://areadetector.github.io/master/ADCore/NDPluginBadPixel.html

xiaoqiangwang, 2021-05-06:
I tested this plugin with a 1-D detector, all _Median_, _Replace_ and _Set_ strategies produce expected results. Thanks!

closed 2021-05-17 by MarkRivers