areaDetector/ADCore #475: NDPluginCircularBuff behavior on trigger event

hinxx, 2021-11-01:
I have a CB set up to take 15 arrays of pre-trigger data and the `NDCircBuffFlushOnSoftTrig` is set to `OnNewImage`. The CB buffering is started and after it stores 15 arrays the driver is stopped (for the purposes of this test, it would be running otherwise). Then the CB is instructed to flush the collected arrays using `NDCircBuffSoftTrigger` PV from OPI. Next, the driver is instructed to generate a single NDArray. That makes the CB _immediately_ send out the collected 15 arrays of data (through the linked NDStdArrays plugin).
When observed with `camonitor` on remote machine, it can be seen that all 15 arrays are dumped to the wire, with no delay between them, resulting in some of the NDarrays (waveforms) to not be seen on the wire (CA feature?). It gets worse if the number of pre-trigger arrays is increased.

Is this expected behavior?

MarkRivers, 2021-11-01:
Yes, that is the expected behavior. The circular buffer plugin does not have any delay between outputting arrays, nor does NDPluginStdArrays. The circular buffer plugin works fine with other plugins, such as file writing, because their input queues can buffer the burst of NDArrays.

hinxx, 2021-11-01:
In my application I would need to throttle these updates a bit. I'm thinking of adding some new controls for that. Basic idea would be that some external event (through PV) could dictate the rate of the updates; one update of that PV would cause single CB collected array to be sent out to next plugin. I need to give it a bit more thought, though. Default behavior would remain as it is today.

Would you accept modifications upstream?

MarkRivers, 2021-11-01:
@hinxx there may be a simpler solution to your problem.  The problem may arising because the callbacks from NDPluginStdArrays are happening faster than the waveform record can process.  By default the asyn device support does not buffer the callback values for waveform records.  
https://epics-modules.github.io/master/asyn/R4-42/asynDriver.html#DeviceCallbackBuffers

You can change this behavior by adding this line to the database, in this case to queue up to 20 arrays.
```
info(asyn:FIFO, "20")
```
You don't actually need to modify  NDStdArrays.template, you can just create a new database file like this that you load after NDStdArrays.template.
```
record(waveform, "$(P)$(R)ArrayData")
{
    info(asyn:FIFO, "20")
}
```
It is also possible that the waveform record is keeping up with processing, and the problem is in the CA layer.  In that case the above fix will not help.  But I suggest trying it first.

hinxx, 2021-11-01:
Thanks for pointing asyn FIFO field out @MarkRivers, will try that out. I recall seeing that somewhere before but I never used it hence the lack of awareness.

hinxx, 2021-11-03:
I've set the FIFO to 20 for the NDStdArray plugin. For the test I set pre-trigger count to 15.
The client still does not get all 15 pre-trigger arrays at trigger time, but only ~6 were seen by the `camonitor`.

hinxx, 2021-11-03:
> It is also possible that the waveform record is keeping up with processing, and the problem is in the CA layer

Would you know how to diagnose this to situation? Is there something documented about this behavior?

MarkRivers, 2021-11-03:
@hinxx 
- What is the size and datatype of your arrays?
- What is the speed of the network link between the IOC and CA client?

hinxx, 2021-11-03:
The datatype is `Float32`, 1D array, ~20000 points. The test is carried out between a host and a client on the lab network with 1Gb link.

MarkRivers, 2021-11-03:
What is the QueueSize of your NDPluginStdArrays plugin?

hinxx, 2021-11-03:
20

MarkRivers, 2021-11-03:
I have recreated the issue using the simDetector.  I used a simple configuration that does not involve the NDPluginCircularBuffer.  The only active plugin is NDPluginStdArrays.  

The st_base.cmd was changed to have this line to load the waveform record for NDPluginStdArrays.  It is FLOAT data type with 20000 elements.
```
# This waveform allows transporting 32-bit float images
dbLoadRecords("NDStdArrays.template", "P=13SIM1:,R=image1:,PORT=Image1,ADDR=0,TIMEOUT=1,NDARRAY_PORT=SIM1,TYPE=Float32,FTVL=FLOAT,NELEMENTS=20000")
```
This is a screen shot of the simDetector driver:
[image]

It is configured with
- 200x100 arrays (20000 elements, same as @hinxx)
- Data type = Float32
- Simulation mode = Linear Ramp, Gain=1.0
- AcquireTime = 1 ms, AcquirePeriod=0
- ImageMode=Multiple, NumImages=200

In this configuration the driver is generating about 500 frames/s, so one image every 2 ms.  I enabled NDPluginStdArrays, and ran camonitor on another Linux machine connected to the IOC Linux machine via 1 Gbit Ethernet.  I directed the output of camonitor to a file.  Each time acquisition is started it will generate a burst of 200 arrays at about 500 Hz.  The camonitor output file should thus contain 201 lines, because it prints one line on the initial connection to the waveform record.

This is what I see when I monitor the entire waveform record, which is 20000 elements:
```
TahoeU18:/corvette/home/epics/devel/measComp> /corvette/usr/local/epics/base-7.0.4/bin/linux-x86_64-ub18/camonitor 13SIM1:image1:ArrayData > array_test1.txt
^C
TahoeU18:/corvette/home/epics/devel/measComp> wc array_test1.txt
     96 1920384 9605664 array_test1.txt
```
So the file is only 96 lines long, meaning that it missed more than 50% of the arrays.

This is what I see if I only monitor 10000 elements:
```
TahoeU18:/corvette/home/epics/devel/measComp> /corvette/usr/local/epics/base-7.0.4/bin/linux-x86_64-ub18/camonitor -#10000 13SIM1:image1:ArrayData > array_test2.txt
^C
TahoeU18:/corvette/home/epics/devel/measComp> wc array_test2.txt
    144 1440576 7208496 array_test2.txt
```
Now it is losing about 25% of the arrays.  

This is monitoring 8000 elements:
```
TahoeU18:/corvette/home/epics/devel/measComp> /corvette/usr/local/epics/base-7.0.4/bin/linux-x86_64-ub18/camonitor -#8000 13SIM1:image1:ArrayData > array_test3.txt
^C
TahoeU18:/corvette/home/epics/devel/measComp> wc array_test3.txt
    195 1560780 7811310 array_test3.txt
```

It is only losing 2% of the arrays.

This is monitoring 7000 elements:
```
TahoeU18:/corvette/home/epics/devel/measComp> /corvette/usr/local/epics/base-7.0.4/bin/linux-x86_64-ub18/camonitor -#7000 13SIM1:image1:ArrayData > array_test4.txt
^C
TahoeU18:/corvette/home/epics/devel/measComp> wc array_test4.txt
    201 1407804 7046658 array_test4.txt
```
It does not drop any arrays.

This proves that the problem is in the CA layer, because the areaDetector driver and plugins have no idea how many elements are being monitored by the client.

The issue comes from the design of CA, which provides no queuing. There is only a single buffer allocated for the CA array.  If the record processes again due to a callback before the previous monitor has been sent out, it just overwrites the data, and only does a single monitor callback, not 2.

This is a calculation of how many bits/s would be sent across the network when monitoring the entire waveform record and the detector is generating 500 frames/s.
```
20000 elements * 4 bytes per element * 8 bits/byte * 500 arrays/s = 320,000,000 bits/s
```
This is about 33% of the theoretical bandwidth of 1 Gbit Ethernet.

I am not sure how pvAccess handles this, but I will test it.

hinxx, 2021-11-03:
I really appreciate your time and effort in explaining the behavior!

> I am not sure how pvAccess handles this, but I will test it.

That sounds very nice!

MarkRivers, 2021-11-03:
I repeated the test with pvAccess.  I first changed the simDetector so that the maximum array size was only 200x100.  This improves the speed, because previously when it was 1024x1024 it actually calculated 1024x1024 arrays and the selected a 200x100 subarray.  That clearly slows things down.  Now when I use AcquireTime=.001 it generates about 825 arrays/s, vs 500 previously.  With .002 seconds it generates about 450 frames/s, similar to my previous test. with .001 second acquire time.

I disabled NDPluginStdArrays and enabled NDPluginPva.  I ran pvMonitor like this:
```
TahoeU18:/corvette/home/epics/scratch> /corvette/usr/local/epics/base-7.0.4/bin/linux-x86_64-ub18/pvmonitor -r "value" 13SIM1:Pva1:Image > test1.txt
^CTahoeU18:/corvette/home/epics/scratchgrep float test1.txt | cut -c1-100 | wc
    201     402   20301
```

- At 50 frames/s there were 201 lines in the file, as expected.
- At 100 frames/s there were 195 lines in the file, so it is dropping 3% of frames.
- At 200 frames/s there were 145 lines in the file, so it is dropping 25% of frames.
- At 450 frames/s there were 59 lines in the file, so it is dropping 75% of frames.

I suspect the problem is actually that pvMonitor is not keeping up, because it does a lot of work to convert to ASCII and write the file.

We need better clients to test with than camonitor and pvMonitor, because we don't know if the dropped frames are actually due to those programs not being able to keep up, or whether the frames never arrive on the wire.

MarkRivers, 2021-11-03:
I just tested using the pvAccess plugin in ImageJ.  It shows that at 500 frames/s I usually get all 200 frames.

[image]

It shows the number of NTNDArrays received every 2 seconds (if any).  I was starting acquisition on the detector less frequently than every 2 seconds, so each time I started it ImageJ should have received 200 frames.  It often did get 200, but sometimes is was less (199, etc.).  At 850 frames/s it typically got 180 OK.  This shows that the results of pvmonitor are not accurate, since the ImageJ client loses a much smaller fraction of the arrays.

In Continuous mode the ImageJ client shows essentially the same frame rate received as the simDetector is generating up to 320 frames/s.  Above that ImageJ is not processing all of the frames.  It is not clear if this is a problem with pvAccess, or with ImageJ not keeping up with the rate that frames are coming in.

If I only send 15 frames then ImageJ got all of them at 450 frames/s.  It misses 2-3 out of 15 at 830 frames/s.

hinxx, 2021-11-04:
@MarkRivers thank you again for helping in understanding this behavior!

It looks like PVA does perform better than CA. Even though that there are still some missed arrays, those loses appear at quite high rates, which my use case will never reach.

In conclusion, I would expect that for my use case of 14 arrays per second in continuous mode client is not supposed to miss any arrays.

closed 2022-05-20 by hinxx