areaDetector/ADCore #482: Looking at an image for a given port?

bsobhani, 2022-06-29:
Is there a way to look at the image for a given port (e.g. "SIM1") directly, without going through the ArrayData PV? Maybe something through the ioc shell, or the AsynIO PV? I want to know for debugging purposes. I recently caught an IOC that have one of its ArrayData PVs configured for the wrong port. This may have been easier to catch if I had the ability to view the ports directly.

MarkRivers, 2022-06-29:
> Is there a way to look at the image for a given port (e.g. "SIM1") directly, without going through the ArrayData PV?

It depends what you mean by "look at".  If you mean send over the network using Channel Access then you must use NDPluginStdArrays.   If you mean send over the network using pvAccess then you must use NDPluginPva.  You can always temporarily switch NDPluginStdArrays to get its data from any other plugin to see what that plugin is doing.

>I recently caught an IOC that have one of its ArrayData PVs configured for the wrong port. This may have been easier to catch if I had the ability to view the ports directly.

The easiest way to quickly see how all of the plugin ports are configured is to look at the commonPlugins.adl screen.  That shows where every plugin is getting its data.

[image]

You suggested through the ioc shell.  Do you really want to look at the raw values of an NDArray, when there could be millions of values?

However, you can get some information about the NDArray for a port using asynReport with details 10 or greater.
```
epics> asynReport 10 SIM1
SIM1 multiDevice:No canBlock:No autoConnect:Yes
    enabled:Yes connected:Yes numberConnects 1
    nDevices 0 nQueued 0 blocked:No
    asynManagerLock:No synchronousLock:No
    exceptionActive:No exceptionUsers 1 exceptionNotifys 0

...
SIM1: pArrays[0] report

NDArray  Array address=0x7f4cd4004600:
  ndims=2 dims=[5000 4000 ]
  dataType=3, dataSize=40000000, pData=0x7f4c89154010
  uniqueId=200, timeStamp=1025384688.103102, epicsTS.secPastEpoch=1025384688, epicsTS.nsec=203243282
  referenceCount=3
  number of attributes=1

NDAttributeList: address=0x7f4cd40030f0:
  number of attributes=1

SIM1: private NDArrayPool report

NDArrayPool:
  numBuffers=36, numFree=27
  memorySize=1240395200, maxMemory=0
  freeList: (index, dataSize, pArray)
    0 184 0x7f4df4049e90
    1 8000 0x7f4df4047fa0
    2 10000 0x7f4df404c000
    3 40000000 0x7f4cd4001fb0
    4 40000000 0x7f4cd4004800
    5 40000000 0x7f4cd4003870
    6 40000000 0x7f4cd4002c10
    7 40000000 0x7f4cd40042e0
    8 40000000 0x7f4cd40012e0
    9 40000000 0x7f4cd40026f0
    10 40000000 0x7f4df404c340
    11 40000000 0x7f4cd4004470
    12 40000000 0x7f4cd4004150
    13 40000000 0x7f4e44000c80
    14 40000000 0x7f4cd4003e00
    15 40000000 0x7f4cd4003b00
    16 40000000 0x7f4cd4003350
    17 40000000 0x7f4cd4002ea0
    18 40000000 0x7f4cd4001a90
    19 40000000 0x7f4cd4001570
    20 40000000 0x7f4cd4002240
    21 40000000 0x7f4cd4002980
    22 40000000 0x7f4cd4003fc0
    23 40000000 0x7f4cd4001d20
    24 40000000 0x7f4cd40035e0
    25 40000000 0x7f4cd4004a60
    26 40000000 0x7f4cd40024d0

SIM1: pAttributeList report

NDAttributeList: address=0x36d3a50:
  number of attributes=0
```

This part of the report tells you information about the last NDArray that it processed.  It gives the dimensions, the data type, uniqueId, etc.

```
SIM1: pArrays[0] report

NDArray  Array address=0x7f4cd4004600:
  ndims=2 dims=[5000 4000 ]
  dataType=3, dataSize=40000000, pData=0x7f4c89154010
  uniqueId=200, timeStamp=1025384688.103102, epicsTS.secPastEpoch=1025384688, epicsTS.nsec=203243282
  referenceCount=3
  number of attributes=1
```

MarkRivers, 2022-07-05:
@bsobhani does this answer your question?

bsobhani, 2022-07-05:
Yes it does. After thinking about this I guess it makes sense that only a PV of the 2D array type can view the image on a port. But hypothetically if there were some kind of back door to access the data on areadetector (or asyn in general) I think this would be nice, although I am not sure technically how it could work. Just like you can send commands to the [...]Asyn PV to communicate with the device, it would be nice if there was a similar PV that a person could use to communicate directly to the areadetector/asyn layer. Like a universal PV that you can use to query the value of any ADDriver member variable. Maybe this would be hard since C++ is a compiled language.

MarkRivers, 2022-07-05:
You can use the asyn record to get the value of any ADDriver scalar parameter.  Here is the simDetector screen.  AcquireTime=0.02 and NumImages=100. 
[image]

This is the asynRecord configured with the ACQ_TIME drvInfo string, which corresponds to AcquireTime.
[image]

This is the asynRegister screen configured to read from the ACQ_TIME parameter on the asynFloat64 interface:
[image]

Note that the value is 0.02, which is the AcquireTime.

This is configured to read the NumImages value (drvInfo=NIMAGES) on the asynInt32 interface.
[image]

[image]

Note that the value is 100, which is the same as on the simDetector screen.

closed 2022-07-05 by MarkRivers

bsobhani, 2022-07-05:
Thank you, I did not know about this.