areaDetector/ADCore #488: No rule to make target ../../../db/NDFFT_settings.req

bsobhani, 2023-03-10:
I am able to build R3-10 without issue. Starting with R3-11 onward I get the following:

```
Installing template file ../../../db/NDColorConvert.template
Installing template file ../../../db/NDAttrPlot.template
Installing template file ../../../db/NDBadPixel.template
make[3]: *** No rule to make target `../../../db/NDFFT_settings.req', needed by `build'.  Stop.
make[3]: Leaving directory `/SNS/users/q8i/ADCore/ADApp/Db/O.linux-x86_64'
make[2]: *** [install.linux-x86_64] Error 2
make[2]: Leaving directory `/SNS/users/q8i/ADCore/ADApp/Db'
make[1]: *** [Db.install] Error 2
make[1]: Leaving directory `/SNS/users/q8i/ADCore/ADApp'
make: *** [ADApp.install] Error 2
```

If I do "touch db/NDFFT_settings.req" that error goes away but then I get an error for "NDFileJPEG_settings.req", and so on. How are these .req files being generated? At first I thought this may have been an issue with the autosave version, but I tried compiling with autosave 5-5 and I am still getting this error for ADCore R3-11 and R3-12

jwlodek, 2023-03-10:
I believe the files are all  being pulled into `ADCore/db` via this line in the `Makefile` in `ADApp/Db`: https://github.com/areaDetector/ADCore/blob/6d661b62ace7704be5c9037d24e2401041238f00/ADApp/Db/Makefile#L16

FWIW Both R3-11 and 3-12 build OK for me

bsobhani, 2023-03-10:
Thanks. It works if I manually do "cp ADApp/Db/*.req db/"

I am using make 3.8.2

MarkRivers, 2023-03-10:
It work fine for me with base 7.0.7 and make 3.8.2.

This is what I see when I run "make clean uninstall" and then "make". 
```
corvette:~/devel/areaDetector/ADCore>make clean uninstall
make -C ./configure clean
make[1]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/configure'
rm -rf O.linux-x86_64 O.vxWorks-ppc32 O.Common
make[1]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/configure'
make -C ./ADApp clean
make[1]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp'
make -C ./Db clean
make[2]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/Db'
rm -rf O.linux-x86_64 O.vxWorks-ppc32 O.Common
make[2]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/Db'
make -C ./ADSrc clean
make[2]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/ADSrc'
rm -rf O.linux-x86_64 O.vxWorks-ppc32 O.Common
make[2]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/ADSrc'
make -C ./ntndArrayConverterSrc clean
make[2]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/ntndArrayConverterSrc'
rm -rf O.linux-x86_64 O.vxWorks-ppc32 O.Common
make[2]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/ntndArrayConverterSrc'
make -C ./pluginSrc clean
make[2]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/pluginSrc'
rm -rf O.linux-x86_64 O.vxWorks-ppc32 O.Common
make[2]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/pluginSrc'
make -C ./pluginTests clean
make[2]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/pluginTests'
rm -rf O.linux-x86_64 O.vxWorks-ppc32 O.Common
make[2]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/pluginTests'
make -C ./op clean
make[2]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/op'
make[2]: Nothing to be done for `clean'.
make[2]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/op'
make[1]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp'
rm -rf ./bin/linux-x86_64 ./lib/linux-x86_64
rm -rf ./bin/vxWorks-ppc32 ./lib/vxWorks-ppc32
rm -rf ./db ./dbd ./doc ./html ./include ./templates ./adl ./alh ./cfg ./edl ./lib/perl ./lib/pkgconfig ./db
rm -rf
corvette:~/devel/areaDetector/ADCore>make
make -C ./configure install
make[1]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/configure'
perl -CSD /corvette/usr/local/epics-devel/base-7.0.7/bin/linux-x86_64/makeMakefile.pl O.linux-x86_64 ../..
perl -CSD /corvette/usr/local/epics-devel/base-7.0.7/bin/linux-x86_64/makeMakefile.pl O.vxWorks-ppc32 ../..
mkdir -p O.Common
make -C O.linux-x86_64 -f ../Makefile TOP=../.. \
    T_A=linux-x86_64 install
make[2]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/configure/O.linux-x86_64'
perl -CSD /corvette/usr/local/epics-devel/base-7.0.7/bin/linux-x86_64/convertRelease.pl checkRelease
make[2]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/configure/O.linux-x86_64'
make -C O.vxWorks-ppc32 -f ../Makefile TOP=../.. \
    T_A=vxWorks-ppc32 install
make[2]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/configure/O.vxWorks-ppc32'
perl -CSD /corvette/usr/local/epics-devel/base-7.0.7/bin/linux-x86_64/convertRelease.pl checkRelease
make[2]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/configure/O.vxWorks-ppc32'
make[1]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/configure'
make -C ./ADApp install
make[1]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp'
make -C ./Db install
make[2]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/Db'
perl -CSD /corvette/usr/local/epics-devel/base-7.0.7/bin/linux-x86_64/makeMakefile.pl O.linux-x86_64 ../../..
perl -CSD /corvette/usr/local/epics-devel/base-7.0.7/bin/linux-x86_64/makeMakefile.pl O.vxWorks-ppc32 ../../..
mkdir -p O.Common
make -C O.linux-x86_64 -f ../Makefile TOP=../../.. \
    T_A=linux-x86_64 install
make[3]: Entering directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/Db/O.linux-x86_64'
Installing ../../../db/NDFFT_settings.req
mkdir ../../../db
Installing ../../../db/NDFileJPEG_settings.req
[... 51 lines cut ...]
Installing template file ../../../db/NDAttrPlotData.template
Installing template file ../../../db/ADPrefixes.template
Installing template file ../../../db/NDAttrPlotAttr.template
Installing template file ../../../db/NDTimeSeries.template
Installing template file ../../../db/NDFileHDF5.template
Installing template file ../../../db/NDROIStatN.template
Installing template file ../../../db/NDCodec.template
Installing template file ../../../db/ADBase.template
Installing template file ../../../db/NDGather.template
Installing template file ../../../db/NDPva.template
Installing template file ../../../db/NDPosPlugin.template
Installing template file ../../../db/NDROIStat8.template
Installing template file ../../../db/NDFileMagick.template
Installing template file ../../../db/NDFileJPEG.template
Installing template file ../../../db/NDProcess.template
Installing template file ../../../db/CCDMultiTrack.template
Installing template file ../../../db/NDROI.template
Installing template file ../../../db/NDTimeSeriesN.template
Installing template file ../../../db/NDColorConvert.template
Installing template file ../../../db/NDPluginBase.template
Installing template file ../../../db/NDTransform.template
make[3]: Leaving directory `/home/epics/devel/areaDetector-3-12-1/ADCore/ADApp/Db/O.linux-x86_64'
```

Note that it installed NDFFT_settings.req.  What do you see when you run those commands?

MarkRivers, 2023-03-10:
Just to be clear, NDFFT_settings.req is distributed in ADCore/ADApp/Db.  Running make in that directory should install that file into ADCore/db.

MarkRivers, 2023-03-10:
This is the commit that changed the Db file to use wildcards, rather than specifying individual files.  f29191b
That is the most recent commit on that file, and was made on April 28, 2021.  

R3-11 was released on May 26, 2021, so it has that change.  Does your version of the Makefile use the wildcards?

If so, and it is not working, then I suspect the problem is because you are running a very old version of base.  It works for me with the same version of make as you (3.8.2) but a newer version of base.

bsobhani, 2023-03-10:
Yes it uses wildcards and I do agree the EPICS Base version is most likely the problem but I do not think the wildcards have anything to do with it because I left in the other two lines that use the wildcards. It is just that files ending in a .req extension do not get copied, even if I bypass the wildcard mechanism.

MarkRivers, 2023-03-10:
I agree, there must be a rule in that version of base that only copies certain file extensions.

closed 2024-01-01 by MarkRivers