areaDetector/ADCore #492: Add thresholding option to NDPluginProcess

xiaoqiangwang, 2023-05-25:
NDPluginProcess currently has the clipping algorithms
https://github.com/areaDetector/ADCore/blob/49f9cce7ffa6331d947986bfa54fa6906d540d79/ADApp/pluginSrc/NDPluginProcess.cpp#L169-L170

In practice, it is often desired to set the outliers to 0 instead of the clipping values. This improves the contrast and reduces the background when x/y profile is summed.

This could be either implemented as a separate processing step, or as  two options (one for low and one for high).
```
enum ClipMode {ClipModeClamp, ClipModeThreshold};
...

int lowClipMode, highClipMode;
double lowClipValue, highClipValue;
getIntegerParam(NDPluginProcessLowClipMode, lowClipMode);
getIntegerParam(NDPluginProcessHighClipMode, highClipMode);
lowClipValue  = (lowClipMode == ClipModeThreshold) ? 0 : lowClip;
highClipValue  = (highClipMode == ClipModeThreshold) ? 0 : highClip;

...
        if (enableHighClip && (value > highClip)) value = highClipValue;
        if (enableLowClip  && (value < lowClip))  value = lowClipValue;
```

MarkRivers, 2023-08-15:
I think it would be better to make this more general replacing with 0 or the clip value.  Users might want to replace the clip values with non-zero user-specified values, for example -1, -2, etc.

I will work on this soon.

xiaoqiangwang, 2023-09-12:
Implemented eb74e5f

closed 2023-09-12 by xiaoqiangwang