areaDetector/ADCore #495: Github Actions failing due to asyn

MarkRivers, 2023-10-01:
Github actions fails because asyn fails to build due to TIRPC issue.

Is there a way to change the asyn build for Github Actions for ADCore to set CONFIG_SITE to use tirpc?

MarkRivers, 2023-10-01:
@ralphlange can you advise on this?

mdavidsaver, 2023-10-01:
An example of setting `TIRPC=yes` from a ci-script hook:

https://github.com/mdavidsaver/ndwarp/blob/cf1bedfd1497dad85ab628c989b694fae11ab75b/.ci-local/defaults.set#L10

https://github.com/mdavidsaver/ndwarp/blob/cf1bedfd1497dad85ab628c989b694fae11ab75b/.ci-local/asyn-config.sh#L9

Although I think  a more efficient, although more complex, solution would be for asyn to auto-detect when tirpc is needed.  Either from make (cf. https://github.com/epics-modules/asyn/pull/187), or from source with eg. `#if __has_include(<tirpc/rpc/rpc.h>)`.