areaDetector/ADCore #502: Support XML string as well as file path in `NDAttributesFile`

coretl, 2024-01-31:
In bluesky we would like to configure an areaDetector without having to have a shared filesystem with the detector. This is almost possible at the moment, but if we need to set `NDAttributesFile` then we need to pass a filesystem path which is read here:
https://github.com/areaDetector/ADCore/blob/9d2c32a210408cdc353331a4353a0b416d95444a/ADApp/ADSrc/asynNDArrayDriver.cpp#L329-L338

The pos plugin has a similar XML reading routine, but it supports the PV being either a blob of XML or a filename:
https://github.com/areaDetector/ADCore/blob/9d2c32a210408cdc353331a4353a0b416d95444a/ADApp/pluginSrc/NDPosPluginFileReader.cpp#L65-L69

I propose we do something similar for `NDAttributesFile`:
```cpp
if (fileName.find("<Attributes>") != std::string::npos){ 
    buffer = fileName;
} else {
    // The existing code
} 
```

Is this a reasonable thing to do? If so, then we can make a PR for it.

MarkRivers, 2024-01-31:
That seems reasonable to me.

coretl, 2024-02-01:
@marcelldls please could you take a look at this?

closed 2024-02-09 by MarkRivers

MarkRivers, 2024-02-10:
Closed via #503

PR #503 (2024-02-09): Support XML string and file path for NDAttributesFile
closes #502 @coretl Tested against the example XML from ADCore/ADApp/ADSrc/asynNDArrayDriver.cpp