areaDetector/ADCore #511: Can we import instrument-specific XML attributes file?

prjemian, 2024-08-27:
We want a main attributes file that imports from a detector-specific file.  Is this possible now?  What work would it take to add it?

EmilioPeJu, 2024-10-03:
Do you mean if including another file is supported?

prjemian, 2024-10-03:
Yes, where the main attributes file imports another attributes file
specific to that detector.

MarkRivers, 2024-10-03:
@prjemian what is the syntax for that? Are there any examples in AdSimDetector or elsewhere?

mp49, 2024-11-11:
We handle this by using an external script that builds multiple XML files, each one for a different detector. 

Although, there seems to be built-in XML support for this:
https://stackoverflow.com/questions/5121052/can-we-import-xml-file-into-another-xml-file