areaDetector/ADCore #512: change "NDAttributes.xsd" to support .xml containing macro characters, $, (, )

cpchuang, 2024-09-18:
current "NDAttributes.xsd" invalidate .xml files containing $, (, )

```
Validating xml file against "./XML_schema/NDAttributes.xsd"
Attri_ADsim.xml:9: element Attribute: Schemas validity error : Element 'Attribute', attribute 'source': '$(P)cam1:Manufacturer_RBV' is not a valid value of the union type 'type_source'.
```
After discussing with @prjemian , propose the following changes to include $, (, ),

```
    <xs:restriction base="xs:string"> <!-- Changed from xs:NMTOKEN to xs:string -->
      <xs:pattern value="[\w_\-:.;\$\(\)]+([.][A-Z0-9]+)?"/>   <!-- include $, (, ),  -->
    </xs:restriction>
  </xs:simpleType>
```

cpchuang, 2024-09-18:
for clarity, 
Here is the example attributes.xml
[image]

and how macro is added in AD.
[image]

prjemian, 2024-12-20:
I will look into this shortly.

PR #522 (2025-01-16): Support macro characters in attributes XML files
- close #512

Thanks, @cpchuang, for the suggestion!
closed 2025-01-16 by MarkRivers