areaDetector/ADCore #521: NDArrayPool statistics not working in main driver

mp49, 2024-12-12:
The array pool statistics for the main driver all report 0 and are never processed except once at IOC startup.

asynReport shows:

Parameter 53 type=asynInt32, name=ARRAY_CALLBACKS, value=1, status=0
Parameter 54 type=asynInt32, name=POOL_MAX_BUFFERS, value is undefined
Parameter 55 type=asynInt32, name=POOL_ALLOC_BUFFERS, value=0, status=0
Parameter 56 type=asynInt32, name=POOL_PRE_ALLOC_BUFFERS, value is undefined
Parameter 57 type=asynInt32, name=POOL_NUM_PRE_ALLOC_BUFFERS, value=100, status=0
Parameter 58 type=asynInt32, name=POOL_FREE_BUFFERS, value=0, status=0
Parameter 59 type=asynFloat64, name=POOL_MAX_MEMORY, value=0, status=0
Parameter 60 type=asynFloat64, name=POOL_USED_MEMORY, value=0, status=0
Parameter 61 type=asynInt32, name=POOL_EMPTY_FREELIST, value is undefined
Parameter 62 type=asynInt32, name=POOL_POLL_STATS, value=0, status=0

So it seems the records are correct (except the parameters shouldn't be zero).

Further down in the asynReport, I see:

T1: private NDArrayPool report

NDArrayPool:
  numBuffers=923, numFree=900
  memorySize=1931974688, maxMemory=-1
  freeList: (index, dataSize, pArray)
    0 184 0x7f5374011de0
    1 2048 0x7f5374011bb0
    2 4096 0x7f53a0037e20
    3 66664 0x7f538c0013a0
    4 66664 0x7f5384006650
etc, up to 899

The plugin drivers report a 'shared' NDArrayPool:

T1.1DROI: shared NDArrayPool report

NDArrayPool:
  numBuffers=923, numFree=900
  memorySize=1931974688, maxMemory=-1
  freeList: (index, dataSize, pArray)
    0 184 0x7f5374011de0
    1 2048 0x7f5374011bb0
    2 4096 0x7f53a0037e20
    3 66664 0x7f538c0013a0
    4 66664 0x7f5384006650

and for the plugins, the records are reporting the correct non-zero values.

Any ideas?

I'm not sure if this is a bug, but it was working pre-3-14.

MarkRivers, 2024-12-12:
The way those records are scanned did change in R3-14. They are now I/O Intr scanned, but still in response to the same PoolPollStats record which is periodically scanned, with a default scan period of 1 second. I am seeing those stats update OK for my drivers. Can you post a screenshot of your main detector screen?

mp49, 2024-12-12:
Thanks Mark, yes I saw the changes and I couldn't see any reason why it wouldn't work. I've attached the ADBase screenshot.
[image]

MarkRivers, 2024-12-12:
It looks like buffer and memory polling is set to “1”.   I think that means the second enum state for the SCAN field, which is probably not periodic. The looks like a bug in the original medm screen where that widget is set to “decimal”, not “string”. Please make sure the SCAN field of that record is 1 second and see if the values update.

mp49, 2024-12-12:
Just to check, I switched back to my 3.12.1 version, and it seems to work fine (screenshot attached - screen is from the newer 3-14 release).
[image]

MarkRivers, 2024-12-12:
I just tested ADSimDetector with R3-14 and it worked for me.

mp49, 2024-12-13:
I found the problem in my driver. I wasn't calling the base class ADDriver::writeInt32 in my derived class writeInt32, so the function in asynNDArrayDriver wasn't being called. 

It worked in earlier ADCore versions because the parameters were being set via the readInt32/readFloat64 functions, which I didn't override in my derived class.

Thanks for your help. I'll close this!

mp49, 2024-12-13:
Closing as user error

closed 2024-12-13 by mp49