areaDetector/ADCore #525: Error building HDF5 support (too few arguments to function H5Oget_info_by_name3)

ronpandolfi, 2025-03-04:
I'm trying to build with hdf5 support and encountered this error:

```
../HDF5FileReader.cpp: In function ‘herr_t file_info(hid_t, const char*, const H5L_info2_t*, void*)’:
../HDF5FileReader.cpp:15:25: error: too few arguments to function ‘herr_t H5Oget_info_by_name3(hid_t, const char*, H5O_info2_t*, unsigned int, hid_t)’
   15 |     H5Oget_info_by_name (loc_id, name, &infobuf, H5P_DEFAULT);
      |                         ^
In file included from /usr/include/H5Apublic.h:22,
                 from /usr/include/hdf5.h:23,
                 from ../HDF5FileReader.h:15,
                 from ../HDF5FileReader.cpp:8:
/usr/include/H5Opublic.h:557:15: note: declared here
  557 | H5_DLL herr_t H5Oget_info_by_name3(hid_t loc_id, const char *name, H5O_info2_t *oinfo, unsigned fields,
      |               ^~~~~~~~~~~~~~~~~~~~
```

A quick search reveals this is from an updated api in hdf5 as of version 1.12. On Rocky 9, that version is default.

MarkRivers, 2025-03-04:
The HDF plugin currently uses API version 1.10.0-patch1.  You cannot use that plugin with more recent API versions.  You can always use that version, because it is provided in ADSupport.

Someone could modify the plugin to work with more recent APIs.  But they would need to either preserve backwards compatibility with 1.10.0 or also update the version in ADSupport, making sure it still builds on Windows, Linux, Mac, vxWorks, etc.

ronpandolfi, 2025-03-05:
My mistake must have been specifying the hdf5 lib paths then. Building with the included version should be fine. I'll close this as its more my mistake.

Thanks again for the support, Mark.

closed 2025-03-05 by ronpandolfi