areaDetector/ADCore #536: github actions build failing on windows-x64

MarkRivers, 2025-06-16:
I tried adding the DLL build for ADCore to .github/workflows/ci-scripts.yml.  However, it fails with this error (I added line breaks to the output for readability):
```
link -nologo -subsystem:windows -dll -LTCG -incremental:no -opt:ref -release  -MACHINE:X64      -out:ADBase.dll -implib:ADBase.lib 
NDAttribute.obj NDAttributeList.obj NDArrayPool.obj NDArray.obj asynNDArrayDriver.obj ADDriver.obj paramAttribute.obj 
CCDMultiTrack.obj PVAttribute.obj functAttribute.obj parseAreaPrefixes.obj myTimeStampSource.obj myAttributeFunctions.obj        
C:/Users/runneradmin/.cache/asyn-master/lib/windows-x64/asyn.lib  C:/Users/runneradmin/.cache/base-7.0/lib/windows-x64/dbRecStd.lib
C:/Users/runneradmin/.cache/base-7.0/lib/windows-x64/dbCore.lib  C:/Users/runneradmin/.cache/base-7.0/lib/windows-x64/ca.lib  
C:/Users/runneradmin/.cache/base-7.0/lib/windows-x64/Com.lib xml2.lib
  LINK : fatal error LNK1181: cannot open input file 'xml2.lib'
```
xml2.lib is in ADSupport, but it is not prefixing xml2.lib with something like:  C:/Users/runneradmin/.cache/adsupport/lib/windows-x64/
If I build locally I see this:
```
link -nologo -subsystem:windows -dll -LTCG -incremental:no -opt:ref -release  -MACHINE:X64      -out:ADBase.dll -implib:ADBase.lib     
NDAttribute.obj NDAttributeList.obj NDArrayPool.obj NDArray.obj asynNDArrayDriver.obj ADDriver.obj paramAttribute.obj 
CCDMultiTrack.obj PVAttribute.obj functAttribute.obj parseAreaPrefixes.obj myTimeStampSource.obj myAttributeFunctions.obj        
J:/epics/devel/asyn-4-45/lib/windows-x64/asyn.lib  H:/epics-devel/base-7.0.8/lib/windows-x64/dbRecStd.lib  
H:/epics-devel/base-7.0.8/lib/windows-x64/dbCore.lib  H:/epics-devel/base-7.0.8/lib/windows-x64/ca.lib  
H:/epics-devel/base-7.0.8/lib/windows-x64/Com.lib  J:/epics/devel/areaDetector-3-14/ADSupport/lib/windows-x64/xml2.lib
Creating library ADBase.lib and object ADBase.exp
Generating code
Finished generating code
```
The dynamic build is told that adsupport is a deps:
```
          - name: Win2022 Base7.0 DLL
            os: windows-2022
            cmp: vs2022
            configuration: default
            base: "7.0"
            deps: adsupport
```
How do I make it look for xml2.lib in ADSupport?

FreddieAkeroyd, 2025-07-04:
I think the problem is with CONFIG_SITE when adCore is built standalone outside of normal areaDetector tree. I've made a few changes to a local fork and the github action now builds - see

https://github.com/areaDetector/ADCore/compare/master...FreddieAkeroyd:ADCore:master

and

https://github.com/FreddieAkeroyd/ADCore/actions/runs/16063481988

Summary of changes
- added options to CONFIG_SITE rather than CONFIG_SITE.local 

I see https://github.com/areaDetector/ADCore/blob/master/.ci-local/adsupport-config.py
run by https://github.com/areaDetector/ADCore/blob/84da95763e24ac2a76b37b99f2276b58afc71e59/.ci-local/adsupport.set#L6
sets a CONFIG_SITE as there is an issue with CONFIG_SITE.local in adsupport, is same true for adCore?

- added  `HDF5_STATIC_BUILD=NO` to CONFIG_SITE 

setting to $(STATIC_BUILD) as in e.g.  
https://github.com/areaDetector/areaDetector/blob/f4a39a23810a89e29f2fbb45798a105cb1665af3/configure/EXAMPLE_CONFIG_SITE.local.WIN32#L3 didn't work, need to look more

I'll investigate a bit more then create a proper PR

mdavidsaver, 2026-01-02:
At present, the DLL build fails to link with some `unresolved external symbol` errors.

```
  link -nologo -subsystem:windows -dll -LTCG -incremental:no -opt:ref -release  -MACHINE:X64      -out:NDPlugin.dll -implib:NDPlugin.lib     NDPluginDriver.obj throttler.obj NDPluginAttribute.obj NDPluginCircularBuff.obj NDArrayRing.obj NDPluginColorConvert.obj NDPluginFFT.obj fft.obj NDPluginGather.obj NDPluginOverlay.obj NDPluginOverlayTextFont.obj NDPluginProcess.obj NDPluginROI.obj NDPluginROIStat.obj NDPluginScatter.obj NDPluginStats.obj NDPluginStdArrays.obj NDPluginTimeSeries.obj NDPluginTransform.obj NDPluginAttrPlot.obj NDPluginCodec.obj NDPosPlugin.obj NDPosPluginFileReader.obj NDPluginFile.obj NDFileNull.obj NDFileHDF5.obj NDFileHDF5Dataset.obj NDFileHDF5AttributeDataset.obj NDFileHDF5LayoutXML.obj NDFileHDF5Layout.obj NDFileJPEG.obj NDFileNetCDF.obj NDFileNexus.obj NDFileTIFF.obj NDPluginPva.obj        ../../../lib/windows-x64/ADBase.lib  C:/Users/runneradmin/.cache/asyn-master/lib/windows-x64/asyn.lib  ../../../lib/windows-x64/ntndArrayConverter.lib  C:/Users/runneradmin/.cache/normativeTypes-master/lib/windows-x64/nt.lib  C:/Users/runneradmin/.cache/pvdatabase-master/lib/windows-x64/pvDatabase.lib  C:/Users/runneradmin/.cache/pvAccess-master/lib/windows-x64/pvAccess.lib  C:/Users/runneradmin/.cache/pvData-master/lib/windows-x64/pvData.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/netCDF.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/tiff.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/xml2.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/NeXus.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/hdf5.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/hdf5_hl.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/bitshuffle.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/blosc.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/szip.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/zlib.lib  C:/Users/runneradmin/.cache/adsupport-master/lib/windows-x64/jpeg.lib  C:/Users/runneradmin/.cache/base-7.0/lib/windows-x64/dbRecStd.lib  C:/Users/runneradmin/.cache/base-7.0/lib/windows-x64/dbCore.lib  C:/Users/runneradmin/.cache/base-7.0/lib/windows-x64/ca.lib  C:/Users/runneradmin/.cache/base-7.0/lib/windows-x64/Com.lib ws2_32.lib user32.lib gdi32.lib oleaut32.lib
     Creating library NDPlugin.lib and object NDPlugin.exp
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_NATIVE_FLOAT_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_NATIVE_INT32_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_NATIVE_UINT8_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5P_CLS_FILE_CREATE_ID_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5P_CLS_DATASET_CREATE_ID_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5P_CLS_DATASET_ACCESS_ID_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_NATIVE_INT16_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5P_CLS_FILE_ACCESS_ID_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_NATIVE_DOUBLE_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_NATIVE_UINT32_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_C_S1_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_NATIVE_INT64_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_NATIVE_UINT16_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_NATIVE_INT8_g
  NDFileHDF5.obj : error LNK2001: unresolved external symbol H5T_NATIVE_UINT64_g
  NDPlugin.dll : fatal error LNK1120: 15 unresolved externals
```

With hdf5 1.14 (circa Debian Linux 13), these symbols come from `libhdf5_serial.a`.  `hdf5.lib` is included, so I so no obvious reason why these symbols should be missing.

Maybe a reach, but I wonder if this is an issue with single pass linker ordering?  `libhdf5_serial_hl.so` has a NEEDS dependency on `libhdf5_serial.so.310`, however in the above `hdf5.lib` appears before `hdf5_hl.lib`.

FreddieAkeroyd, 2026-01-02:
I'm slightly suspicious of these symbol names - normally undefined DLL symbols start `__imp__` so this might suggest that `NDFileHDF5.obj` has been compiled to expect a static hdf5 library but is being linked to a DLL version.  It could be `HDF5_STATIC_BUILD` is getting set incorrectly/inconsistently during the build? My only worry on this is that it looks like it can't find a few global variables as opposed to a lot of functions, i'd probably expect a lot more errors if static/dynamic was getting mixed in general.  I don't believe the `.lib` library order usually matters with the windows linker, but there may be some exceptions.